hathach/tinyusb #858: HID Keyboard Endpoint Descriptor can not be according to standard causing -EOVERFLOW.

Algolian, 2021-05-29:
Im using a rp2040 with now behind tinyusb.

Im setting CFG_TUD_HID_BUFSIZE to 8 and recompile.
It works as expected and 
lsusb on a linux shows:
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10

This what the standard expect for HID keyboards. 
However calling tud_hid_keyboard_report(REPORT_ID_KEYBOARD, modifier, pressed_keys);
causes problem. This is what is sent to the host.

Frame 815: 73 bytes on wire (584 bits), 73 bytes captured (584 bits) on interface usbmon2, id 5
    Interface id: 5 (usbmon2)
    Encapsulation type: USB packets with Linux header and padding (115)
    Arrival Time: May 29, 2021 21:51:41.935900000 CEST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1622317901.935900000 seconds
    [Time delta from previous captured frame: 0.000887000 seconds]
    [Time delta from previous displayed frame: 0.000887000 seconds]
    [Time since reference or first frame: 1.645144000 seconds]
    Frame Number: 815
    Frame Length: 73 bytes (584 bits)
    Capture Length: 73 bytes (584 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: usb:usbhid]
USB URB
    [Source: 2.13.1]
    [Destination: host]
    URB id: 0xffff9b25869af900
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_INTERRUPT (0x01)
    Endpoint: 0x81, Direction: IN
    Device: 13
    URB bus id: 2
    Device setup request: not relevant ('-')
    Data: present (0)
    URB sec: 1622317901
    URB usec: 935900
    URB status: Value too large for defined data type (-EOVERFLOW) (-75)
    URB length [bytes]: 9
    Data length [bytes]: 9
    [Request in: 618]
    [Time from request: 0.055942000 seconds]
    [bInterfaceClass: HID (0x03)]
    Unused Setup Header
    Interval: 8
    Start frame: 0
    Copy of Transfer Flags: 0x00000204, No transfer DMA map, Dir IN
    Number of ISO descriptors: 0
HID Data: 010000040000000001
0000   00 f9 9a 86 25 9b ff ff 43 01 81 0d 02 00 2d 00   ....%...C.....-.
0010   4d 9b b2 60 00 00 00 00 dc 47 0e 00 b5 ff ff ff   M..`.....G......
0020   09 00 00 00 09 00 00 00 00 00 00 00 00 00 00 00   ................
0030   08 00 00 00 00 00 00 00 04 02 00 00 00 00 00 00   ................
0040   01 00 00 04 00 00 00 00 01                        .........

The URB length is 9 while the  wMaxPacketSize  is 8 so it is a overflow.
This is a key pressed event, and the first 8 bytes on the 0040 line is right. But there is a 01 as the last byte that
should not be there!

hathach, 2021-05-30:
Please help us to reproduce by providing enough info
- Fill out the requirement information for the hw setup
- Provide your simplest sketch to reproduce in quote
- Provide your serial output in debug mode in quote.

hathach, 2021-05-31:
The issue still doesn't match the template, to reproduce and analyze the issue one need the exact firmware, the log. Also you will need to use the latest version on the master to count as a bug. Ping me when you got it updated.

closed 2021-05-31 by hathach

Algolian, 2021-05-31:
Your branch does not compile on pico pi branch. It make the pio fail. And the maintainer of pico
does want to update for some reason.  However I don't think this bug is releated to that. 
Im not sure if the the bug is pico related or generic.  You seems to be very uninterested that people
help you to test your code.

Algolian, 2021-06-03:
Now raspberry have updated the SDK so I think im on latest and gratest.
It is very easy to reproduce with hid_composite example.
Apply this little change:
Subject: [PATCH] Buffer overflow demo

This chages the endpoint buffer size from 16 to 8.
This is what usb keyboards when they are hid keyboard.
---
 examples/device/hid_composite/src/tusb_config.h | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/examples/device/hid_composite/src/tusb_config.h b/examples/device/hid_composite/src/tusb_config.h
index 3e608ed3..2951c37d 100644
--- a/examples/device/hid_composite/src/tusb_config.h
+++ b/examples/device/hid_composite/src/tusb_config.h
@@ -102,7 +102,7 @@
 #define CFG_TUD_VENDOR            0
 
 // HID buffer size Should be sufficient to hold ID (if any) + Data
-#define CFG_TUD_HID_EP_BUFSIZE    16
+#define CFG_TUD_HID_EP_BUFSIZE    8
 
 #ifdef __cplusplus
  }
-- 
2.31.1

It will change the hid descriptor:
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0xcafe 
  idProduct          0x4004 
  bcdDevice            1.00
  iManufacturer           1 TinyUSB
  iProduct                2 TinyUSB Device
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
[... 187 lines cut ...]
However it fails in the other end.
From the host log:
[805997.220073] usb 2-1: new full-speed USB device number 96 using uhci_hcd
[805997.528442] usb 2-1: New USB device found, idVendor=cafe, idProduct=4004, bcdDevice= 1.00
[805997.528463] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[805997.528473] usb 2-1: Product: TinyUSB Device
[805997.528482] usb 2-1: Manufacturer: TinyUSB
[805997.528490] usb 2-1: SerialNumber: 123456
[805997.624441] input: TinyUSB TinyUSB Device Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1:1.0/0003:CAFE:4004.0098/input/input161
[805997.679481] input: TinyUSB TinyUSB Device Mouse as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1:1.0/0003:CAFE:4004.0098/input/input162
[805997.680975] input: TinyUSB TinyUSB Device as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1:1.0/0003:CAFE:4004.0098/input/input163
[805997.683759] hid-generic 0003:CAFE:4004.0098: input,hidraw2: USB HID v1.11 Keyboard [TinyUSB TinyUSB Device] on usb-0000:00:1d.0-1/input0
[806013.055527] usb 2-1: input irq status -75 received
[806013.087518] usb 2-1: input irq status -75 received
[806013.099525] usb 2-1: input irq status -75 received
[806013.127554] usb 2-1: input irq status -75 received
[806013.139521] usb 2-1: input irq status -75 received
[806013.167592] usb 2-1: input irq status -75 received
[806013.179533] usb 2-1: input irq status -75 received
[806013.207526] usb 2-1: input irq status -75 received
[806013.219531] usb 2-1: input irq status -75 received

-75 is a buffer overflow and wireshare also detect this error.

The tinyusb log thing it is sending 8, but it get 9 on the other end.

Algolian, 2021-06-07:
There is some similarities with #813 as they both send 8 bytes payloads. @Makurisan