hathach/tinyusb #872: CDC hardware flow control can not be disabled

robert-rozee, 2021-06-03:
in a particular user case i need to be able to disable CDC hardware flow control at the device end. it is not practical for the host end to simply assert DTR due to the need for backwards compatibility with other software. the feature i'm requesting is a flag to disable hardware flow control that is exposed via the api.

 fyi, the  hardware i working with is an RP2040. 

my proposal is a bit of a hack, i'd welcome better suggestions. in **class/cdc/cdc_device.c** make the following changes:

line 116
`return tud_ready() && (tu_bit_test(_cdcd_itf[itf].line_state, 0) || flag_ignore_DTR);`

line 394
`if (!flag_ignore_DTR) tu_fifo_set_overwritable(&p_cdc->tx_ff, !dtr);`

where **flag_ignore_DTR** is exposed through the api.

cheers,
rob   :-)

hathach, 2021-06-20:
The existing code still works without DTR. However due to no flow control, it wouldn't know if there is any terminal connected and therefore have no way to tell when the fifo will be drained. Therefore the fifo mode is set to overwritable.

robert-rozee, 2021-06-21:
on the RP2040, after sending around 1000 characters from a (linux) host **without** DTR being asserted (characters which are being echoed back by the RP2040), the USB subsystem on the host locks up solidly for around 30 seconds before returning a 'try again' error message back to the user application. while this seems to be a problem with the host/linux, it would be good for the RP2040/tinyUSB to instead be configurable to ignore the state of DTR and just keep sending characters back to the host even when DTR is not asserted. i can think of few cases where the host would **genuinely** need to signal to tinyUSB to stop sending characters.

the test environment i'm using here now is micropython (RPi foundations, built less than a week back) running on the RP2040, and a terminal emulator on the linux host that is able to independently manipulate DTR.

surely if there is no USB connection present, the are less subtle signs than checking the state of DTR? such as missing heartbeat from the host, Vusb not detected, etc?

cheers,
rob   :-)

hathach, 2021-06-21:
what is the issue with current implementation that you want to change ?

robert-rozee, 2021-06-21:
at the moment, with RP2040+tinyUSB, if DTR is not asserted by the host (in my case a desktop PC running linux), the RP2040 stops sending characters to the host. it looks like tinyUSB attempts to buffer these characters. after approximately 1000 characters have been buffered a malfunction occurs. this scenario can be demonstrated with micropython running on a RP2040 (micropython uses tiny USB).

**what i would like to be able to do, is to configure tinyUSB so that it ignores the state of DTR and keeps sending characters irrespective of the current state of DTR. ie, i would like the connection to act as if hardware flow control is turned OFF.**

in my case, there are good reasons why the terminal emulator can not alter the state of DTR. my application running on the RP2040 is written in C, however the behaviour of micropython is able to sufficiently reproduce and demonstrate the problem.

cheers,
rob   :-)

hathach, 2021-06-21:
> at the moment, with RP2040+tinyUSB, if DTR is not asserted by the host (in my case a desktop PC running linux), the RP2040 stops sending characters to the host. it looks like tinyUSB attempts to [...]

Have you tried to figured out why tinyusb stop sending data to host. Any log from stack ? It seems to be a bug to me.

> **what i would like to be able to do, is to configure tinyUSB so that it ignores the state of DTR and keeps sending characters irrespective of the current state of DTR. ie, i would like the connection to act as if hardware flow control is turned OFF.**

Could you provide information as bug template, and steps to reproduce the issue. Once we figure out exactly what went wrong. We could decide which is the best fix.

robert-rozee, 2021-06-22:
the software engineer developing for the RP2040 is located in the UK, the other side of the world to where i am (in New Zealand). his perspective is that it works ok for him (win10, using teraterm as a terminal emulator) and he has no interest in testing for anything other than his own setup; he is a 'challenging' person to deal with. the source code will only become available to me once his code is 'released to production'.

hence my reproducing the behaviour with micropython, where the source is readily to hand. i'm working at the host/pc end, with my area of expertise is there - not development on the PR2040. if a solution can be found within tinyUSB, this will flow through into the RP2040 SDK, which will then flow through into our project.

what i can do is ring-fence the behaviour as precisely as possible, so that others may reproduce it.

cheers,
rob   :-)

hathach, 2021-06-22:
I see and understand your problem since you are not familiar with mcu. However for this issue to be addressed, and if you still want it to. Please update your 1st post to provide all the setup information with specific version of each software (mp, picosdk, tinyusb) and step to reproduce issue. Then change the label to Bug. Though since I don't often use mp, this will have huge delay for me to even try to reproduce it.

closed 2021-06-29 by hathach