hathach/tinyusb #884: net_lwip_webserver reaches "endpoint busy" race condition on STM32F411DISCO

kamathba, 2021-06-08:
**Set Up**
Compile the example: `make DEBUG=1 BOARD=stm32f411disco all`

- **Ubuntu 20.04.2 LTS**
- **STM32F411DISCO**
- **master (1b8473902fb0e808c32183f306938accc9ce0d5)**
- **Firmware** e.g examples/device/net_lwip_webserver

**Describe The Bug**

I reach the following assert when connecting to my host pc over usb:
```
usbd_edpt_xfer (rhport=rhport@entry=0 '\000', ep_addr=129 '\201', buffer=buffer@entry=0x2000001c <ndis_report> "\001", total_bytes=<optimized out>)
    at /home/bkamath/.dev/fb/tinyusb/src/device/usbd.c:1249
1249	  TU_ASSERT(_usbd_dev.ep_status[epnum][dir].busy == 0);
```

With the following backtrace:
```
#0  usbd_edpt_xfer (rhport=rhport@entry=0 '\000', ep_addr=129 '\201', buffer=buffer@entry=0x2000001c <ndis_report> "\001", total_bytes=<optimized out>)
    at /home/bkamath/.dev/fb/tinyusb/src/device/usbd.c:1249
#1  0x080014bc in netd_report (buf=buf@entry=0x2000001c <ndis_report> "\001", len=len@entry=8) at /home/bkamath/.dev/fb/tinyusb/src/class/net/net_device.c:122
#2  0x08001526 in rndis_report () at /home/bkamath/.dev/fb/tinyusb/lib/networking/rndis_reports.c:81
#3  0x080015e6 in rndis_query_cmplt32 (status=status@entry=0, data=data@entry=0) at /home/bkamath/.dev/fb/tinyusb/lib/networking/rndis_reports.c:94
#4  0x08002298 in rndis_query () at /home/bkamath/.dev/fb/tinyusb/lib/networking/rndis_reports.c:127
#5  0x0800245c in rndis_class_set_handler (data=data@entry=0x20001f6c <notify> "\004", size=<optimized out>) at /home/bkamath/.dev/fb/tinyusb/lib/networking/rndis_reports.c:267
#6  0x08003306 in netd_control_xfer_cb (rhport=<optimized out>, stage=<optimized out>, request=0x20000068 <_ctrl_xfer>) at /home/bkamath/.dev/fb/tinyusb/src/class/net/net_device.c:339
#7  0x080016e8 in usbd_control_xfer_cb (rhport=<optimized out>, ep_addr=<optimized out>, result=<optimized out>, xferred_bytes=32) at /home/bkamath/.dev/fb/tinyusb/src/device/usbd_control.c:209
#8  0x080036a4 in tud_task () at /home/bkamath/.dev/fb/tinyusb/src/device/usbd.c:545
#9  0x0800a246 in main () at /home/bkamath/.dev/fb/tinyusb/examples/device/net_lwip_webserver/src/main.c:237
```

I'm not super familiar with the code-base yet, so I'm not sure if this is just some sort of RTOS/platform issue, or the net_device code not meeting the pre-conditions for calling `usbd_edpt_xfer` inside of `netd_report`.

Not sure if it's helpful, but this is what my dmesg output looks like on the host side:
```
[ 2895.868565] usb 1-1.1.2: USB disconnect, device number 27
[ 2896.618695] usb 1-1.1.2: new full-speed USB device number 28 using xhci_hcd
[ 2896.737422] usb 1-1.1.2: New USB device found, idVendor=cafe, idProduct=4020, bcdDevice= 1.01
[ 2896.737424] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2896.737425] usb 1-1.1.2: Product: TinyUSB Device
[ 2896.737426] usb 1-1.1.2: Manufacturer: TinyUSB
[ 2896.737426] usb 1-1.1.2: SerialNumber: 123456
[ 2901.925172] rndis_host 1-1.1.2:1.0: RNDIS_MSG_QUERY(0x00010202) failed, -110
[ 2907.044517] rndis_host 1-1.1.2:1.0: RNDIS_MSG_QUERY(0x01010101) failed, -110
[ 2907.044524] rndis_host 1-1.1.2:1.0: rndis get ethaddr, -110
```

**To Reproduce**

Build example and connect to a linux machine.

**Log**

WIP, will upload if requested

kamathba, 2021-06-08:
Found issue #289 after posting this. It (along with a few of the MRs) have a fair amount of discussion, I'm not positive this isn't a duplicate.

hathach, 2021-06-14:
thanks for the issue, I also encouter the ASSERT with 0x81 endpoint, however, it seems to be "**normal**" since currently the net driver doesn't check the busy state of the notification endpoint and can send one while previous one doesn't received by host yet. This will be fixed later but in general it does affect the example as this also appear on other working MCUs as well.

kamathba, 2021-06-14:
Thanks for confirming. What would be the correct way to handle this at the application layer if I wanted to work around it? Do I need to make it more into a queue + state machine? Or can I just not send if the endpoint is busy (will the host eventually retry)?

Happy to make a PR if I can understand the problem well enough.

majbthrd, 2021-08-01:
Alas, this seems to be a "feature" of the RNDIS driver bundled into recent Linux distributions.  This newer Linux RNDIS driver fires off a bunch of RNDIS queries (via EP0) without first reading the notification EP response to the query/queries that it already sent.  D'oh.

The workaround that I’m tempted to submit as a PR would be to add an if-busy check to netd_report() in ./src/class/net/net_device.c:

```
void netd_report(uint8_t *buf, uint16_t len)
{
  // skip if previous report not yet acknowledged by host
  if ( usbd_edpt_busy(TUD_OPT_RHPORT, _netd_itf.ep_notif) ) return;
  usbd_edpt_xfer(TUD_OPT_RHPORT, _netd_itf.ep_notif, buf, len);
}
```