hathach/tinyusb #900: Only first bulk OUT EP write works

diggit, 2021-06-18:
###  Set Up

- Archlinux, distro kernel 5.12.8, 
- custom board with STM32H730 and USB PHY USB3343 (may provide schematics on request)
- **TinyUSB version** b34724215bea28ef93fd80ed9c2e4ddc87874093 with [synopsys hotfix ](https://github.com/diggit/tinyusb/commit/69e9397b8ac5201db2a79041593e76b31563c9d9) #496
#### Firmware
- based on RTOS ([modified Argon](https://github.com/diggit/argon-rtos))
- osal implemented and queue passes events to tud_task
##### EPs
```c
uint8_t const desc_hs_configuration[] = {
	// Config number, interface count, string index, total length, attribute, power in mA
	TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 500),

	TUD_VENDOR_DESCRIPTOR(ITF_NUM_VENDOR_0, 4, EPNUM_VENDOR_0, 0x80 | EPNUM_VENDOR_0, 512)
};
```

### Describe The Bug
Driver callback is called only after first bulk OUT xfer.

### To Reproduce

Steps to reproduce the behavior:
1. Write more than once from master to bulk OUT EP.
3. Observe, that vendor callback is called only in response to first xfer.

### Logs
I've added logging of `int_status` in the beginning of `dcd_int_handler` in `dcd_synopsys.c` (`TU_LOG2("(IS %08X)", int_status);`)
This log is enclosed in parentheses for easier reading due to its async nature.
There is less interrupts during second xfer.

#### First try
```
(IS C4808428)USBD Resume 
(IS 44808438)(IS 44808428)
USBD Setup Received 80 00 00 00 00 00 02 00 
  Get Status
  Queue EP 80 with 2 bytes ... (IS 44848428)(IS 44848428)OK
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with 0 bytes ... (IS 44808438)(IS 44808428)OK

USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
  Queue EP 80 with 0 bytes ... O(IS 44848428)K
USBD Xfer Co(IS 44808438)(IS 44808428)mplete on EP 00 with 0 bytes
USBD Xfer Complete on EP 80 with 0 bytes
USBD Xfer Complete on EP 01 with 4 bytes
  VENDOR xfer callback
  Queue EP 01 with 512 bytes ... OK
(IS 44808C28)USBD Suspend 
```
#### Second try
```
(IS C4808428)USBD Resume 
(IS 44808438)(IS 44808428)
USBD Setup Received 80 00 00 00 00 00 02 00 
  Get Status
  Queue EP 80 with 2 bytes ... (IS 44848428)(IS 44848428)OK
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with 0 bytes ... (IS 44808438)(IS 44808428)OK

USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
  Queue EP 80 with 0 bytes ... (IS 44848428)OK
USBD Xfer Complete on EP 00 with 0 bytes
USBD Xfer Complete on EP 80 with 0 bytes
(IS 44808C28)USBD Suspend 
```
#### Full log without int flags logging
[log_sniff_fail_on_second_xfer_wr.txt](https://github.com/hathach/tinyusb/files/6679536/log_sniff_fail_on_second_xfer_wr.txt)

#### pcap capture from host
[usb_sniff_fail_on_second_xfer_wr.zip](https://github.com/hathach/tinyusb/files/6679541/usb_sniff_fail_on_second_xfer_wr.zip)
Everything seems nominal or at least I can't spot difference between two writes.

hathach, 2021-06-20:
There is no clues if this is an tinyusb issue. There is too many custom changes from your setup. And there is no way for me to understand and track down the issue. If you could reproduce the issue with
- one of supported board
- one of example with slight modification

Then I could possible help you with narrow it down.

diggit, 2021-06-28:
With `python-libusb1` library,  issue seems to be non-existent. I'll try to compare behavior of  `python-pyusb` and `python-libusb1.`

hathach, 2021-06-29:
The issue seems to be the **stack's vendor driver**, currently it doesn't guard against race condition as cdc. I haven't updated the vendor driver once introducing the usbd_edpt_claim() for race protection. I actually have an plan to abstract these to stream/buffered endpoints API for driver and haven't got time to do so.

https://github.com/hathach/tinyusb/blob/master/src/class/vendor/vendor_device.c#L109
https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c#L83

diggit, 2021-06-30:
Would it be feasible to port cdc's guarding to vendor class driver?
IMO It would be better to have vendor class driver working until there is some unified streamed EPs API.

hathach, 2021-06-30:
yeah, it is better to fix existing problem with vendor right now. If you could submit PR, I am happy to review. Otherwise, give me a bit of time, I will do it when I have time.

diggit, 2021-07-04:
OT: looking onto claim/release functions, they are using mutex with RTOS... Did you consider using [atomic_compare_exchange](https://en.cppreference.com/w/cpp/atomic/atomic_compare_exchange)? It may require chaging that struct to bitfield (I am not very familiar with atomics in C, I work in C++), but locking mutex seems like an overkill for this.

kkitayam, 2021-07-04:
I guess the method is still under consideration. #662 mentions atomic operations for ARM architecture.
FYI, RTX implements [atmic_wr8](https://github.com/ARM-software/CMSIS_5/blob/2ccc9e92637fe80f50d5e8b9d503bb715112fe69/CMSIS/RTOS2/RTX/Source/rtx_core_cm.h#L465) by using these instructions.

hathach, 2021-07-04:
> OT: looking onto claim/release functions, they are using mutex with RTOS... Did you consider using [atomic_compare_exchange](https://en.cppreference.com/w/cpp/atomic/atomic_compare_exchange)? It may [...]

There are 2 issue
- atomic is gcc extension, also it kind of requires C11 or higher, not sure but it is technically resolvable (just take time to do it more portable)
- only M3/M4/M7 have mutex instructions that does not requires disable global interrupt. Global interrupts is even worse than using rtos mutex and/or disable USB IRQ.

Anyway, yeah, I am aware of those atomic function, just need time to execute it. For now, we will reply on rtos/ usb irq to provide mutex.

mitchellcairns, 2022-11-07:
Possibly related? https://github.com/espressif/esp-idf/issues/10118

HiFiPhile, 2024-02-01:
Closed by https://github.com/hathach/tinyusb/commit/876f49f6ad10ef2bca000602fe862889c3e0268f

PR #2443 (2024-02-01): Update vendor_device.c fifo mutex config similar to cdc_device.c.
**Describe the PR**
Modified vendor_device.c fifo mutex creation/configuration in the spirit of cdc_device.c.
This hopefully fixes #900.

Works on my machine (device STM32G474-DPOW1). ;)
closed 2024-02-01 by HiFiPhile