hathach/tinyusb #935: audiod_tx_done_cb overflow

HiFiPhile, 2021-06-29:
### Operating System

Windows 10

### Board

Core4357

### Firmware

examples\device\uac2_headset

### What happened ?

Playing with uac2_headset I found there are output samples in the buffer, but mic input is always silence.

Initially there are IN xfer, but as soon as OUT intf is opened, IN xfer stops.
```
00>   AUDIO xfer callback
00>   Queue EP 81 with 0 bytes ... OK
00> USBD Xfer Complete on EP 81 with 0 bytes
00>   AUDIO xfer callback
00>   Queue EP 81 with 0 bytes ... OK
00> USBD Xfer Complete on EP 81 with 0 bytes
00>   AUDIO xfer callback
00>   Queue EP 81 with 0 bytes ... OK
00> USBD Xfer Complete on EP 81 with 0 bytes
00>   AUDIO xfer callback
00>   Queue EP 81 with 0 bytes ... OK
00> USBD Xfer Complete on EP 81 with 0 bytes
00>   AUDIO xfer callback
00>   Queue EP 81 with 0 bytes ... OK
00> USBD Xfer Complete on EP 81 with 0 bytes
00>   AUDIO xfer callback
00> 
00> USBD Setup Received 01 0B 01 00 01 00 00 00 
00>   Set Interface
00>   AUDIO control request
00>   Set itf: 1 - alt: 1
00>   Open EP 01 with Size = 192
00> Set interface 1 alt 1
00>   Queue EP 01 with 192 bytes ... OK
00>   Queue EP 80 with 0 bytes ... OK
00> USBD Xfer Complete on EP 80 with 0 bytes
00> 
00> USBD Setup Received 01 0B 00 00 01 00 00 00 
00>   Set Interface
00>   AUDIO control request
00>   Set itf: 1 - alt: 0
00>   CLOSING Endpoint: 0x01
00>   Queue EP 80 with 0 bytes ... OK
00> USBD Xfer Complete on EP 80 with 0 bytes
00> 
00> USBD Setup Received 01 0B 01 00 01 00 00 00 
00>   Set Interface
00>   AUDIO control request
00>   Set itf: 1 - alt: 1
00>   Open EP 01 with Size = 192
00> Set interface 1 alt 1
00>   Queue EP 01 with 192 bytes ... OK
00>   Queue EP 80 with 0 bytes ... OK
00> USBD Xfer Complete on EP 80 with 0 bytes
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
[... 2607 lines cut ...]
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback
00>   Queue EP 01 with 192 bytes ... OK
00> USBD Xfer Complete on EP 01 with 192 bytes
00>   AUDIO xfer callback

PanRe, 2021-06-29:
I am a bit puzzled how this come to happen, but before digging deep into the code a few questions:
- Did you use the fix #849?
- Did you try disabling the debug messages? In my setup, streaming does not work at all if i enable the debug messages

If nothing helps, we need to debug `audiod_get_AS_interface_index()`...

PanRe, 2021-06-29:
> You have different return in your board ?

I never tested this example on my own. I only adapted the changes required for the new version of the UAC2 driver.

Looking at the example in more detail i see that there are two AS interfaces defined in the audio descriptors, yet `CFG_TUD_AUDIO_FUNC_1_N_AS_INT = 1` is defined within `tusb_config.h`. It should be `CFG_TUD_AUDIO_FUNC_1_N_AS_INT = 2` yielding `alt_setting_1` to be of length two. If interface 1 is opened, `idxItf = 0` should be the result of `audiod_get_AS_interface_index().` Likewise, for interface 2 it should be `idxItf = 1`.

Could you verify the descriptor boarders `audio->p_desc` and `p_desc_end` within `audiod_get_AS_interface_index()`?

HiFiPhile, 2021-06-29:
I think it's a logic problem on `tmp` increment, `tmp` should increase only on alt0 interface descriptor, now it's good :)
```
    uint8_t tmp = 0;
    while (p_desc < p_desc_end)
    {
      // We assume the number of alternate settings is increasing thus we return the index of alternate setting zero!
      if (tu_desc_type(p_desc) == TUSB_DESC_INTERFACE && ((tusb_desc_interface_t const * )p_desc)->bAlternateSetting == 0)
      {
        if (((tusb_desc_interface_t const * )p_desc)->bInterfaceNumber == itf)
        {
          *idxItf = tmp;
          *pp_desc_int = p_desc;
          return true;
        }
        // Increase index, bytes read, and pointer
        tmp++;
      }
      p_desc = tu_desc_next(p_desc);
    }
```

PanRe, 2021-06-29:
Bingo, that should it be! Thanks for fixing this bug... my working example while developing that stuff did not cover this situation! I hope there are not many other possibilities i did not cover :D

Is the example working now?

HiFiPhile, 2021-06-29:
> Is the example working now?

Yes it works now.  

I'll also add `feature_unit_set_request` for `UAC2_ENTITY_CLOCK`, change frequency is quite common.

PanRe, 2021-06-29:
Jep feel free to enhance it! Change of sample resolution from 16/24/32 bit may also be nice but is not mandatory. Some time ago i tried to do this, but i did not manage it to get 24 bits working on Windows. The standard UAC2 Windows driver ignored my 24 bit alternate interface setting for some reason i don't know yet. Have you done this before?

HiFiPhile, 2021-06-29:
I haven't tried 24bit yet.
For EP `wMaxPacketSize` it can be figured out more smartly with maximum sample frequency, for example with xmos:
```
/* Max packet sizes:
 * Samples per channel. e.g (192000+7999)/8000 = 24
 * Must allow 1 sample extra per chan (24 + 1) = 25
 * Multiply by number of channels and bytes      25 * 2 * 4 = 200 bytes
 * TODO Output doesn't get modified by channel count
*/
#define MAX_PACKET_SIZE_MULT_OUT_HS ((((MAX_FREQ+7999)/8000)+1) * NUM_USB_CHAN_OUT)
#define MAX_PACKET_SIZE_MULT_OUT_FS ((((MAX_FREQ_FS+999)/1000)+1) * NUM_USB_CHAN_OUT_FS)

#define HS_STREAM_FORMAT_OUTPUT_1_MAXPACKETSIZE (MAX_PACKET_SIZE_MULT_OUT_HS * HS_STREAM_FORMAT_OUTPUT_1_SUBSLOT_BYTES)
#define HS_STREAM_FORMAT_OUTPUT_2_MAXPACKETSIZE (MAX_PACKET_SIZE_MULT_OUT_HS * HS_STREAM_FORMAT_OUTPUT_2_SUBSLOT_BYTES)
#define HS_STREAM_FORMAT_OUTPUT_3_MAXPACKETSIZE (MAX_PACKET_SIZE_MULT_OUT_HS * HS_STREAM_FORMAT_OUTPUT_3_SUBSLOT_BYTES)

#if (HS_STEAM_FORMAT_OUPUT_1_MAXPACKETSIZE > 1024) || (HS_STEAM_FORMAT_OUPUT_2_MAXPACKETSIZE > 1024) \
    || (HS_STEAM_FORMAT_OUPUT_3_MAXPACKETSIZE > 1024)
#error
#endif

#define FS_STREAM_FORMAT_OUTPUT_1_MAXPACKETSIZE (MAX_PACKET_SIZE_MULT_OUT_FS * FS_STREAM_FORMAT_OUTPUT_1_SUBSLOT_BYTES)
#define FS_STREAM_FORMAT_OUTPUT_2_MAXPACKETSIZE (MAX_PACKET_SIZE_MULT_OUT_FS * FS_STREAM_FORMAT_OUTPUT_2_SUBSLOT_BYTES)
#define FS_STREAM_FORMAT_OUTPUT_3_MAXPACKETSIZE (MAX_PACKET_SIZE_MULT_OUT_FS * FS_STREAM_FORMAT_OUTPUT_3_SUBSLOT_BYTES)
```

PanRe, 2021-06-29:
Mh i tried something like that... i read that the standard windows driver selects the alternate interface whose wMaxPacketSize is just big enough to handle the desired bandwidth. So i had
_(48000 / 1000 + 1) * N_Ch * N_BYTES_PER_SAMPLE_ for _N_BYTES_PER_SAMPLE = 2, 3, 4_
But this did not work so far...
Anyway, thanks for fixing the example!!

HiFiPhile, 2021-06-29:
Maybe it's because `bInterval`, in this example `bInterval` was 4 which should be 1, only the feedback EP should set `bInterval` to 4.

With `bInterval` set to 1 I can set `wMaxPacketSize` like in theory.

I pushed my changes in #938

PanRe, 2021-06-30:
Thanks a lot! A working example is always nice to have! Which program did you use to get the descriptors and the nice parsing of them?

HiFiPhile, 2021-06-30:
> Which program did you use to get the descriptors and the nice parsing of them?

This one : https://www.uwe-sieber.de/usbtreeview_e.html

PanRe, 2021-06-30:
Thanks a lot! :)

HiFiPhile, 2021-07-06:
Closed by #938

closed 2021-07-06 by HiFiPhile