hathach/tinyusb #939: tu_fifo overflow correction error

HiFiPhile, 2021-06-30:
### Operating System

Windows 10

### Board

STM32L0

### Firmware

Reproduce code:
```
tu_fifo_t ft;
uint8_t dummy[512];
uint8_t dummy2[512];
uint16_t n;
void main(void)
{
    tu_fifo_config(&ft, dummy, 512, 1, 1);
    ft.wr_idx = 466;
    ft.rd_idx = 827;
    printf("Count %d, rd_idx %d \r\n", tu_fifo_count(&ft), ft.rd_idx);
    n = tu_fifo_read_n(&ft, dummy2, 200);
    printf("Read %d, count %d rd_idx %d  \r\n", n, tu_fifo_count(&ft), ft.rd_idx);
    n = tu_fifo_read_n(&ft, dummy2, 200);
    printf("Read %d, count %d rd_idx %d  \r\n", n, tu_fifo_count(&ft), ft.rd_idx);
    n = tu_fifo_read_n(&ft, dummy2, 200);
    printf("Read %d, count %d rd_idx %d  \r\n", n, tu_fifo_count(&ft), ft.rd_idx);
}
```

### What happened ?

Playing with cdc, sometimes IN EP goes into crazy mode which sends infinite data to host 🤣 ...

Apparently there is a problem with overflow correction in https://github.com/hathach/tinyusb/blob/81d238bd463c694af6ef31015107c0727805c7c7/src/common/tusb_fifo.c#L405,
my rd_idx went into 64xxx with 512 fifo depth.

### How to reproduce ?

Run reproduce code above, which gives:
```
Count 512, rd_idx 827
Read 200, count 512 rd_idx 64666
Read 200, count 512 rd_idx 64666
Read 200, count 512 rd_idx 64666
```
Basically the fifo become infinite....

HiFiPhile, 2021-06-30:
It's ok ,just need to force cast to `uint16_t` before comparison.
` if ((p < (uint16_t)(p - offset)) || ((uint16_t)(p - offset) > f->max_pointer_idx))`

PanRe, 2021-06-30:
Ah thank you! The code was orginially designed for `uint32_t` (with all the overflows in mind). Could you do me a favor, could you include this case in the `tu_fifo` CI tests?

PanRe, 2021-06-30:
Mh i thought a bit of the problem you described but i don't really see where the problem comes from and why casting to `uint16_t` solves the problem. After the first read, the second condition should hit as
`(p + offset > f->max_pointer_idx)` which is `827 + 200 = 1027 > 1023`
and the read index should become
`rd_idx = mod(827 + 200 + 64512, 16) = 3`,
where `64512` is `non_used_index_space `. Do you have any ideas on this?

HiFiPhile, 2021-06-30:
> After the first read, the second condition should hit as

It failed at 1st read, since the fifo is in overflowed state, in `_tu_fifo_peek` it tries to correct the rd_idx by backward the pointer.

I'm not sure which C language rule apply, but without casting to uint it generate signed compare machine code, since `(p - offset)` is negative the result is always false.

hathach, 2021-06-30:
The GCC++17 have the warning when compiling with this as well, since haven't got time to trace it down, I just put the warning comment there :D. For C : I think any expression involved minus will be treated as signed number ( I guess ).

```
// TODO warning: assuming signed overflow does not occur when assuming that (X + c) < X is always false [-Wstrict-overflow]
```

HiFiPhile, 2021-06-30:
Fixed by #940

closed 2021-06-30 by HiFiPhile

PanRe, 2021-07-01:
> It failed at 1st read, since the fifo is in overflowed state, in `_tu_fifo_peek` it tries to correct the rd_idx by backward the pointer.
>
> I'm not sure which C language rule apply, but without [...]

I see, thanks for fixing!