hathach/tinyusb #966: Documentation

perigoso, 2021-07-16:
What is the current status with the documentation? It seems it has been left on hold while prioritizing other parts of the project, not criticizing ofc, I say this because of things like the domain  www.tinyusb.org, is this the case?

That said I wanted to contribute and do some work on documentation, I think some online pages [like this](https://openinput.readthedocs.io/en/latest/) are best, is it ok If I work on that?
If so, what approach would be best? we can use something like github pages or a service like [readthedocs.io](readthedocs.io) or some other solution like something selfhosted.

hathach, 2021-07-17:
thank you for your willing to contribute, indeed, documentation is lacking behind. since I don't quite have time to do it. For the approach, indeed, I prefer to to readthedocs #6 , the docs will be served via github pages. The domain is mostly for redirecting. Though, I haven't got time to set them up just yet, any PRs are welcome.

perigoso, 2021-07-17:
Great, in a week when I return from vacation I try to allocate some time for this

perigoso, 2021-07-27:
@hathach I started some work on the docs;
here's a very early preview, lots to be done still:
https://perigoso.github.io/tusb_docs_showcase/

I tried to setup the readthedocs pages, but I don't have enough permissions on the repository, I would need admin rights. So you would need to set that up yourself, or give me admin rights temporarily but I understand that comes with trust issues.

perigoso, 2021-07-27:
You might notice the logo looks different, theres no particular reason for the change, except that I wanted an svg logo to embed in the docs and I had to make one from scratch, and I took some creative liberties.

hathach, 2021-07-29:
> You might notice the logo looks different, theres no particular reason for the change, except that I wanted an svg logo to embed in the docs and I had to make one from scratch, and I took some creative liberties.

Ah right, I notice that as well, to be honest, I like the old logo font more, I will try to export an svg using existing desgin file to use with doc page. Give me a bit of time, I am not really good with these image tool.

perigoso, 2021-08-04:
Close after readthedocs is setup

perigoso, 2021-08-04:
@hathach `github-pages` branch is now redundant no?

hathach, 2021-08-04:
@perigoso it currently is only used for hosting the webusb-serial server part. we could move it as part of the new docs site, it is pretty static.

hathach, 2021-08-05:
Update: I am moving the webusb server to its own repo to free up the gh-page, it will be pointed to the example.tinyusb.org while the readthedocs will be pointed to both doc.tinyusb.org and www.tinyusb.org