hathach/tinyusb #973: custom HID

zhangslice, 2021-07-20:
Do Tiny USB have the corresponding demo for custom HID devices

HiFiPhile, 2021-07-20:
There is an example of generic IN/OUT : `examples/device/hid_generic_inout`

It's pretty easy to customize, you can take a look of predefined descriptors in `/src/class/hid/hid_device.h` and replace `TUD_HID_REPORT_DESC_GENERIC_INOUT` with your own descriptor.

```
//--------------------------------------------------------------------+
// HID Report Descriptor
//--------------------------------------------------------------------+

uint8_t const desc_hid_report[] =
{
  TUD_HID_REPORT_DESC_GENERIC_INOUT(CFG_TUD_HID_EP_BUFSIZE)
};

// Invoked when received GET HID REPORT DESCRIPTOR
// Application return pointer to descriptor
// Descriptor contents must exist long enough for transfer to complete
uint8_t const * tud_hid_descriptor_report_cb(uint8_t itf)
{
  (void) itf;
  return desc_hid_report;
}
```

hathach, 2021-07-21:
@zhangslice next time, if you want to ask an question, please post it in the discussion. Issue for code development

closed 2021-07-21 by hathach