hathach/tinyusb #1008: Examples not building for RP2040

Frogieder, 2021-08-08:
### Operating System

Linux

### Board

Raspberry pi pico

### Firmware

All example files

### What happened ?

I tried building the example files according to the getting started guide. I copy-pasted all commands, choosing raspberry_pi_pico as BOARD. The build failed. Everything worked well when I chose feather_nrf52840_express as BOARD.

### How to reproduce ?

Follow the steps in getting started guide and choose raspberry_pi_pico or pico_sdk as BOARD

### Debug Log

**_../examples/device/dfu_** $ make BOARD=raspberry_pi_pico all LOG=3
cmake -S . -B _build/raspberry_pi_pico -DFAMILY=rp2040 -DBOARD=raspberry_pi_pico -DPICO_BUILD_DOCS=0 -DLOG=3
Using PICO_SDK_PATH from environment ('/home/martin/pico/pico-sdk')
PICO_SDK_PATH is /home/martin/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
PICO target board is pico.
Using board configuration from /home/martin/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.9 (found version "3.9.5") found components: Interpreter 
TinyUSB available at /home/martin/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
ELF2UF2 will need to be built
-- Configuring done
CMake Error at CMakeLists.txt:13 (add_executable):
  Cannot find source file:

    /home/martin/tinyusb/hw/bsp/raspberry_pi_pico/board_raspberry_pi_pico.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [/home/martin/tinyusb/hw/bsp/rp2040/family.mk:9: _build/raspberry_pi_pico] Error 1

hathach, 2021-08-09:
which pico-sdk you are running, if it is not latest 1.2.0, please update it.

Frogieder, 2021-08-09:
I'm sorry, I wasn't running the latest SDK. After upgrade, it works. I just didn't see pico SDK 1.2.0 as a requirements, so I didn't bother upgrading.

closed 2021-08-09 by Frogieder