hathach/tinyusb #1018: stm32_fsdev builded by keil with --gnu option - enumeration fails

NikitaKhalyavin, 2021-08-12:
### Operating System

Windows 10

### Board

custom board with stm32f103c8t6

### Firmware

examples/device/cdc_msc, but adopted manually for keil project with --gnu compiler option

### What happened ?

When project is builded by keil armcc compiler, enumeration fails. The reason is a weak attribute in dcd_edpt0_status_complete function definition (dcd.h, string 135). The armcc compiler ignores function implementation in dcd_stm32_fsdev.c, and because of that device address is never actually set.
I dont know, if it's necessary to fix it, but I think that this issue will be useful for other keil users anyway (I spent a week looking for a reason)

### How to reproduce ?

1. Add tinyusb into stm32f103c8t6 keil project with --gnu compiler option
2. Build and run

hathach, 2021-08-13:
Your `How to reproduce` is not reproducible. You need to make sure others can actual reproduce what you are doing. At least
- Attach your keil project, also tell people where to put it in order to compile.
- I don't why keil didn't pick up that weak function. There is several other weak funcs which are seem to be fine. Maybe you should raise this issue in keil forum.

NikitaKhalyavin, 2021-08-13:
Ok, sorry for this. I've created a repo with working project.
How to reproduce:
1. git clone https://github.com/NikitaKhalyavin/tinyusb_stm32f103_test_in_keil
2. cd tinyusb_stm32f103_test_in_keil/
3. git submodule update --init --recursive
4. open usbTest.uvprojx in Keil
5. build and run on bluepill board - this causes enumeration error
6. rebuild with removed weak attribute in function dcd_edpt0_status_complete definition (file tinyusb/src/device/dcd.h) and rebuild - this works correctly
Unfortunately, I have no bluepill board (or any other standart board) right now, I've tested project on my custom board with the same mcu, but different pinout and clocksource. But the version from repo should work correctly on bluepill board.
Sorry for separated stm32bluepill.c file, the only one difference is removing asm("bkpt") from HardFault_Handler, it causes a compilation error in Keil and it's not important.

NikitaKhalyavin, 2021-08-13:
Compiler armcc v5.06 update 1 (build 61)

hathach, 2021-09-14:
just want to say, since I don't use Keil, hopefully some people would know how to fix.

HiFiPhile, 2021-09-14:
I think it's better to contact keil support for this compiler issue.

Based on my experience of IAR & Segger SES they are quite helpful.

XelaRellum, 2023-08-29:
I think the problem here is the different semantics of weak functions in Keil and in the GCC.

I solved the problem by removing TA_ATTR_WEAK from:
void dcd_edpt0_status_complete(uint8_t rhport, tusb_control_request_t const * request);

In usbd_control.c I added an empty, yet weak declaration:
TU_ATTR_WEAK void dcd_edpt0_status_complete(uint8_t rhport, tusb_control_request_t const * request)
{
  (void)rhport;
  (void)request;
}

In "usbd_control_xfer_cb" I removed the check for the existence and just added the call.

This should be done probably for all optional functions.

commit 85807748 (2024-01-12): Merge pull request #2239 from XelaRellum/bugfix_stm32_fsdev_keil

Fixes #1018 the weak dcd_edpt0_status_complete for Keil Compiler
closed 2024-01-12 by hathach