hathach/tinyusb #1022: GD32VF103 issue with MSC

hathach, 2021-08-15:
### Operating System

Linux

### Board

Sipeed Longan Nano

### Firmware

examples/device/cdc_msc

### What happened ?

The cdc_msc example have the cdc echo fine, however, the msc does not, I hook it into analyzer. Look like there is some issue when host read  4k of data. Notice that the data is **corrupted** with 0x55AA repeatedly somehow. Host give up and decide device isn't properly formatted. It is probably buffer/endpoints config/overflow issue etc ..  which is typical for a new port based on existing dcd driver. I will spend sometime this week (or so) to poke at GDVF103 manual to see if I could spot any issue. Hopefully it would be only an minor change.

_Originally posted by @hathach in https://github.com/hathach/tinyusb/issues/959#issuecomment-895378892_

### How to reproduce ?

1. flash the firmware and attached to pc
2. ram disk won't appeared due to corrupted READ10 response.

### Debug Log

Log is attached, though it doesn't show any abnormal, since the issue is caused by the corrupted buffer/payload of READ10.
[1019.txt](https://github.com/hathach/tinyusb/files/6988049/1019.txt)

### Screenshots

[image]

KarlK90, 2021-11-27:
Hi @hathach out of curiosity I ran the MSC demo again with the latest changes to master and...

```
[Sa Nov 27 12:25:14 2021]  sdc:
[Sa Nov 27 12:25:14 2021] sd 9:0:0:0: [sdc] Attached SCSI removable disk
[Sa Nov 27 12:26:27 2021] usb 3-4.3.4: USB disconnect, device number 20
[Sa Nov 27 12:26:28 2021] usb 3-4.3.4: new full-speed USB device number 21 using xhci_hcd
[Sa Nov 27 12:26:28 2021] usb 3-4.3.4: New USB device found, idVendor=cafe, idProduct=4003, bcdDevice= 1.00
[Sa Nov 27 12:26:28 2021] usb 3-4.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sa Nov 27 12:26:28 2021] usb 3-4.3.4: Product: TinyUSB Device
[Sa Nov 27 12:26:28 2021] usb 3-4.3.4: Manufacturer: TinyUSB
[Sa Nov 27 12:26:28 2021] usb 3-4.3.4: SerialNumber: 123456789012
[Sa Nov 27 12:26:28 2021] cdc_acm 3-4.3.4:1.0: ttyACM0: USB ACM device
[Sa Nov 27 12:26:28 2021] usb-storage 3-4.3.4:1.2: USB Mass Storage device detected
[Sa Nov 27 12:26:28 2021] scsi host8: usb-storage 3-4.3.4:1.2
[Sa Nov 27 12:26:28 2021] FAT-fs (sdc): unable to read boot sector to mark fs as dirty
[Sa Nov 27 12:26:29 2021] scsi host8: scsi scan: INQUIRY result too short (5), using 36
[Sa Nov 27 12:26:29 2021] scsi 8:0:0:0: Direct-Access     TinyUSB  Mass Storage     1.0  PQ: 0 ANSI: 2
[Sa Nov 27 12:26:29 2021] sd 8:0:0:0: Attached scsi generic sg2 type 0
[Sa Nov 27 12:26:29 2021] sd 8:0:0:0: [sdc] 16 512-byte logical blocks: (8.19 kB/8.00 KiB)
[Sa Nov 27 12:26:29 2021] sd 8:0:0:0: [sdc] Write Protect is off
[Sa Nov 27 12:26:29 2021] sd 8:0:0:0: [sdc] Mode Sense: 03 00 00 00
[Sa Nov 27 12:26:29 2021] sd 8:0:0:0: [sdc] No Caching mode page found
[Sa Nov 27 12:26:29 2021] sd 8:0:0:0: [sdc] Assuming drive cache: write through
[Sa Nov 27 12:26:29 2021]  sdc:
[Sa Nov 27 12:26:29 2021] sd 8:0:0:0: [sdc] Attached SCSI removable disk
```

[image]

It works now! :tada:

I haven't investigated what commit fixed the issue but I'm super exited about it and will start to port TinyUF2.

KarlK90, 2021-11-27:
This works with the new `dcd_dwc2` and "old" `dcd_synopsys` driver.

hathach, 2021-11-29:
@KarlK90 great to hear that works, actually I did test it out with gd32 when doing new dwc2 driver. The msc drive does show up, so thing is definitely improved. However, there is still issue here and there. It may not work as well as other ports and I haven't got time to look further at this port. Let me know if you see any issue with gd32