hathach/tinyusb #1039: CDC device receives garbage if it transmits first

diggit, 2021-08-21:
### Operating System

Linux

### Board

Custom board with ULPI PHY and STM32H730

### Firmware

cdc_dual example, with these changes:
- with only one CDC enabled
- loopback code disabled
- Instead of looping back data from host back to host, device transmits data periodically since boot
- CDC buffers raised to 4 kiB

```cpp
	tusb_init();
	auto prev = ticks; // ticks are volatile variable incremented every 1 ms (systick).
	std::uint8_t data = 0;
	for(;;) {
		tud_task();

		const auto now = ticks;
		if (now - prev > 200) {
			prev = now;

			tud_cdc_write(&data, 1); // TX one byte
			tud_cdc_write_flush();
			data++; // increment data value for each transmission
			pins::LedStatus.setLevel(data&1); // and flip LED
		}

		if (tud_cdc_available()) {
			// capture all received data
			pins::LedErr.setLevel(true);
			offset += tud_cdc_read(&buffer.at(offset), buffer.size() - offset);
		}
	}
```

### What happened ?

1) power on device and let i live (1s or so, just to have some data in TX buffer)
2) open UART (`ACM0`) in serial terminal (iserterm in ma case)
3) data in TX buffer are received by host, but also some new data ends in RX buffer (ERR led lit up and some garbage is in `buffer`, but I no (explicit) data transmission from host occurred

### How to reproduce ?

Basically replicate "What happened" scenario. I am not sure if this is issue of Linux or tinyUSB stack.

### Debug Log

```
USBD init
CDC init
SystemCoreClock = 48000000
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes ...
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
SystemCoreClock = 48000000
USBD Bus Reset : Full Speed

USBD Setup Received 00 05 69 00 00 00 00 00 
  Set Address
[... 338 lines cut ...]
  Queue EP 82 with 1 bytes ...
USBD Xfer Complete on EP 82 with 1 bytes
  CDC xfer callback
  Queue EP 82 with 1 bytes ...
USBD Xfer Complete on EP 82 with 1 bytes
  CDC xfer callback

USBD Setup Received 21 20 00 00 00 00 07 00 
  CDC control request
  Set Line Coding
  Queue EP 00 with 7 bytes ...
USBD Xfer Complete on EP 00 with 7 bytes
  0000:  80 25 00 00 00 00 08
  CDC control complete
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 21 22 00 00 00 00 00 00 
  CDC control request
  Queue EP 80 with 0 bytes ...
USBD Xfer Complete on EP 80 with 0 bytes
  Set Control Line State: DTR = 0, RTS = 0
  Queue EP 82 with 1 bytes ...
USBD Suspend : Remote Wakeup = 0
```

HiFiPhile, 2021-08-21:
You could try to capture USB traffic with Wireshark.

From the log there are OUT transfers came from PC, or the USB transfer complete interrupt was falsely triggered.

hathach, 2021-08-22:
Please put long log as attached txt file to make issue easier to read

hathach, 2021-08-28:
```C
pins::LedErr.setLevel(true);
```

it is not buildable, please also give the link to your fork that contain the exact code, it must be as minimal changes as possible.

diggit, 2021-09-04:
I've updated code snippet. Now it should be C compatible and also not dependent on the rest of my code.
Just use `examples/device/cdc_dual_ports/src/main.c` example and replace main function block with that snippet.
Adjust `coreFrequency` according to your MCU frequency.

hathach, 2021-09-06:
Please attached your example folder here, that makes sure we are testing the same piece of code.

PR #1789 (2023-01-07): Fix fifo memory overflow
**Describe the PR**

While trying to fix an issue reported in https://github.com/hathach/tinyusb/issues/1729 (detected using fuzzing #1716). I am able to fix a couple of issues with current tusb_fifo when using with overwritable mode
- Prevent double overflowed to happen which make fifo count invalid. Though due to the cost of moving memory, the data at read index is not replaced by newer one.
- fix _tu_fifo_remaining() return incorrect value which cause memory overflowed, detected by fuzzing
- minimize tu_fifo size to 16
- simplify and remove _tu_fifo_empty, _tu_fifo_ [... cut]
closed 2023-01-07 by hathach