hathach/tinyusb #1040: Potential fifo racing on self powered device

HiFiPhile, 2021-08-22:
### Operating System

Windows 10

### Board

Custom NUC126

### Firmware

In loop:
```
   uint8_t data[64];
    if(tud_vendor_write_available() >= sizeof(data))
    {
        tud_vendor_write(data, sizeof(data));
    }     
```

### What happened ?

This could happen on CDC & Vendor class which use fifo when USB is re-plugged.

Take Vendor class as example, fifo is cleared in `vendord_reset` when re-plugged, if fifo is fully written between `vendord_reset` and `vendord_open`, it will enter a deadlock condition that fifo is full but no transfer is scheduled.

Either `maybe_transmit` or `tu_fifo_clear` need to be called in `vendord_open`.

### How to reproduce ?

1. Run a minimum vendor class on self-powered device
2. Write data in a loop
3. Plug & Unplug multiple times
4. Tx fifo is written but no data is transfered

hathach, 2021-08-25:
closed by #1041

closed 2021-08-25 by hathach

HiFiPhile, 2021-08-25:
Do you want to add `tud_cdc_n_write_flush`  in `cdcd_open` ?