hathach/tinyusb #1042: API to get _ctrl_xfer status

zxtxin, 2021-08-23:
I am porting tinyusb to the USB IP 'musbfdrc' from Mentor Graphics .
musbfdrc deals with the control transfer in a slighly different way that it needs CPU to explicitly denote a packet if it is the last packet of the transaction.
[musbfdrc_ep0.pdf](https://github.com/hathach/tinyusb/files/7029829/musbfdrc_ep0.pdf)
Here is a brief introduction about what the CPU needs to do when dealing with control transfer. We can see that during IN DATA PHASE, when we put the last packet into the FIFO, we must set DATAEND bit. 
I could not judge whether a packet of IN DATA PHASE is the last packet only if I can get _ctrl_xfer status. So I would like tinyusb to add relevant APIs.

hathach, 2021-08-23:
> when we put the last packet into the FIFO, we must set DATAEND bit.

you can detect that by using the xfer length of the dcd_edpt_xfer().  Maybe I miss something, would mind illustrate the issue you are having with the pseudo code based, with comment what is your issue, which info is missing. It is much easier to discuss over actual work.

kkitayam, 2021-08-23:
I guess that `total_bytes` is always less than or equal to `max_packet_size` in control transfer. https://github.com/hathach/tinyusb/blob/master/src/device/usbd_control.c#L89

But, if dcd layer can read setup packet, it can check wLength of setup packet and calculate how many interrupts will occur in the current transfer. I think a typical dcd can read setup packet.

kkitayam, 2021-08-28:
I think that when host requests a control read transfer with a multiple of 64(=max packet size) bytes length, it needs to check `wLength`. 

In `Universal Serial Bus Specification, Revision 2.0` 5.5.3, data stages are described as follows.
[image]

My understanding is followings. When `wLength` is greater than the actual data size and is a multiple of 64, USBD layer needs to send ZLP. When `wLength` is equal to the actual data size and is a multiple of 64, USBD layer does not send ZLP.

HiFiPhile, 2021-08-28:
> My understanding is followings. When wLength is greater than the actual data size and is a multiple of 64, USBD layer needs to send ZLP. When wLength is equal to the actual data size and is a multiple of 64, USBD layer does not send ZLP.

You're right.

Sorry I missed out that no ZLP is sent on data stage if length is multiple of 64.
https://github.com/hathach/tinyusb/blob/8d97f0e81f775d8c7646dfc75ee59b7e938bbf8d/src/device/usbd_control.c#L197

hathach, 2021-08-29:
@zxtxin please follow the suggestion from @kkitayam to store the wlength and direction if DATA in, or you could keep an copy of 8-byte setup packet. It totally depends on your port. It is too specific with your port to have an dedicated API for it.

zxtxin, 2021-08-30:
> @zxtxin please follow the suggestion from @kkitayam to store the wlength and direction if DATA in, or you could keep an copy of 8-byte setup packet. It totally depends on your port. It is too specific with your port to have an dedicated API for it.

OK. I will follow this.

closed 2021-08-30 by zxtxin