hathach/tinyusb #1048: tud_ready returning true when not connected to mass storage host

jcummings2, 2021-08-24:
### Operating System

Windows 10

### Board

Feather M0 with added flash and battery including switch

### Firmware

```C++
/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This example demo how to expose on-board external Flash as USB Mass Storage.
 * Following library is required
 *   - Adafruit_SPIFlash https://github.com/adafruit/Adafruit_SPIFlash
 *   - SdFat https://github.com/adafruit/SdFat
 *
 * Note: Adafruit fork of SdFat enabled ENABLE_EXTENDED_TRANSFER_CLASS and FAT12_SUPPORT
 * in SdFatConfig.h, which is needed to run SdFat on external flash. You can use original
 * SdFat library and manually change those macros
 */

#include "SPI.h"
#include "SdFat.h"
#include "Adafruit_SPIFlash.h"
#include "Adafruit_TinyUSB.h"

// Uncomment to run example with FRAM
#define FRAM_CS   5
#define FRAM_SPI  SPI

#if defined(FRAM_CS) && defined(FRAM_SPI)
  Adafruit_FlashTransport_SPI flashTransport(FRAM_CS, FRAM_SPI);

#elif defined(ARDUINO_ARCH_ESP32)
  // ESP32 use same flash device that store code.
  // Therefore there is no need to specify the SPI and SS
  Adafruit_FlashTransport_ESP32 flashTransport;

#else
  // On-board external flash (QSPI or SPI) macros should already
  // defined in your board variant if supported
  // - EXTERNAL_FLASH_USE_QSPI
  // - EXTERNAL_FLASH_USE_CS/EXTERNAL_FLASH_USE_SPI
  #if defined(EXTERNAL_FLASH_USE_QSPI)
    Adafruit_FlashTransport_QSPI flashTransport;

  #elif defined(EXTERNAL_FLASH_USE_SPI)
    Adafruit_FlashTransport_SPI flashTransport(EXTERNAL_FLASH_USE_CS, EXTERNAL_FLASH_USE_SPI);

  #else
    #error No QSPI/SPI flash are defined on your board variant.h !
  #endif
#endif

Adafruit_SPIFlash flash(&flashTransport);

// file system object from SdFat
FatFileSystem fatfs;

[... 159 lines cut ...]
* ready - just a prefix used on all files

Notice that I ran the `computer_none_brick_computer` scenario 2 times (file names ending in 0 and 1). It is the only case that seemed to behave correctly. However, the `brick_none_computer_brick` case did not yield the desired results (I ran that one twice as well just to confirm).

[ready_chargerSamsung.txt](https://github.com/hathach/tinyusb/files/7041718/ready_chargerSamsung.txt)
[ready_chargerSamsung_none_computer_chargerSamsung.txt](https://github.com/hathach/tinyusb/files/7041719/ready_chargerSamsung_none_computer_chargerSamsung.txt)
[ready_computer.txt](https://github.com/hathach/tinyusb/files/7041720/ready_computer.txt)
[ready_computer_none_brick_computer_0.txt](https://github.com/hathach/tinyusb/files/7041721/ready_computer_none_brick_computer_0.txt)
[ready_computer_none_brick_computer_1.txt](https://github.com/hathach/tinyusb/files/7041722/ready_computer_none_brick_computer_1.txt)
[ready_computer_none_brick_computer_2.txt](https://github.com/hathach/tinyusb/files/7041723/ready_computer_none_brick_computer_2.txt)
[ready_computer_none_chargerAmazon_computer.txt](https://github.com/hathach/tinyusb/files/7041724/ready_computer_none_chargerAmazon_computer.txt)
[ready_computer_none_chargerSamsung_computer.txt](https://github.com/hathach/tinyusb/files/7041725/ready_computer_none_chargerSamsung_computer.txt)
[ready_brick.txt](https://github.com/hathach/tinyusb/files/7041726/ready_brick.txt)
[ready_brick_none_computer_brick_0.txt](https://github.com/hathach/tinyusb/files/7041727/ready_brick_none_computer_brick_0.txt)
[ready_brick_none_computer_brick_1.txt](https://github.com/hathach/tinyusb/files/7041728/ready_brick_none_computer_brick_1.txt)
[ready_chargerAmazon.txt](https://github.com/hathach/tinyusb/files/7041729/ready_chargerAmazon.txt)
[ready_chargerAmazon_none_computer_chargerAmazon.txt](https://github.com/hathach/tinyusb/files/7041730/ready_chargerAmazon_none_computer_chargerAmazon.txt)

### Debug Log

Didn't capture but can re-run the pertinent scenario(s) if needed.

### Screenshots

N/A

hathach, 2021-08-25:
please post your sketch as attached file for readability. Also I cannot reproduce your issue, since I don't have the MX25L25673G device. Please try to reproduce it with one of the Adafruit board without any hardware modification.

jcummings2, 2021-08-25:
[msc_external_flash_bug.ino](https://github.com/hathach/tinyusb/files/7046713/msc_external_flash_bug.ino.txt)

hathach, 2021-08-25:
> [msc_external_flash_bug.ino](https://github.com/hathach/tinyusb/files/7046713/msc_external_flash_bug.ino.txt)

I have updated 1st post to make it easier to follow. but I still cannot reproduce your issue due to your sketch using additional flash device that I don't have. Can you try to reproduce this on one of adafruit M0 board that has on-board flash and update the sketch accordingly.