hathach/tinyusb #1059: USB Compliance Verification Test Suite (USBCV)

hathach, 2021-08-30:
**What is this issue for**

The test suite include many corner cases which require both update of usbd and dcd to pass. This issue is used to keep track of  current status of compliance test of tinyusb **device stack** on each DCD ports. Once it is all passed, we could close this.

**What is USB Compliance Verification (USBCV)**

USBCV is test suite tool used by usb-if to verify all usb hardware/software/controller which is required to passed in order to have USB logo on the product. https://www.usb.org/compliancetools 

**How to run USBCV**

usbcv tool only runs on windows , depending on usb controller type on your PC you will need to either download the USB3CV (xHCI) or USB2CV  (EHCI) from https://www.usb.org/compliancetools . Note: even your mcu is only full speed, but testing on xHCI host, you would need to run USB3CV.

**Which tests suite to run**

Within the scope of tinyusb device stack, we only need to run follow test suite
- Chapter 9 Test [USB 2 devices]
- HID Tests
- MSC Tests

[image]

**Compliance Status**

Example/Application can have large impact on the test result, the follow result is tested mostly for **cdc_msc** and **hid_composite**

| MCU \ Test Suite | Chapter9 | HID  | MSC  |
| :--------------- | :------: | :--: | :--: |
| nRF52840         |    :heavy_check_mark:    |  :heavy_check_mark:  |  :heavy_check_mark:  |
| SAMD21, SAMD51   |    :question:     |  :question:   |  :question:   |
| RP2040           |   :question:    |  :question:  |  :question:   |

HiFiPhile, 2021-08-30:
I'm trying to test my project based on STM32L071.

SetConfiguration passed even without #1058
```
TD 9.13 - SetConfiguration Test (Configuration Index 0)Passed
INFO
Start time: Aug 30, 2021 - 14:23:20
INFO
SetConfiguration with configuration value : 0x1
INFO
Unconfigured the device
INFO
SetConfiguration with configuration value : 0x1
INFO

Stop time: Aug 30, 2021 - 14:23:21
INFO
Duration:  1 second.
INFO
Stopping Test [ TD 9.13 - SetConfiguration Test (Configuration Index 0):
     Number of: Fails (0); Aborts (0); Warnings (0) ]

TD 9.30 Configuration Summary Descriptor Test (Configuration Index 0x00) - Device State ConfiguredPassed
INFO
Start time: Aug 30, 2021 - 14:23:21
INFO
Configuration Index 0x00 has a Configuration Value of 1.
INFO
Initialized the device to the Configured state with Configuration Value 1.
INFO
Retrieved Device Descriptor.
INFO
Device Under Test supports USB Version 0x2.0x10.
INFO
Retrieved BOS Descriptors.
INFO
No Configuration Summary Descriptors found.
INFO

Stop time: Aug 30, 2021 - 14:23:22
INFO
Duration:  1 second.
INFO
Stopping Test [ TD 9.30 Configuration Summary Descriptor Test (Configuration Index 0x00) - Device State Configured:
     Number of: Fails (0); Aborts (0); Warnings (0) ]

TD 9.14 - Suspend/Resume Test (Configuration Index 0)Passed
INFO
Start time: Aug 30, 2021 - 14:23:22
INFO
Suspended the parent port of the Device Under Test.
INFO
Resumed the parent port of the Device Under Test.
INFO
Got device descriptor of Device Under Test
INFO

Stop time: Aug 30, 2021 - 14:23:23
INFO
Duration:  1 second.
INFO
Stopping Test [ TD 9.14 - Suspend/Resume Test (Configuration Index 0):
     Number of: Fails (0); Aborts (0); Warnings (0) ]
```

Then it fails at `L1Suspend/Resume Test (Configuration Index 0)`
```
INFO
Start time: Aug 30, 2021 - 14:23:25
INFO
Checking Device Under Test for LPM L1 Compatibility...
INFO
USB version of device is 2.10.
INFO
DUT IS compatible with LPM.
INFO
LPM IS required for DUT
FAIL
(USB: 9.6.2.1.9) A USB device with a USB version greater than or equal to 0x210 must include a USB 2.0 Extension descriptor.
INFO

Stop time: Aug 30, 2021 - 14:23:26
INFO
Duration:  1 second.
INFO
Stopping Test [ L1Suspend/Resume Test (Configuration Index 0):
     Number of: Fails (1); Aborts (0); Warnings (0) ]
```

I set `bcdUSB` to 0x0210 to be driver-free like WebUSB. However I looked USB 2.0 and 3.2 specifications but can find this `9.6.2.1.9` chapter.  Do you have idea where it is ?

hathach, 2021-08-30:
@HiFiPhile USB 2.0 extension section 9.6.2.1 is part of USB 3.2 revision 1.0 specs

HiFiPhile, 2021-09-09:
NUC126 passed chapter 9 tests with custom vendor class.
[Chapter 9 Tests - USB 2 - Passed - 2021-09-09 20-35-38.zip](https://github.com/hathach/tinyusb/files/7138985/Chapter.9.Tests.-.USB.2.-.Passed.-.2021-09-09.20-35-38.zip)

hathach, 2021-09-10:
> NUC126 passed chapter 9 tests with custom vendor class.
> [Chapter 9 Tests - USB 2 - Passed - 2021-09-09 20-35-38.zip](https://github.com/hathach/tinyusb/files/7138985/Chapter.9.Tests.-.USB.2.-.Passed.-.2021-09-09.20-35-38.zip)

Thank for the testing, though, did you test with the hid composite, one of the chapter9 is remote wakeup which is only tested with hid example due to `TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP`

HiFiPhile, 2021-09-10:
> Thank for the testing, though, did you test with the hid composite, one of the chapter9 is remote wakeup which is only tested with hid example due to TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP

It didn't work.
There is a typo of `|=` but still something more.
https://github.com/hathach/tinyusb/blob/d66d817c38d3da25198dbef90e8e0cd516f4508e/src/portable/nuvoton/nuc121/dcd_nuc121.c#L258

HID test passed.
[USB3CV.zip](https://github.com/hathach/tinyusb/files/7142653/USB3CV.zip)