hathach/tinyusb #1066: assert() failure in hcd_rp2040.c

csBlueChip, 2021-09-02:
### Operating System

RaspberryPi OS

### Board

RP2040

### Firmware

examples/host/hid_controller

### What happened ?

`assertion "ep->active" failed: file "/path/to/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 164, function: hw_trans_complete`

Workaround: comment out the assert [dirty]

### How to reproduce ?

* `mkdir ~/tmp ; cd ~/tmp`
* `cmake -DFAMILY=rp2040 -DCMAKE_BUILD_TYPE=Debug /path/to/tinyusb/examples/host/hid_controller/`
* `make`
* Attach `minicom` (or such) to the Pico UART (FTDI/PicoProbe/etc.)
* Connect a keyboard to the RP2040 USB port

This (?obviously?) does not happen with debug DISabled.

### Debug Log

TinyUSB Host HID Controller Example
USBH init
HID init
hcd_init 0
USBH DEVICE ATTACH
hcd_port_connect_status
Open EP0 with Size = 8 (addr = 0)
hcd_edpt_open dev_addr 0, ep_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Get 8 byte of Device Descriptor
Control Setup (addr = 0): 80 06 00 01 00 00 08 00
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Transfer complete
Sent setup packet
on EP 00 with 0 bytes
hcd_edpt_xfer dev_addr 0, ep_addr 0x80, len 8
hw_endpoint_init dev 0 ep 0 in xfer 0
dev 0 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Prepare Buffer Control:
  Buffer Control 1 0x0: len = 0, available = 0, full = 0, last = 0, stall = 0, reset = 0, toggle = 0
  Buffer Control 0 0x7408: len = 8, available = 1, full = 0, last = 1, stall = 0, reset = 1, toggle = 1
Buffer complete
buf_status 0x00000001
Single Buffered: ep_ctrl = A0000180
_hw_endpoint_xfer_sync:
  Buffer Control 1 0x0: len = 0, available = 0, full = 0, last = 0, stall = 0, reset = 0, toggle = 0
  Buffer Control 0 0xe008: len = 8, available = 0, full = 1, last = 1, stall = 0, reset = 0, toggle = 1
Completed transfer of 8 bytes on ep 0 in
Transfer complete
assertion "ep->active" failed: file "/path/to/tinyusb/src/portable/raspberrypi/rp2040/hcd_rp2040.c", line 164, function: hw_trans_complete

hathach, 2021-09-03:
> This (?obviously?) does not happen with debug DISabled.
what do you mean by this ? Can you confirm if the issue occur when debug is disabled ??

BTW, hid controller is for mounting game controller only, and currently only tested with Sony PS4. If you want to test keyboard please use the cdc_msc_hid example.

csBlueChip, 2021-09-03:
> > This (?obviously?) does not happen with debug DISabled.
>
>what do you mean by this ? Can you confirm if the issue occur when debug is disabled ??

If you do NOT use `-DCMAKE_BUILD_TYPE=Debug` then the HID code (seemingly) works well... You can connect/disconnect a keyboard and the PID/VID are shown as expected ...Enable debug, and the driver just crashes :/

> BTW, hid controller is for mounting game controller only, and currently only tested with Sony PS4. If you want to test keyboard please use the cdc_msc_hid example.

I do not care to make a keyboard work, and I do not own a PS4. I am trying to write two new class host drivers, and was simply reversing the HID code as a working-example ...all help gratefully appreciated ;)

hathach, 2021-09-05:
it is not an bug, it is due to print log in the ISR with slow channel like uart will mess up the usb timing. You could try to reduce the debug level CFG_TUSB_DEBUG to 2 or 1 instead of 3.

closed 2021-09-05 by hathach

BabumonGopi, 2022-04-03:
HID is working fine in my case. MSC is giving the same error in my case, even after building with "release". Please help how to solve it.

pm-mic, 2022-05-31:
https://github.com/hathach/tinyusb/issues/1296#issuecomment-1018895297