hathach/tinyusb #1067: /examples/host/hid_controller is not compiled

csBlueChip, 2021-09-02:
### Operating System

Linux

### Board

RP2040

### Firmware

/examples/host/hid_controller

### What happened ?

hid_controller does not build ...because `/examples/host/CMakeLists.txt` lacks the line `family_add_subdirectory(hid_controller)`

### How to reproduce ?

* `mkdir ~/tmp ; cd ~/tmp`
* `cmake -DFAMILY=rp2040 /path/to/tinyusb/examples`
...note that hid_controller is not built

### Debug Log

n/a

hathach, 2021-09-03:
since you already know how to fix it, would you mind making an PR to fix this ?

csBlueChip, 2021-09-03:
I am trying to write two new class host drivers (by reverse engineering the existing driver code) ...When I have done that, I will need to learn how Git PRs work ...So my answer is: Yes, but give me time ;)

PR #1120 (2021-10-01): house keeping update
**Describe the PR**
Mostly house keeping work
- Update ISO endpoint for nrf5x to be 8 for audio example
- fix #1067 
- correct drivers interface binding for MIDI and BTH, where it have 2 interfaces but does not use IAD.
- clean up usbh configuration parser a bit  
- Make vendor driver more flexible with varies of custom descriptors between Interfaces and Endpoints. Also can have 0-2 endpoints (previously fixed 2 endpoints).
closed 2021-10-01 by hathach