hathach/tinyusb #1079: Add MTP class device support

zhangshengping, 2021-09-08:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
some time mcu need to get file from sd card and also usb host need get file from sd card.
MTP is for this solution.

pwsoftwaredesigns, 2024-09-26:
I was trudging through the ST's implementation of MTP in the latest release of their USB library (the only USB stack I could find that has support for MTP).  However, it's poorly-written, difficult to understand, and requires that you do everything in an ISR.

So, I'm feeling (foolishly) optimistic that I might be able to implement the MTP class in TinyUSB (even though I am just learning how the USB stack works).  I'm reading through the MTP spec and ST's implementation, and making use of Wireshark to see if it's possible.

Though I'd much prefer someone who is actually an expert in TinyUSB and the USB stack in general to implement this.

Blenderpics, 2025-01-26:
@pwsoftwaredesigns any update on this? I am currently emulating a fat32 block device. It would be so much easier if I could use mtp for my usecase

WMXZ-EU, 2025-04-27:
I second the request for MTP device class. It turned out to very useful for Teensy based data acquisition systems.

f-hoepfinger-hr-agrartechnik, 2025-04-28:
i would give a Plus too.

roundby, 2025-07-04:
This is just to inform that we have submitted a simple implementation within PR #3023. You can find the details and test it from there.

hathach, 2026-11-06:
Fixed by 283b06bb5, which added the MTP device class (`src/class/mtp`) with examples.

closed 2026-11-06 by hathach