hathach/tinyusb #1095: setting keyboard led error

qingliyu, 2021-09-19:
### Operating System

Windows 7

### Board

nrf52840

### Firmware

\tinyusb\examples\device\hid_boot_interface\src\main.c

### What happened ?

Execution fails forever on \tinyusb\examples\device\hid_boot_interface\src\main.c：line 216.
report_type is always equal to 0

### How to reproduce ?

empty

hathach, 2021-09-22:
could you provide 
- how you test and steps to reproduce the issue
- the debug log with `LOG=3` around the time the incident occur.

qingliyu, 2021-10-10:
@hathach

hathach, 2021-10-14:
@qingliyu I see you didn't use the hid_boot_interface as it is, but a custom firmware. It is what confuse me, please make sure you specify it exactly as it is in the bug report. The reason for  HID_REPORT_TYPE_INVALID, is lacking of hid parser as mentioned above. For now when encounter this, you need to process the data as how you specify it in the report descriptor i.e read the first byte as report ID if you use report ID for output report. I will try to implement an simple hid parser later on to resolve this.

```
it is what it means at current state of the device stack when receiving report via OUT interrupt endpoint. device stack doesn't attempt to parse the report descriptor, and have no idea if report ID is present or not. Therefore, it will present the SET report as it is with report id as 1st byte. When receiving report via control endpoints, the report id and type is explicitly specified by host in the setup packet and can be presented to the app. I will do more testing with OUT interrupt endpoint later on and see if I could fit a simple parser to detect report id to device stack later on.
```

qingliyu, 2021-10-15:
＠

qingliyu, 2021-10-15:
@hathach I seem to understand what you mean．Using TUD_HID_DESCRIPTOR, the ｐｃ will use the control endpoint to transmit. Using TUD_HID_INOUT_DESCRIPTOR, the ｐｃ will use the ＯＵＴ interrupt　endpoint for transmission．

closed 2021-10-15 by qingliyu