hathach/tinyusb #1097: Update Code for Class Vendor_Host

jmark1m, 2021-09-22:
The type definitions for pipe_handle_t were removed in #449 for all class drivers except for class Vendor_Host (/class/vendor_host.h and /class/vendor_host.c).

The Class Vendor_Device also does not use the type pipe_handle_t so it seems Vendor_Host should follow a similar pattern.

I want to simply define,

`#define CFG_TUH_VENDOR 1` 

and then simply be able to use these functions for bulk data transfer.

`
tusb_error_t tusbh_custom_read(uint8_t dev_addr, uint16_t vendor_id, uint16_t product_id, void * p_buffer, uint16_t length);
`
`
tusb_error_t tusbh_custom_write(uint8_t dev_addr, uint16_t vendor_id, uint16_t product_id, void const * p_data, uint16_t length);
`

I can't compile currently because
`
/class/vendor/vendor_host.h:37:3: error: pipe_handle_t does not name a type (pipe_handle_t pipe_in)
`

hathach, 2021-09-22:
yes, vendor host is lacking behind the migration/upgrade and currently I don't have plan/time to work on it.

Chiminakust, 2022-01-02:
What needs to be done in order to migrate/upgrade this? I want to communicate with a ft232r with the rp2040's USB.
I want to try to contribute, but I have no idea what the needed changes are.

Docfch, 2024-02-29:
@jmark1m did you manage to get the vendor specific host to work?
I also want to install bidirectional communication. Do you have an example of implementation or can you help me do it?

IngHK, 2024-02-29:
> What needs to be done in order to migrate/upgrade this? I want to communicate with a ft232r with the rp2040's USB. I want to try to contribute, but I have no idea what the needed changes are.

See #2488.
Any kind of feedback, review and test is welcome.

jmark1m, 2024-02-29:
I guess I did manage to get usb host to work just using the tinyusb submodule that comes with the pico 2040 sdk.

For my project it was known what devices were going to be plugged in to it so I assigned them fixed end points and didn't go through all of the things a host should do to assign endpoints.  

But eventually I just sent data from my host to the device with this simple function.

Good luck.

```
bool hid_send( board::host_command_t command )
{
  tuh_xfer_t xfer =
  {
    .daddr       = btdev_addr,
    .ep_addr     = 1,
    .buflen      = 64,
    .buffer      = &command.Command,
    .complete_cb = hid_xfer_complete
  };
  return tuh_edpt_xfer(&xfer);
}

```

Docfch, 2024-03-01:
@jmark1m thanks for your reply.
Can i see the full code of your project ?

maaamcube, 2024-10-29:
No news since 2021 ?