hathach/tinyusb #1105: MSP430 failed to response Get Device Descriptor

hathach, 2021-09-25:
### Operating System

Linux

### Board

msp_exp430f5529lp

### Firmware

examples/device/cdc_msc

### What happened ?

Failed to enumerate on both Linux and Windows. The root cause is msp430 responses with incorrect data, basically using the control request instead of the device descriptor payload.

### How to reproduce ?

1. Run cdc_msc
2. PC will complain with read device descriptor error
```
[89841.141609] usb 3-2.1: new full-speed USB device number 45 using xhci_hcd
[89846.329389] usb 3-2.1: device descriptor read/64, error 8
```

### Debug Log

After 3 attempts, host issue bus reset and it loops over

```
USBD init
CDC init
MSC init
USBD Bus Reset : Full Speed

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 8 bytes ...
USBD Xfer Complete on EP 80 with 8 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 8 bytes ...
USBD Xfer Complete on EP 80 with 8 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 40 00 
  Get Descriptor Device
  Queue EP 80 with 8 bytes ...
USBD Xfer Complete on EP 80 with 8 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset : Full Speed
```

### Screenshots

screenshot where 8 bytes DATA is same with setup request (with wlength changed from 64 to 8 bytes).
[image]

hathach, 2021-09-25:
It is probably buffer overflow and/or mis-management somewhere. Could try to do some checkout and bisect to narrow down the changes.

cr1901, 2021-09-25:
@hathach Would you be willing to handle this for me? If not, I'll take a look in the coming days.

Is that screenshot first transaction that happens after bus reset (i.e. beginning of enumeration)? If so, that looks somewhat  (see italics) normal to me.

MSP430 is hardcoded for control endpoint to xfer 8 bytes at a time. I know the first xfer an OS will send should be 64 bytes, and the OS should notice that the MSP430 only responded w/ 8 bytes. _MSP430 responding by echoing the setup packet is incorrect_, but I don't understand why _that_ particular transaction should fail. I would expect the OS to ignore the device payload, and then set the EP0 packet size appropriately.

I would then expect subsequent SETUPs to fail because of what you describe:

>The root cause is msp430 responses with incorrect data, basically using the control request instead of the device descriptor payload.

hathach, 2021-09-25:
@cr1901 no worries, I will try to fix this whenever I got time (currently I am focusing on esp32 work). This issue is discovered when testing #1102 (for ref https://github.com/hathach/tinyusb/pull/1096#issuecomment-926547381) , and is reminder for me since I have a bad memory :)

dcd msp430 isn’t modified for a while. It is probably recent changes since it work flawlessly previously.

maemoKKK, 2021-10-03:
I have similar issue.
I use STM32F103 BLUEPILL board, Win10-PC-Host send usb packet （ 80 06 00 01 00 00 12 00 ）means descriptor length is 18 bytes. in tiny usb code, the descriptor is prepared by desc_device, and the length is 18bytes exactly.
However, I use another MCU IC manufacture by Chinese facturary ( they said it is as the same as the STM32F103C8T6). after power on, usb enumeration ，Win10-PC-Host send usb packet （ 80 06 00 01 00 00 40 00 ）. my board still response 18bytes.

then issue happen!