hathach/tinyusb #1119: XCore Support

xmos-jmccarthy, 2021-09-29:
**Is your feature request related to a problem? Please describe.**
XCore supports TinyUSB.  We have support that exists in our [sdk](https://github.com/xmos/xcore_sdk), built on top of the [FreeRTOS kernel modified for SMP](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/35b95d2a4debe36f38e0c8768a5260dd586b7d2c), that is now being maintained by Amazon.  Between the SMP kernel and some build requirements unique to our platform, we opted to test the example applications within our SDK.

**Describe the solution you'd like**
I want to get XCore support into the supported devices, but do not want to bloat tinyusb with our build requirements and full sdk.

Is it acceptable to add the platform and hardware to the [supported](https://github.com/hathach/tinyusb/blob/master/docs/reference/supported.rst) page and top level readme; And then add a tinyusb/hw/mcu/xcoreai/readme.txt that just contains a link to our sdk?  I could add the sdk as a submodule, but this will add a lot of other submodules to your repo in the recursive clone case.

hathach, 2021-09-30:
Hi @xmos-jmccarthy ,

> [sdk](https://github.com/xmos/xcore_sdk), built on top of the [FreeRTOS kernel modified for SMP](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/35b95d2a4debe36f38e0c8768a5260dd586b7d2c), that is [...]

Yeah please add your platform/mcu to the supported.rst and your specific board as well. People often forget to do this, and I need to manually add it later on (will make an PR template for this later on).

Not including as submodule is a great approach for complicated sdk. actually there are 2 ports that are using this approach already. The esp32s2 and rp2040 . Both required user to clone/install their sdk IDF & pico-sdk respectively since they have a sophisticated building system. Most other mcu doesn't since their sdk is mostly for low level driver and register defs.

SInce it is not included as submodule, you need to manually checkout it in the ci like pico-sdk here https://github.com/hathach/tinyusb/blob/master/.github/workflows/build.yml#L69

xmos-jmccarthy, 2021-10-04:
Thanks.

Closing for now while I wait for some things to happen on my end.  Will reopen and PR when ready.

closed 2021-10-04 by xmos-jmccarthy

hathach, 2021-10-07:
looking forward to your PR, if you are unsure, just make it as a draft, I will try to help with