hathach/tinyusb #1129: tud_hid_set_protocol_cb() is not called when SET_PROTOCOL is sent

dhalbert, 2021-10-07:
### Operating System

Others

### Board

Adafruit MacroPad RP2040

### Firmware

circuitpython (post 7.0.0, PR under development) using TinyUSB commit 43aac707

I'm testing this in the BIOS screen on a Dell Latitude E4310, which requires a boot keyboard,, and sends a SET_PROTOCOL. Interestingly, some later Dell BIOS'es no longer require a boot keyboard. I also see this problem on an older MSI motherboard.

### What happened ?

I'm implementing boot protocol support in CircuitPython. I set up an HID interface descriptor specifying boot protocol, using the keyboard (device 1), and a keyboard HID report descriptor with no report ID (for failover).

CircuitPython never sees a call to its `tud_hid_set_protocol_cb()`, to confirm that the host wants to switch to boot protocol. I instrumented the calling routine.

I also instrumented `hidd_control_xfer_cb()`, which is supposed to call `tud_hid_set_protocol_cb()`. This routine is never entered, which seems odd. I do see `hidd_control_xfer_cb()` entered when the same setup is just plugged into Linux, when that the host doesn't ask for boot protocol. So I think my instrumentation is OK.

(The instrumentation consists of setting a float global to various values so I can see which pieces of code were entered.  print the value out at various places in my CircuitPython test program, and can see the printouts on the MacroPad display. I. I am hijacking `microcontroller.cpu.temperature` to get the value: it returns the global instead of the temperature.)

### How to reproduce ?

The setup of this is complicated, but I can help you get it set up if necessary. I think this should be testable with your `examples/hid_boot_interface`. Right now that example does nothing in the callback, but it could be changed to do something (or you could set a breakpoint there).

The trick is to find an old-enough machine whose BIOS wants a boot keyboard.

### Debug Log

Here is a Beagle .tdc trace of the USB traffic. I wait five seconds at the beginning of the program to allow plenty of
time for enumeration:
[macropad1.tdc.zip](https://github.com/hathach/tinyusb/files/7297868/macropad1.tdc.zip)

For comparison, here is a conventional Dell keyboard responding to the SET_PROTOCOL request:
[dell1.tdc.zip](https://github.com/hathach/tinyusb/files/7297870/dell1.tdc.zip)

### Screenshots

Macropad: I think the "Set Configuration" is the start of the SET_PROTOCOL transaction, and the `21 0B` is the actual SET_PROTOCOL message. Note the repeated Control Transfer stalls after it.
[image]

In comparison, here's the Dell example, which seems to work smoothly:
[image]

hathach, 2021-10-07:
thanks @dhalbert for a detail issue, I will do my own test. meanwhile can you provide following detail, to see if I could spot any isue in the configuration.
1. the branch/code for cpy you are working on along with code.py for reproducing purpose.
2. your configuration descriptor via command `lsusb -v -d 239a:`
3. hid descriptor via command `sudo usbhid-dump -d 239a | grep -v : | xxd -r -p | hidrd-convert -o spec`

dhalbert, 2021-10-07:
Here's a patch file of the instrumentation of `src/class/hid/hid_device.c`. I just add an `extern float indicator`, and then set it to various values at various points in the code:
[instrumentation.hid_device.c.patch.txt](https://github.com/hathach/tinyusb/files/7303213/instrumentation.hid_device.c.patch.txt)

CircuitPython changes are here:
https://github.com/dhalbert/circuitpython/tree/hid-boot-protocol. There are two commits on this branch.

boot.py and code.py (renamed for uploading):
[boot.py.txt](https://github.com/hathach/tinyusb/files/7303255/boot.py.txt)
[code.py.txt](https://github.com/hathach/tinyusb/files/7303262/code.py.txt)
`code.py` implements just two keys: up arrow and down arrow, which is sufficient for moving around in the BIOS screen.

`lsusb -v -d 239a:`
[lsusb.txt](https://github.com/hathach/tinyusb/files/7303275/lsusb.txt)

 `hidrd-convert` is not easily available for Ubuntu 20.04 (it was in 18.04), but here is the descriptor:
```
001:019:003:DESCRIPTOR         1633610483.029616
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 03 75 01
 05 08 19 01 29 05 91 02 95 01 75 05 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 C0
```
passed through http://eleccelerator.com/usbdescreqparser/. This should be the same as what is in `boot.py`.
```c
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x03,        //   Report Count (3)
0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x01,        //   Report Count (1)
0x75, 0x05,        //   Report Size (5)
0x91, 0x01,        //   Output (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
```

hathach, 2021-10-08:
@dhalbert  Changing the bcd device number when changing driver combo is a good way to keep the same vid/pid. Though you still need to do a further research. Since my info could be obsolete.

Anyway, I think you should just go ahead with it, just to try to see we could quickly get the boot interface working first. Then we could decide how we walkaround this later on.

hathach, 2021-10-09:
- If you want to load Usbser.sys automatically, set the class code to 02 and subclass code to 02 in the Device Descriptor. For more information, see USB communications device class. With this approach, you are not required to distribute INF files for your device because the system uses Usbser.inf.
  - If your device specifies class code 02 but a subclass code value other than 02, Usbser.sys does not load automatically. Pnp Manager tries to find a driver. If a suitable driver is not found, the device might not have a driver loaded. In this case, you might have to load your own driver or write an INF that references another in-box driver.
  - If your device specifies class and subclass codes to 02, and you want to load another driver instead of Usbser.sys, you have to write an INF that specifies the hardware ID of the device and the driver to install. For examples, look through the INF files included with sample drivers and find devices similar to your device. For information about INF sections, see Overview of INF Files.

@dhalbert it requires the CDC class & subclass code to be set in the Device Descriptor, just make sure you test with configuration without CDC (the device descriptor should unset the class code). However, this does not seem to work with win8 and prior

[image]

dhalbert, 2021-10-11:
I tried a device id like:
```
"USB Serial"=DriverInstall, USB\Class_02&SubClass_02
```
This does not work because a third-party driver cannot be a class driver: it cannot recognize a device solely by its class (its "Compatible ID"). The  `.cat` file creator complains and throws an error on the above. Instead, a device must be recognized by its complete hardware ID, e.g. `USB\VID_0239a&PID_1234`. It's not possible to wildcard the PID either.

So I am going to give up on this for now. We cannot get old BIOSes to work without an Interface number of 0, and we are already using Interface 0 to designate the CircuitPython CDC in the Windows driver package. We cannot change that easily, for legacy support reasons.

If we stopped supporting Windows 7 and 8.1 completely, we could start messing around with changing the Interface numbers.

It appears that at least some new BIOSes don't need boot keyboards anyway, so it's not terrible to give up on this.