hathach/tinyusb #1192: Error on plug and unplug of keyboard from USB Host

rppicomidi, 2021-11-12:
### Operating System

Others

### Board

raspberry pi pico (rp2040)

### Firmware

examples/host/cdc_msc_hid/ master branch commit 6f59a2c

I am developing in Visual Studio Code and using a picoprobe for debug and serial console display.

### What happened ?

The first time you plug a keyboard to the pico board, the keyboard works fine. The second time, the host fails to request a report and the keyboard does not work.

I did some digging on this and have two commits that seem to fix the issue for me. The commits were:
1. Implement the hcd_device_close function in the hcd_rp2040.c file. The master branch version seems to be stub
2. Clear the dev->ep_status in the process_device_unplugged function in usbh.c (after the hcd_device_close function call).

I am new to this project, new to the raspberry pi pico and new to github. Do you accept pull requests from newbies?

### How to reproduce ?

1. Build the cdc_msc_hid example (make sure the tinyusb code is up to date on the master branch)
2. Load the debug build of the code to the target pico board using a picoprobe board
3. Connect an Acer AU-0833 generic USB keyboard to the pico board via a microUSB to USB A adapter
4. Observe the board enumerates
5. Type some characters and observe they show on a terminal window
6. Unplug the keyboard and plug it back in
7. Observe error messages and verify keys typed do not display on the console

### Debug Log

I didn't enable logging, but the console log seems clear

TinyUSB Host CDC MSC HID Example
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Keyboard
HID device address = 1, instance = 1 is mounted
HID Interface Protocol = None
HID has 2 reports 
A device with address 1 is mounted
Couldn't find the report info for this report !
abcde I will now unplug and replugA device with address 1 is unmounted 
HID device address = 1, instance = 0 is unmounted
HID device address = 1, instance = 1 is unmounted
HID device address = 1, instance = 0 is mounted
HID Interface Protocol = Keyboard
Error: cannot request to receive report
HID device address = 1, instance = 1 is mounted
HID Interface Protocol = None
HID has 2 reports 
Error: cannot request to receive report
A device with address 1 is mounted

hathach, 2021-11-15:
Thanks for reporting the issue, seems like host stack does not clean up all its state nicely. I will try to reproduce the issue and checkout your Pr #1193 as soon as I could.

hathach, 2021-11-30:
fixed by #1193

closed 2021-11-30 by hathach