hathach/tinyusb #1196: Issues with ESP32-S3

203Null, 2021-11-15:
### Operating System

Windows 10

### Board

ESP32-S3-DevKitC-1

### Firmware

examples\device\cdc_msc_freertos

### What happened ?

tud_task() caused a crash in examples\device\cdc_msc_freertos

### How to reproduce ?

Compile the example for espressif_addax_1. (I know it's not the same board, but I can't tell any difference)

### Debug Log

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4201ee3e
0x4201ee3e: esp_pm_impl_waiti at C:/esp-idf/esp-idf-4.4/components/esp_pm/pm_impl.c:839

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1718
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2eb4
entry 0x403b61c0
I (29) boot: ESP-IDF v4.4-dev-3657-gdf0929009a-dirty 2nd stage bootloader
I (29) boot: compile time 06:12:06
I (29) boot: chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08b58h ( 35672) map
I (112) esp_image: segment 1: paddr=00018b80 vaddr=3fc922c0 size=046fch ( 18172) load
I (117) esp_image: segment 2: paddr=0001d284 vaddr=40374000 size=02d94h ( 11668) load
I (125) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1fa98h (129688) map
I (154) esp_image: segment 4: paddr=0003fac0 vaddr=40376d94 size=0b520h ( 46368) load
I (165) esp_image: segment 5: paddr=0004afe8 vaddr=50000000 size=00010h (    16) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
W (173) bootloader_random: RNG for ESP32-S3 not currently supported
I (192) cpu_start: Pro cpu up.
I (192) cpu_start: Starting app cpu, entry point is 0x4037524c
0x4037524c: call_start_cpu1 at C:/esp-idf/esp-idf-4.4/components/esp_system/port/cpu_start.c:156

I (170) cpu_start: App cpu up.
I (206) cpu_start: Pro cpu start user code
I (206) cpu_start: cpu freq: 160000000
I (206) cpu_start: Application information:
I (209) cpu_start: Project name:     cdc_msc_freertos
I (214) cpu_start: App version:      0.12.0-122-g6f59a2ce-dirty
I (221) cpu_start: Compile time:     Nov 15 2021 06:10:19
I (227) cpu_start: ELF file SHA256:  b8cb4fc117a1d896...
I (233) cpu_start: ESP-IDF:          v4.4-dev-3657-gdf0929009a-dirty
I (240) heap_init: Initializing. RAM available for dynamic allocation:
I (247) heap_init: At 3FC98628 len 000479D8 (286 KiB): D/IRAM
I (254) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (260) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (267) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (273) spi_flash: detected chip: generic
I (278) spi_flash: flash io: dio
W (282) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (295) sleep: Configure to isolate all GPIO pins in sleep state
I (302) sleep: Enable automatic switching of GPIO sleep configuration
I (309) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (20) TUSB: TUSB TASK
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4037a41e  PS      : 0x00050033  A0      : 0x4037a367  A1      : 0x3fc981c0  
0x4037a41e: vPortYieldFromInt at C:/esp-idf/esp-idf-4.4/components/freertos/port/xtensa/portasm.S:625

0x4037a367: _frxt_int_exit at C:/esp-idf/esp-idf-4.4/components/freertos/port/xtensa/portasm.S:231

A2      : 0x00000022  A3      : 0x00000000  A4      : 0x3fc956a0  A5      : 0x4037a2e8
0x4037a2e8: _frxt_int_enter at C:/esp-idf/esp-idf-4.4/components/freertos/port/xtensa/portasm.S:119

A6      : 0x00000001  A7      : 0xffffffff  A8      : 0x80378623  A9      : 0x3fc93460
A10     : 0x00000000  A11     : 0x3fc98190  A12     : 0x80378698  A13     : 0x3fc98250
A14     : 0x600c0034  A15     : 0x00000001  SAR     : 0x00000007  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000022  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe

Backtrace:0x4037a41b:0x3fc981c00x4037a364:0x3fc981d0 0x0005001e:0xa5a5a5a5  |<-CORRUPTED
0x4037a41b: vPortYieldFromInt at C:/esp-idf/esp-idf-4.4/components/freertos/port/xtensa/portasm.S:624

0x4037a364: _frxt_int_exit at C:/esp-idf/esp-idf-4.4/components/freertos/port/xtensa/portasm.S:230

ELF file SHA256: b8cb4fc117a1d896

Rebooting...

203Null, 2021-11-15:
I tried to compile the example because I run into issues with my custom firmware, tud_task() simply hangs my code.

203Null, 2021-11-16:
I manage to fix it by change the 
`#define USBD_STACK_SIZE     (3*configMINIMAL_STACK_SIZE/2) `
to 
`#define USBD_STACK_SIZE     (3*configMINIMAL_STACK_SIZE)`

I guess during the development, the stack size increased and no longer fits inside the original static stack

203Null, 2021-11-16:
For some reason, my ESP_LOGI() doesn't want to print again inside the task if followed by tud_task(). What could be the cause of this?

[image]

hathach, 2021-11-16:
currently, the tud_task() won't return, it will pend until there is new event. I have been thinking to tweak this behavior a bit, may do it later on. This issue is now closed, since it is not bug but a stack overflow.

closed 2021-11-16 by hathach

203Null, 2021-11-16:
> currently, the tud_task() won't return, it will pend until there is new event. I have been thinking to tweak this behavior a bit, may do it later on. This issue is now closed, since it is not bug but a stack overflow.

I guess it is a bug for the examples but not the stack.

hathach, 2021-11-16:
> > currently, the tud_task() won't return, it will pend until there is new event. I have been thinking to tweak this behavior a bit, may do it later on. This issue is now closed, since it is not bug but a stack overflow.
> 
> I guess it is a bug for the examples but not the stack.

yeah, you are right, I don't have any esp3s3 now for testing, will update it later one when testing with one.

203Null, 2021-11-16:
> yeah, you are right, I don't have any esp3s3 now for testing, will update it later one when testing with one.

I replicated the same issue with ESP32S2.

me-no-dev, 2021-11-16:
what is the value of `configMINIMAL_STACK_SIZE`? When printing inside a task, more stack is required. To be on the safe side, I would suggest 4KB as good value on ESP chips

hathach, 2021-11-16:
@me-no-dev `configMINIMAL_STACK_SIZE` probably 128 or something, I always set the buffer size to smallest in example to prevent RAM buffer on smaller mcu. Since there is already an #if for ESP, I will push an update to bump it up to 4k

chegewara, 2021-11-16:
Can we make it configurable? Maybe from Kconfig or something like that?
Currently esp-idf is optimized and 3kB is enough for tasks with logs printing, and tinyusb not always will require to print logs.
Just idea, not requirement.

me-no-dev, 2021-11-16:
@chegewara it's an example, so setting it to the safe 4K should be enough. If it was a library feature, then maybe it would have made sense to be configurable.

hathach, 2021-11-17:
@chegewara this is only for simple example for demonstration only. There is no way they will use all sram on the esp32sx, there is really no need to make it configurable from kconfig. The issue is with smaller mcu with limited sram.

PR #1215 (2021-11-29): use 4k stack for example with esp32sx
**Describe the PR**
resolve #1196, give 4k stack size for esp32s2/3 in freertos example