hathach/tinyusb #1201: Broadcom USB address

tannewt, 2021-11-20:
### Operating System

Linux

### Board

Pi Zero 2W

### Firmware

CircuitPython: https://github.com/tannewt/circuitpython/tree/rpi/

### What happened ?

usb_init hangs

### How to reproduce ?

Run CircuitPython on the Zero 2W (BCM2837). It's USB is at a different address than the BCM2711. I think it's that address for the BCM2836 as well. How do you want to manage the different locations? Should the client code set it instead of having TinyUSB do it?

```patch
diff --git a/src/portable/synopsys/dwc2/dwc2_bcm.h b/src/portable/synopsys/dwc2/dwc2_bcm.h
index fcec0b0d..18e9d874 100644
--- a/src/portable/synopsys/dwc2/dwc2_bcm.h
+++ b/src/portable/synopsys/dwc2/dwc2_bcm.h
@@ -34,7 +34,7 @@
 #include "broadcom/interrupts.h"
 #include "broadcom/caches.h"
 
-#define DWC2_REG_BASE       0xFE980000UL
+#define DWC2_REG_BASE       0x3F980000UL
 #define DWC2_EP_MAX         8
 #define DWC2_EP_FIFO_SIZE   4096
```

hathach, 2021-11-20:
We should use the `USB_OTG_GLOBAL_BASE` defined by broadcom peripheral repo. I hardcoded it since initially there is some macro naming conflict, plan to change it later but kind of forget.

lurch, 2021-11-22:
I believe extracting these addresses from DeviceTree is the most portable way of handling it? But that might be a whole different kettle of fish...?

hathach, 2021-11-22:
@lurch yeah, it is already done by https://github.com/adafruit/broadcom-peripherals which generate the header for peripherals and `USB_OTG_GLOBAL_BASE` https://github.com/adafruit/broadcom-peripherals/blob/main-build/broadcom/gen/bcm2837_lpa.h#L3558

TinyUSB simply picks that up

PR #1206 (2021-11-22): Bcm dwc2 address
**Describe the PR**
- update broadcom peripheral
- fix #1201
closed 2021-11-22 by hathach