hathach/tinyusb #1205: USB CDC device crashes when under heavy load

nikisalli, 2021-11-22:
### Operating System

Linux

### Board

custom rp2040 board

### Firmware

this will stop receiving or writing data after 20 seconds when being put under stress using `cat /dev/urandom | pv > /dev/ttyACM1` or `cat /dev/urandom | pv -L 14400 > /dev/ttyACM1` to limit the throughput

main.c
```
#include <pico/stdlib.h>
#include <string.h>
#include <tusb.h>

uint32_t get_time_ms(){
    return to_ms_since_boot(get_absolute_time());
}

int main (void) {
    tusb_init();

    char buf[200];

    while (1) {
        tud_task();

        if (tud_cdc_n_connected(0)) {
            uint32_t len = tud_cdc_n_available(0);
            while (len--) {
                tud_cdc_n_write_char(0, tud_cdc_n_read_char(0));
                // tud_cdc_n_write_flush(0);
            }
		}

        if (tud_cdc_n_connected(1)) {
            uint32_t len = tud_cdc_n_available(1);
            while (len--) {
                tud_cdc_n_write_char(1, tud_cdc_n_read_char(1));
                // tud_cdc_n_write_flush(1);
            }
		}

        if (get_time_ms() % 20 == 0) {
            sprintf(buf, "TIME: %.2f\n", get_time_ms() / 1000.0);
            tud_cdc_n_write_str(2, buf);
            sleep_ms(3);
		    // tud_cdc_n_write_flush(2);
        }
    }
}
```
tusb_config.h
```
#if !defined(_TUSB_CONFIG_H_)
#define _TUSB_CONFIG_H_

#include <tusb_option.h>

#define CFG_TUD_CDC_EP_BUFSIZE 512
#define CFG_TUD_CDC_EPSIZE 512
#define CFG_TUD_EP_MAX 16
#define CFG_TUSB_RHPORT0_MODE OPT_MODE_DEVICE

#define CFG_TUD_CDC 3 // two hardware uarts and one fake uart for communication
#define CFG_TUD_CDC_RX_BUFSIZE 512
#define CFG_TUD_CDC_TX_BUFSIZE 512

#endif /* _TUSB_CONFIG_H_ */
```
usb-descriptors.c
```
#include <tusb.h>

#define DESC_STR_MAX 20

#define USBD_VID 0x2E8A /* Raspberry Pi */
#define USBD_PID 0x000A /* Raspberry Pi Pico SDK CDC */

#define USBD_DESC_LEN (TUD_CONFIG_DESC_LEN + TUD_CDC_DESC_LEN * CFG_TUD_CDC)
#define USBD_MAX_POWER_MA 500

#define USBD_ITF_CDC_0 0
#define USBD_ITF_CDC_1 2
#define USBD_ITF_CDC_2 4
#define USBD_ITF_MAX 6

#define USBD_CDC_0_EP_CMD 0x81
#define USBD_CDC_0_EP_OUT 0x02
#define USBD_CDC_0_EP_IN 0x82

#define USBD_CDC_1_EP_CMD 0x83
#define USBD_CDC_1_EP_OUT 0x04
#define USBD_CDC_1_EP_IN 0x84

#define USBD_CDC_2_EP_CMD 0x85
#define USBD_CDC_2_EP_OUT 0x06
#define USBD_CDC_2_EP_IN 0x86

#define USBD_CDC_CMD_MAX_SIZE 8
#define USBD_CDC_IN_OUT_MAX_SIZE 64

#define USBD_STR_0 0x00
#define USBD_STR_MANUF 0x01
#define USBD_STR_PRODUCT 0x02
#define USBD_STR_SERIAL 0x03
#define USBD_STR_CDC 0x04

static const tusb_desc_device_t usbd_desc_device = {
	.bLength = sizeof(tusb_desc_device_t),
	.bDescriptorType = TUSB_DESC_DEVICE,
	.bcdUSB = 0x0200,
	.bDeviceClass = TUSB_CLASS_MISC,
	.bDeviceSubClass = MISC_SUBCLASS_COMMON,
	.bDeviceProtocol = MISC_PROTOCOL_IAD,
	.bMaxPacketSize0 = CFG_TUD_ENDPOINT0_SIZE,
	.idVendor = USBD_VID,
	.idProduct = USBD_PID,
	.bcdDevice = 0x0100,
	.iManufacturer = USBD_STR_MANUF,
	.iProduct = USBD_STR_PRODUCT,
	.iSerialNumber = USBD_STR_SERIAL,
	.bNumConfigurations = 1,
};

static const uint8_t usbd_desc_cfg[USBD_DESC_LEN] = {
	TUD_CONFIG_DESCRIPTOR(1, USBD_ITF_MAX, USBD_STR_0, USBD_DESC_LEN,TUSB_DESC_CONFIG_ATT_SELF_POWERED, USBD_MAX_POWER_MA),
	TUD_CDC_DESCRIPTOR(USBD_ITF_CDC_0, USBD_STR_CDC, USBD_CDC_0_EP_CMD,USBD_CDC_CMD_MAX_SIZE, USBD_CDC_0_EP_OUT, USBD_CDC_0_EP_IN,USBD_CDC_IN_OUT_MAX_SIZE),
	TUD_CDC_DESCRIPTOR(USBD_ITF_CDC_1, USBD_STR_CDC, USBD_CDC_1_EP_CMD,USBD_CDC_CMD_MAX_SIZE, USBD_CDC_1_EP_OUT, USBD_CDC_1_EP_IN,USBD_CDC_IN_OUT_MAX_SIZE),
	TUD_CDC_DESCRIPTOR(USBD_ITF_CDC_2, USBD_STR_CDC, USBD_CDC_2_EP_CMD,USBD_CDC_CMD_MAX_SIZE, USBD_CDC_2_EP_OUT, USBD_CDC_2_EP_IN,USBD_CDC_IN_OUT_MAX_SIZE),
};

static const char *const usbd_desc_str[] = {
	[USBD_STR_MANUF] = "Raspberry Pi",
	[USBD_STR_PRODUCT] = "Pico",
	[USBD_STR_SERIAL] = "000000000000",
	[USBD_STR_CDC] = "Board CDC",
};

const uint8_t *tud_descriptor_device_cb(void)
{
	return (const uint8_t *) &usbd_desc_device;
}

const uint8_t *tud_descriptor_configuration_cb(uint8_t index)
{
	return usbd_desc_cfg;
}

const uint16_t *tud_descriptor_string_cb(uint8_t index, uint16_t langid)
{
	static uint16_t desc_str[DESC_STR_MAX];
	uint8_t len;

	if (index == 0) {
		desc_str[1] = 0x0409;
		len = 1;
	} else {
		const char *str;

		if (index >= sizeof(usbd_desc_str) / sizeof(usbd_desc_str[0]))
			return NULL;

		str = usbd_desc_str[index];
		for (len = 0; len < DESC_STR_MAX - 1 && str[len]; ++len)
			desc_str[1 + len] = str[len];
	}

	desc_str[0] = (TUSB_DESC_STRING << 8) | (2 * len + 2);

	return desc_str;
}
```

### What happened ?

tinyusb stops writing or reading data when under heavy load on one of the three usb cdc devices plus some sensor reading overhead (the 3ms delay at 50Hz).
What I tried so far:
- running the usb routine on core1
- modifying https://github.com/Noltari/pico-uart-bridge to read my sensors in the main loop (crashes after a couple minutes with the sensors but worked for many hours without the sensors for some reason)

I tried stepping through my firmware using gdb and from what I see, tinyusb application api functions simply stop returning data or writing it but the program doesn't get stuck

### How to reproduce ?

run the attached firmware, spam one port and wait for it to completely stop responding (happens after 30 seconds in my tests)

hathach, 2021-11-22:
thank you for the issue
1. can this be reproduced with the stock example `/examples/device/cdc_msc` or `cdc_dual_ports` ? If not, then please add an example based on those with minimal changes so that the example is readily for compiling and testing. (it is how I normally test/develop the stack). Also that makes it easier to review/comment on your specific line of code. The easiest way is
   - Fork this repo, checkout a new branch
   - Add your-own-example based on stock one
   - Push and post it here.
2. which exactly way/command for spamming the cdc port.

nikisalli, 2021-11-24:
ok after some tests I was able to reproduce the issue here https://github.com/nikisalli/tinyusb/commit/92152ccdb5dcef8636f957a8659cd247efb1ed2a
just run this two commands in two terminals:
cat /dev/ttyACM0
cat /dev/urandom | dd of=/dev/ttyACM1 status=progress

it halts after a couple seconds and never reads a packet again until reset

HiFiPhile, 2021-11-25:
It might be RP2040 specific since I can't reproduce on STM32L053.

[image]

nikisalli, 2021-11-25:
I found out now that the bug occurs only when the program is loaded in RAM using a debugger. I've tried running again all the programs that crashed when loaded in ram using a picoprobe and it didn't crash not even one time. At this point I think the problem might lay in picoprobe's firmware so I'm closing the issue as clearly tinyusb isn't what's causing the problem

closed 2021-11-25 by nikisalli

hathach, 2021-11-25:
thanks for confirmation, sometime running in debug mode can cause some issue. It is normally a race. Feel free to re-open the issue if you figure out other way to reproduce it. Thanks @HiFiPhile for testing it out.