hathach/tinyusb #1231: Need accurate SOF event on RP2040 for explicit feedback

vmilea, 2021-12-04:
First of all, thanks for TinyUSB! It's an awesome little library.

I'm implementing a sound card using the RP2040. The device provides an asynchronous output endpoint and an explicit feedback endpoint to manage the data rate.

The feedback method is described in `5.12.4.2` of the USB 2.0 spec: 

> To generate the desired data rate Ff, the device must measure its actual sampling rate Fs, referenced to the USB notion of time, i.e., the USB (micro)frame frequency. [...] A sink can determine Ff by counting cycles of the master clock Fm for a period of 2^(K-P) (micro)frames. The counter is read into Ff and reset every 2^(K-P) (micro)frames. As long as no clock cycles are skipped, the count will be accurate over the long term.

So at each SOF we measure elapsed CPU clock cycles. The CPU clock frequency is fixed to a multiple of the sample rate, and by dividing we get the exact number of samples played each frame. This approach is very reliable and low-latency.

But the SOF event needs to be processed straight from IRQ handler to minimize jitter. When I tried dispatching on the `dcd_event_bus`, the extra 1-2us of jitter was causing audio glitches.

vmilea, 2021-12-04:
Proposal: Declare a weak callback in `dcd_rp2040.c`. If available, the callback is to be triggered on SOF. For minimal jitter, `dcd_rp2040_irq` should be declared `__not_in_flash_func`, and `USB_INTS_DEV_SOF_BITS` should be checked first.

If this sounds ok, please let me know and I'll submit a pull request.

HiFiPhile, 2021-12-04:
I've implemented a 32bit/384kHz sound card on LPC4357 which runs perfectly without glitch at about 10% CPU usage.

In fact the way of feedback calculation in the spec is not optimal, as you said the jitter of 1-2 us was causing audio glitchs.

What I'm using and proved reliable is measure the sample count of the out buffer in every `tud_audio_rx_done_pre_read_cb`, and use this count to calculate feedback value.
Here is my snippet:
```
bool tud_audio_rx_done_pre_read_cb(uint8_t rhport, uint16_t n_bytes_received, uint8_t func_id, uint8_t ep_out, uint8_t cur_alt_setting)
{
    (void)rhport;
    (void)func_id;
    (void)ep_out;
    (void)cur_alt_setting;

    fifo_count = tud_audio_available();

    // filter fifo_count
    fifo_count_flt = (fifo_count_flt * 255 + ((uint32_t)fifo_count << 8)) >> 8;

    // enough sample in buffer, start I2S playback
    if(fifo_count >= buffer_threshhold)
    {
        if(!buffer_ready)
        {
            buffer_ready = true;
            fifo_count_flt = fifo_count << 8;
        }
    }
    return true;
}

bool tud_audio_fb_done_cb(uint8_t rhport)
{
    (void)rhport;

    current_feedback = feedback_std - ((fifo_count_flt >> 8) - buffer_threshhold);
    if(current_feedback < feedback_min) current_feedback = feedback_min;
    if(current_feedback > feedback_max) current_feedback = feedback_max;

    tud_audio_fb_set(current_feedback);
    return true;
}
```

vmilea, 2021-12-04:
Thanks! This looks viable provided you have the magic constants figured out (min, max, threshold). It's the sort of code I was hoping for in the device examples. :)

That said, the SOF way definitely works once you mitigate the jitter. It's running glitch-free with a small TinyUSB patch on the Raspberry Pi Pico.

To be clear, the project I'm working on is for my own entertainment - to learn and see how far I can push cheap hardware. I'm aiming for low audio latency, and the advantage of the SOF approach is it locks the data rate immediately with minimal delay for buffering. In effect `2 * TUD_AUDIO_EP_SIZE` FIFO.

So I think it's worth supporting this use case for latency-sensitive applications.

HiFiPhile, 2021-12-04:
> To be clear, the project I'm working on is for my own entertainment 

Me too :)

I've uploaded my repo, you can find more details in `audio_callbacks.c`

vmilea, 2021-12-04:
> I've uploaded my repo, you can find more details in audio_callbacks.c

Awesome, thank you!

hathach, 2021-12-06:
yeah, adding user sof make sense for this case. SOF callback isn't used yet (on purpose) since it is triggered too often (1ms for FS). I will try to see what would be the best way to add it.

hathach, 2021-12-07:
> Here is my current implementation: [vmilea@bf61702](https://github.com/vmilea/tinyusb/commit/bf617028f90d87475a3864dac9bb7a09311563d7). The SOF interrupt is enabled only when `dcd_sof_cb()` is defined, so there is zero overhead unless used.

thanks, IMHO, sof interrupt must be actively enabled/disabled e.g when going into low power like wfi/wfe or when audio interface is not active. We don't want to have SOF triggered on every 1 ms in those scenario. This should be an tud_ API that applies to all ports. And since not all application needs this, its code should be explicitly enabled by and CFG_TUD (though I think the additional code size is pretty small).

geekbozu, 2021-12-14:
I quite like @vmilea's implementation. I actually needed this for a project as well and ported it over to the stm32 synopsys driver and it works great!

I do think adding in proper support for low power modes would be preferred however.

PanRe, 2022-03-01:
@hathach Do you think we could implement the SOF nicely into the audio framework together with low power mode handling? I think new users would be happy of this feature.

hathach, 2022-03-07:
thanks for the reminder, I just made an pr #1381 that added the tud_sof_isr_set() to set sof handler. This will enable/disable the SOF irq respectively. Please let me know your feedback/review on that PR.

PR #1381 (2022-05-31): Add SOF IRQ Handler 
**Describe the PR**
mostly implement and close #1231 , add new
- dcd_sof_enable(uint8_t rhport, bool en)
- void tud_sof_isr_set(tud_sof_isr_t sof_isr)
- implement dcd_sof_enable() for rp2040
- add dcd_sof_enable() stubs for other ports for now

Other unrelated clean up
- remove broadcom/synopsys (replaced by synopsys/dwc2)
- add CFG_TUD_INTERFACE_MAX for max number of interfaces, default to <s>8</s> 16.
closed 2022-05-31 by hathach