hathach/tinyusb #1249: stm32_fsdev isochronous support

leptun, 2021-12-18:
### Operating System

Windows 10

### Board

stm32g474nucleo

### Firmware

I was using the uac2_headset example code. I tried using it on a stm32g474nucleo board, but any mcu which uses stm32_fsdev should have this issue.

### What happened ?

An assert is hit at runtime:
```
dcd_edpt_open 760: ASSERT FAILED
```
Which basically says that isochronous mode is not implemented in the stm32_fsdev driver.

Is there any interest in implementing the isochronous mode in this driver? Right now the audio class can't work on a large portion of the stm32 devices (basically the cheaper ones, where OTG is not present). I imagine there might be more classes that use this endpoint mode, but this was the only class I needed at the moment. I wish I was able to switch to an f4 mcu (dwc2 driver), but with the current chip shortage, that is not possible.
If there is no plan to make isochronous mode work, can there be a note added to these devices so that examples with unimplemented features are not built?

### How to reproduce ?

Just run any of the examples which use isochronous transfers on the stm32_fsdev driver. Easiest board to test with might be a blue pill or similar. nucleo g4 boards with a USB shield also work. In my setup I'm using a X-NUCLEO-USBPDM1 TypeC power delivery shield which also connects the data lines to the usb port.

### Debug Log

[log_iso.txt](https://github.com/hathach/tinyusb/files/7739990/log_iso.txt)

hathach, 2021-12-20:
> Is there any interest in implementing the isochronous mode in this driver?

Yeah, there is the interest but just not enough time and people to implement it.

pigrew, 2021-12-30:
In order to implement this, we'd also want to implement DMA in order to support the higher data rates. To perform DMA, we'd want some sort of abstraction layer in the case that DMA is used in other parts of the application code. Also, DMA will be slightly different on the various variants of the STM32, so I'd expect adding DMA support would be a large undertaking.

If you don't think that DMA is necessary, this could be done without _too_ much effort, though I personally don't need ISO endpoints, so I don't plan to write it myself.

HiFiPhile, 2021-12-30:
> To perform DMA, we'd want some sort of abstraction layer in the case that DMA is used in other parts of the application code.

That's something I'm also want to be implemented. 

But I don't think stm32_fsdev has DMA support ?

PanRe, 2022-03-01:
For DMA support, we would need the `dcd_edpt_xfer_fifo()` to get working. I once pushed some untested code implementing that feature. Curiously, right now all of that code is commented out but present in the main branch. All parts are marked with _// TODO support dcd_edpt_xfer_fifo API_. I was not able to test the code because i don't have any hardware.
Directly copying data into the USB hardware buffer by use of DMA is not possible also not in the synopsis case. The trick is to copy the data into the FIFO by use of two DMAs (ringbuffer -> linear and wrapped part) and than schedule `dcd_edpt_xfer_fifo()`. The process of copying the data from the fifo into the hardware buffer is taken care of by my untested code. This way, the DMA stuff is left to the user. Works the same way as for synopsis.

Iktek, 2022-06-07:
Hi guys, I Also stumbled over missing isoc support on fsdev. 
For my current "fist test" I would be fine if there's no dma-support for now, could we split this issue or is it mandatory to support dma in order to get isoc transfers running?
I'm willing to help with implementation and test but maybe I'd need someone to discuss my changes as I'm not that deep into fsdev driver and tinyusb stack up to now. 
I'll first try to comment in that untested stuff and to adopt the changed routines. 
Is there somebody willing to discuss the "missing bits" in order to get that working?

I try to implement that here:
https://github.com/Iktek/tinyusb/pull/new/1249-feature-stm32_fsdev_isocronous_support

With first tests I got my linux kernel acceping some isoc-transfers but for some reason the transfer stops after a while not kowing what is wrong from within the device:

[wireshark_stm32_fsdev_iso_out.txt](https://github.com/hathach/tinyusb/files/8851670/wireshark_stm32_fsdev_iso_out.txt)

leptun, 2022-06-07:
@HiFiPhile I think that is correct. The stm32f103rc for example doesn't have any way to connect the DMA to the USB. It also doesn't have an internal dma controller.
I also checked the stm32g474re and it's the same in this sense. That one has a much more sophisticated DMA controller, but even that one can't connect to the USB (although I noticed in one of the diagrams there is some extra FIFO, but I didn't read more into that).

As far as I know, there is no stm32 that actually can talk to the `DMA1` or `DMA2`. In all cases where a DMA is involved, the usb block incorporates an extra DMA just for the usb. Also, this internal DMA always appears for `USB_OTG_HS` (dwc2) interfaces. It rarely appears on `USB_OTG_FS` (still dwc2, take the stm32h743zi as a counterexample for this). The `USB` interface is the one that needs the fsdev driver, but that one seems to never incorporate an internal DMA. 

Here is an example from the stm32f439zi document where only the `USB_OTG_HS` gets the DMA:
[image]
[image]

And here it is for the stm32f103rc, no dma:
[image]

And here for stm32g474re, no dma:
[image]

Iktek, 2022-06-07:
@HiFiPhile @leptun : Mhh. You may be right, but it could also be that m2m dma is still possible as the pma memory can be accessed from within the normal adressing-space. So for my understanding there only needs to be a dma-connection if requests of an periperal need to be made which would not be the case for pma. 
Nevertheless there should also be a way to implement isocronous support without dma. 
I'll first try to also add the ff-api support to the driver as already proposed by the "commented-out" code.
However maybe one could have a look in my wireshark-dissections if one has an idea why I see that -EXDEV erros and why the linux-host stops the transfer after a short while. Also with dynamic_debug enabled in the linux-kernel I don't see a cause. I'm also wondering why there are 2 iso-descriptors per out-transfer as one 64byte packet would be sufficient for the configured audio-stream (16khz 16bit stereo).
Thanks in advance
Pascal

Iktek, 2022-06-07:
One step back: there seems to be an issue with my usb-host driver -> with another notebook the stream stays alive. 
It seems that isoc-support is basically running with my minor fixes / additions.. I'll test that further and add an PR for that bit if it's running here. 
DMA is on another page and might be tackled in the future.

Iktek, 2022-06-07:
Sure:
#define AUDIO_CTRL_ID_SPK_INPUT_STREAM   0x01
#define AUDIO_CTRL_ID_SPK_FUNIT			 0x02
#define AUDIO_CTRL_ID_SPK_OUTPUT 		 0x03
#define AUDIO_CTRL_ID_MIC_INPUT			 0x04
#define AUDIO_CTRL_ID_MIC_FUNIT			 0x05
#define AUDIO_CTRL_ID_MIC_OUTPUT_STREAM  0x06
#define AUDIO_CTRL_ID_CLOCK				 0x41
#define AUDIO_CTRL_ID_CLOCKSEL		     0x40

#define AUDIO_NUM_INTERFACES 			 0x03
#define AUDIO_NUM_INCHANNELS			 0x02
#define AUDIO_NUM_OUTCHANNELS			 0x02

#define TUD_AUDIO_CTRL_TOTAL_LEN (\
			TUD_AUDIO_DESC_CLK_SRC_LEN+\
			TUD_AUDIO_DESC_INPUT_TERM_LEN+\
			TUD_AUDIO_DESC_OUTPUT_TERM_LEN+\
			TUD_AUDIO_DESC_INPUT_TERM_LEN+\
			TUD_AUDIO_DESC_OUTPUT_TERM_LEN)

#define TUD_AUDIO_IO_DESC_LEN (\
    TUD_AUDIO_DESC_IAD_LEN\
  + TUD_AUDIO_DESC_STD_AC_LEN\
  + TUD_AUDIO_DESC_CS_AC_LEN\
  + TUD_AUDIO_CTRL_TOTAL_LEN\
  /* Speaker Interface */\
  + TUD_AUDIO_DESC_STD_AS_INT_LEN\
  + TUD_AUDIO_DESC_STD_AS_INT_LEN\
  + TUD_AUDIO_DESC_CS_AS_INT_LEN\
  + TUD_AUDIO_DESC_TYPE_I_FORMAT_LEN\
  + TUD_AUDIO_DESC_STD_AS_ISO_EP_LEN\
  + TUD_AUDIO_DESC_CS_AS_ISO_EP_LEN\
  + TUD_AUDIO_DESC_STD_AS_ISO_EP_LEN\
  /* Microphone Interface */ \
  + TUD_AUDIO_DESC_STD_AS_INT_LEN\
  + TUD_AUDIO_DESC_STD_AS_INT_LEN\
  + TUD_AUDIO_DESC_CS_AS_INT_LEN\
  + TUD_AUDIO_DESC_TYPE_I_FORMAT_LEN\
  + TUD_AUDIO_DESC_STD_AS_ISO_EP_LEN\
  + TUD_AUDIO_DESC_CS_AS_ISO_EP_LEN)

#define TUD_AUDIO_MIC_ONE_CH_DESC_N_AS_INT 1 	// Number of AS interfaces

#define TUD_AUDIO_IO_DESCRIPTOR(_itfnum, _stridx, _nBytesPerSample, _nBitsUsedPerSample, _epin, _epinsize, _epout, _epoutsize, _epfb) \
  /* Standard Interface Association Descriptor (IAD) */\
  TUD_AUDIO_DESC_IAD(_itfnum, AUDIO_NUM_INTERFACES, /*_stridx*/ 0x00),\
  /* Audio Control Interface */ \
  /* Standard AC Interface Descriptor(4.7.1) */\
  TUD_AUDIO_DESC_STD_AC(/*_itfnum*/ _itfnum, /*_nEPs*/ 0x00, /*_stridx*/ _stridx),\
  /* Class-Specific AC Interface Header Descriptor(4.7.2) */\
  TUD_AUDIO_DESC_CS_AC(/*_bcdADC*/ 0x0200, AUDIO_FUNC_CONVERTER, TUD_AUDIO_CTRL_TOTAL_LEN, /*_ctrl*/ AUDIO_CS_AS_INTERFACE_CTRL_LATENCY_POS),\
  TUD_AUDIO_DESC_CLK_SRC(AUDIO_CTRL_ID_CLOCK, AUDIO_CLOCK_SOURCE_ATT_INT_FIX_CLK, (AUDIO_CTRL_R << AUDIO_CLOCK_SOURCE_CTRL_CLK_FRQ_POS), /*_assocTerm*/ 0x01,  /*_stridx*/ 0x00),\
  /* Speaker Terminals */ \
  TUD_AUDIO_DESC_INPUT_TERM(AUDIO_CTRL_ID_SPK_INPUT_STREAM, AUDIO_TERM_TYPE_USB_STREAMING, AUDIO_CTRL_ID_MIC_OUTPUT_STREAM, AUDIO_CTRL_ID_CLOCK, AUDIO_NUM_INCHANNELS, AUDIO_CHANNEL_CONFIG_FRONT_LEFT | AUDIO_CHANNEL_CONFIG_FRONT_RIGHT, /*_idxchannelnames*/ 0x00, /*_ctrl*/ AUDIO_CTRL_R << AUDIO_IN_TERM_CTRL_CONNECTOR_POS, /*_stridx*/ 0x00),\
  TUD_AUDIO_DESC_OUTPUT_TERM(AUDIO_CTRL_ID_SPK_OUTPUT, AUDIO_TERM_TYPE_OUT_GENERIC_SPEAKER, AUDIO_CTRL_ID_SPK_INPUT_STREAM, AUDIO_CTRL_ID_SPK_INPUT_STREAM, AUDIO_CTRL_ID_CLOCK, /*_ctrl*/ 0x0000, /*_stridx*/ 0x00),\
  /* Microphone Terminals */ \
  TUD_AUDIO_DESC_INPUT_TERM(AUDIO_CTRL_ID_MIC_INPUT, AUDIO_TERM_TYPE_IN_GENERIC_MIC, AUDIO_CTRL_ID_MIC_OUTPUT_STREAM, AUDIO_CTRL_ID_CLOCK, AUDIO_NUM_INCHANNELS, AUDIO_CHANNEL_CONFIG_FRONT_LEFT | AUDIO_CHANNEL_CONFIG_FRONT_RIGHT, /*_idxchannelnames*/ 0x00, /*_ctrl*/ AUDIO_CTRL_R << AUDIO_IN_TERM_CTRL_CONNECTOR_POS, /*_stridx*/ 0x00),\
  TUD_AUDIO_DESC_OUTPUT_TERM(AUDIO_CTRL_ID_MIC_OUTPUT_STREAM, AUDIO_TERM_TYPE_USB_STREAMING, AUDIO_CTRL_ID_MIC_INPUT, AUDIO_CTRL_ID_MIC_INPUT, AUDIO_CTRL_ID_CLOCK, /*_ctrl*/ 0x0000, /*_stridx*/ 0x00),\
  \
  /* Speaker Interface */ \
  /* Interface 1, Alternate 0 - default alternate setting with 0 bandwidth */\
  TUD_AUDIO_DESC_STD_AS_INT(/*_itfnum*/ (uint8_t)((_itfnum) + 1), /*_altset*/ 0x00, /*_nEPs*/ 0x00, /*_stridx*/ 0x00),\
  /* Interface 1, Alternate 1 - alternate interface for data streaming */\
  TUD_AUDIO_DESC_STD_AS_INT(/*_itfnum*/ (uint8_t)((_itfnum) + 1), /*_altset*/ 0x01, /*_nEPs*/ 0x02, /*_stridx*/ 0x00),\
  /* Class-Specific AS Interface Descriptor(4.9.2) */\
  TUD_AUDIO_DESC_CS_AS_INT(AUDIO_CTRL_ID_SPK_INPUT_STREAM, AUDIO_CTRL_NONE, AUDIO_FORMAT_TYPE_I, AUDIO_DATA_FORMAT_TYPE_I_PCM, AUDIO_NUM_OUTCHANNELS, AUDIO_CHANNEL_CONFIG_FRONT_LEFT | AUDIO_CHANNEL_CONFIG_FRONT_RIGHT, /*_stridx*/ 0x00),\
  /* Type I Format Type Descriptor(2.3.1.6 - Audio Formats) */\
  TUD_AUDIO_DESC_TYPE_I_FORMAT(_nBytesPerSample, _nBitsUsedPerSample),\
  /* Standard AS Isochronous Audio Data Endpoint Descriptor(4.10.1.1) */\
  TUD_AUDIO_DESC_STD_AS_ISO_EP(_epout,(TUSB_XFER_ISOCHRONOUS | TUSB_ISO_EP_ATT_ASYNCHRONOUS | TUSB_ISO_EP_ATT_DATA), _epoutsize, TUD_OPT_HIGH_SPEED ? 0x04 : 0x01),\
  /* Class-Specific AS Isochronous Audio Data Endpoint Descriptor(4.10.1.2) */\
  TUD_AUDIO_DESC_CS_AS_ISO_EP(AUDIO_CS_AS_ISO_DATA_EP_ATT_NON_MAX_PACKETS_OK, AUDIO_CTRL_NONE, AUDIO_CS_AS_ISO_DATA_EP_LOCK_DELAY_UNIT_UNDEFINED, 0x0000),\
  /* Standard AS Isochronous Feedback Endpoint Descriptor(4.10.2.1) */\
  TUD_AUDIO_DESC_STD_AS_ISO_FB_EP(_epfb, 1),\
  \
  /* Microphone Interface */ \
  /* Standard AS Interface Descriptor(4.9.1) */\
  /* Interface 1, Alternate 0 - default alternate setting with 0 bandwidth */\
  TUD_AUDIO_DESC_STD_AS_INT((uint8_t)((_itfnum)+2), /*_altset*/ 0x00, /*_nEPs*/ 0x00, /*_stridx*/ 0x00),\
  /* Standard AS Interface Descriptor(4.9.1) */\
  /* Interface 1, Alternate 1 - alternate interface for data streaming */\
  TUD_AUDIO_DESC_STD_AS_INT(/*_itfnum*/ (uint8_t)((_itfnum)+2), /*_altset*/ 0x01, /*_nEPs*/ 0x01, /*_stridx*/ 0x00),\
  /* Class-Specific AS Interface Descriptor(4.9.2) */\
[... cut ...]
  TUD_AUDIO_DESC_TYPE_I_FORMAT(_nBytesPerSample, _nBitsUsedPerSample),\
  /* Standard AS Isochronous Audio Data Endpoint Descriptor(4.10.1.1) */\
  TUD_AUDIO_DESC_STD_AS_ISO_EP(_epin, (TUSB_XFER_ISOCHRONOUS | TUSB_ISO_EP_ATT_ASYNCHRONOUS | TUSB_ISO_EP_ATT_DATA), _epinsize, TUD_OPT_HIGH_SPEED ? 0x04 : 0x01),\
  /* Class-Specific AS Isochronous Audio Data Endpoint Descriptor(4.10.1.2) */\
  TUD_AUDIO_DESC_CS_AS_ISO_EP(AUDIO_CS_AS_ISO_DATA_EP_ATT_NON_MAX_PACKETS_OK, AUDIO_CTRL_NONE, AUDIO_CS_AS_ISO_DATA_EP_LOCK_DELAY_UNIT_UNDEFINED, 0x0000)

Iktek, 2022-06-07:
Same descriptor is working properly with stm32l4 + dwc2
I'm wondering if the "doublebuffering" is needed in order to support isoc-audio as it seems that maybe every 2nd pice of data is not handled. 
I'm also wondering if it's supported by the fsdev that there are 2 isoc-descriptors with 64 byte in each urb. Maybe one of them is not handled properly.

HiFiPhile, 2022-06-07:
> I'm wondering if the "doublebuffering" is needed in order to support isoc-audio as it seems that maybe every 2nd pice of data is not handled.

I don't think it's the cause. As there is no ack or retry for ISO transfer, host will continue to send even device can't handle it properly.
I remember Windows continued to send even if I paused the execution of LPC4337.
In your case host is sending only half of the data.

Seems like feature unit is missing in descriptors.

Iktek, 2022-06-07:
Mhh.. where do you see that the host is missing to send the data? In the dissection I see that the host is sending about 128bytes every 2 ms which is the correct data amount. I would suggest that the device can't handle every 2nd packet (or e.g the 2nd descriptor in the iso-out-urb)

HiFiPhile, 2022-06-07:
> Mhh.. where do you see that the host is missing to send the data? 

I'm sorry I misread your Wireshark log 😅 I'll take a deeper look after vacation.

pigrew, 2022-06-07:
> Hi @Iktek , I believe DMA is not supported on fsdev after reading reference manual and ST driver, I could be wrong.

The peripheral cannot make DMA requests, but the operating system/tinyusb can periodically initiate "MEM2MEM" transfers as needed (preferably using the double-buffer mode of the USB peripheral). Handling DMA complete interrupts manually shouldn't be so hard to do. Error handling (i.e., when USB disconnects happen) could be complicated if a DMA transfer is in progress in the background.

Currently, the stm32_fsdev driver copies data into the peripheral buffer during interrupts. Perhaps a simple optimization to speed things up a bit would be to synchronously perform a DMA MEM2MEM transfer (i.e., start the transfer and busy-loop until the transfer finishes) instead of the memcpy. We maybe would want to define an OS API to request/reserve a DMA channel for the exclusive use of tinyusb and to enable/disable the DMA controller. Do we have any sort of OS DMA API that's used for other architectures?

Having the operating system manage a queue of DMA requests in the background seems doable and probably would give the best speed, but I'm not sure if that's needed? Is the current (slow) implementation limiting performance at the moment?

pigrew, 2022-06-07:
> Same descriptor is working properly with stm32l4 + dwc2 I'm wondering if the "doublebuffering" is needed in order to support isoc-audio as it seems that maybe every 2nd pice of data is not handled.

I think `stm32_fsdev` must use double-buffering, based on the reference manual section `23.4.4 isochronous transfers`:

> Isochronous endpoints implement double-buffering to ease application software development, using both ‘transmission’ and ‘reception’ packet memory areas to manage buffer swapping on each successful transaction in order to have always a complete buffer to be used by the application, while the USB peripheral fills the other.

Iktek, 2022-06-08:
> The peripheral cannot make DMA requests, but the operating system/tinyusb can periodically initiate "MEM2MEM" transfers as needed
I agree. 

I think a busy-loop will not be a solution which would speed up much as irq is still blocked during memcpy. 
I don't expect the dma-transfer being much faster then the manual memcpy. I'ts only worth being used if we could make it asyncronous.

I'll have a look these days if double-buffering is doable easy. 
I think dma is not mandatory for now but would maybe ease up things.
If DMA would not help in Design, I would postpone it and try to implement double-buffering with copying in user-space first as this seems to be the crux. 

As I think I have to dig in more I'll come up with a proposal here these days.

Iktek, 2022-06-08:
Ok, i digged in a bit and things seem even more complicated as thought before:
Double-buffered isoc-endpoints can't be bedirectional. So one would have to use 2 endpoint indices here. 
As the driver states endpoint id and ix are mixed up all-over-the-world, so it would be hard to get them devided again. 
Unfortunately UAC2 needs the sync endpoint being on the same address ( 0x01 <->0x81 ) as the corresponding data-endpoint.. so we have a no-go here.

In addition my customer phone me he would want to pause the current STM32G4 project (as the lead time for this cpu is currently to high), which means that I currently can't spent hours any more on pushing this further :-(, I would have liked to but getting all that stuff in would be way to much for now. Hopefully I can resume that project on a given point in time and hopefully I can remember what needs to be done. If someone else would want to push this further here are some steps I think which need to be done to get isynch-transfers in:
1. separate endpoint-ix and endpoint-id in order to allow bidirectional, double-buffered endpoints
2. Introduce some API or automated way to allocate endpoint-ix without binding to endpoint-addr
3. allocate a twice-the-maxPktSize pma buffer for double-buffered endpoints and set the addresses accordingly 
( e.G. TX, which corresponds to RX0 or TX0 to &buffer, RX, which corresponds to RX1 or TX1 to &buffer[maxPktSize] )
4. implement the dtog-bit-handling
5. move the data-copying and buffer-swapping to thread-context

So if nobody else has time to work on this I'll hopefully continue soon and now have a list to remember what to do.. I'm also happy about every input / ideas which would make things easier. Maybe I also got something wrong...

Regards Pascal

rolandvs, 2022-07-28:
@Iktek Is there any change you or perhaps others @hathatch are/have been working on getting this to work or made any progress since the last post?

pigrew, 2022-07-28:
@rolandvs, It's not really on my list of things to do, since I'm not that interested in media streaming. I don't think anyone else has worked on it.

I have, however, been thinking about the implementation of double-buffering, and using FIFO for data transfer, and ISO should be simple after double-buffering is implemented. I'd like @hathach's input on how endpoint configuration could be implemented: How should an application request that an endpoint be double-buffered or not, and also allow an application to specify a buffer's location in the PMA. Do we need to add a application callback to get parameters? This would vary across architectures, so could make the code difficult....  Should we default to single-buffered, except for ISO which would be double-buffered?

I'm imagining perhaps:

```
struct stm32_fsdev_ep_cfg {
bool double_buffered;
uint32_t pma_start; // -1 for automatic allocation. 
};

int tusb_get_ep_cfg(int rhport, int config, int interface, int ep, struct stm32_fsdev_ep_cfg *cfg);
```

Other architectures would have different parameters in the configuration struct.

rolandvs, 2022-07-28:
Thanks for your quick reply! Ok, we have a need to get this running on an G491. Is it OK to drop a few questions as you have pursued this subject before?

pigrew, 2022-07-28:
> Thanks for your quick reply! Ok, we have a need to get this running on an G491. Is it OK to drop a few questions as you have pursued this subject before?

Sure (I wrote the stm32fsdev driver here...), as long as they are related to ISO... otherwise, we could open a separate discussion.

Iktek, 2022-07-29:
> @Iktek Is there any change you or perhaps others @hathatch are/have been working on getting this to work or made any progress since the last post?

Hi @rolandvs , there have been some "experiments" in https://github.com/Iktek/tinyusb/tree/1249-feature-stm32_fsdev_isocronous_support where I enabled ISO without doublebuffering, which did not work out, so: nothing worthy I think. Unfortunately project priorities shifted and other things got more urgent for now (thanks to the chip-crisis).. So basically I'm also still interested getting things running (now escpecially on STM32L151), but unfortunately I have no chance to invest much time here now. Nevertheless I'm willing to do some testing on other chips / situations / descriptors or give some discussion-input / code-reviewing if this could be helpful.

@pigrew : As there seem to be (only) a few people which seem to need this feature I would vote that we do not invest tooo much time in API-questions for now, but first rework the driver-structure in a way to allow doublebuffering by design (e.g. separate endpoint-ix and endpoint-id a.s.o.) and get it working (maybe just for iso-xfers for now as they do generally not work without). We should continue discussing how to allow the user to enable/disable DB for non-iso-transfers but this is IMHO not the root problem for now. 

On the other Hand having a PMA-location would be needed. In other drivers (like dcw2) the allocation of the fifos is done "automatically" on endpoint-creation/deletion with a few limitations (e.g. closed endpoint's fifos are not freed, but re-used on re-open). Maybe we could also follow this example to create PMA-allocations.

As you already said that streaming is not your on your list to do, maybe you could just give some hint's how you would get that running / do organization of Id's etc. Are you on the same page then I that this reworking is neede? Hopefully s.o. will get time to do the implementation then ( If nobody is found, I'll hopefully get more time at the end of the year).

Expecting your inputs
Best Regards 
Pascal

pigrew, 2022-07-29:
The endpoint-ix vs endpoint-ix separation isn't necessarily needed if the application doesn't have a second endpoint with that particular ID (e.g., if the ISO EP is 3OUT, then there couldn't be a 3IN endpoint).

The driver already has an allocation method for the PMA. It's correct for most cases (where PMA is not also used for CAN).

The minimum changes need:
1. Modify allocator to support double-buffering.
2. Support longer PMA buffers (for packet lengths up to 512 bytes?), including changes to allocator
3. Update xfer callbacks/interrupt handler for double-buffering

--
Some other notes:
* Data toggle should always be set to DATA0 when transmitting. (peripheral handles this internally)
* Peripheral uses the data toggle bit in the control register to specify which buffer to use when operating double-buffered.
* ISO endpoint never send a handshake response (no NAK or ACK response, ever).

There are a few things that I don't know how to handle:
* Should lost packets be detected? What's the mechanism for error handling? The FSDEV peripheral has a `USBFNR.FN` frame number register which could be used to detect if a packet was skipped.
* During double-buffering, should the xfer_complete callback be called when the data has been transmitted, or when there is at least one free buffer?
* Does the FIFO interface need to be implemented? For the FIFO, how does one transmit a short or zero-length packet? (needed if FIFO were used for other protocols.). I'm more familiar with the TI USBD API, where an endpoint can be setup as a FIFO or using block buffers, but one cannot switch between them after the endpoint is opened.

rolandvs, 2022-07-31:
Thanks for all your comments. I'll see if I can make an effort, but unfortunately it will have to wait as I'm currently working on some issues with higher priority.

pigrew, 2022-07-31:
> Thanks for all your comments. I'll see if I can make an effort, but unfortunately it will have to wait as I'm currently working on some issues with higher priority.

I started looking at the driver, too, and have found at least one bug (#1583), and got confused about error handling while trying to fix it (#1584 ). I'm reworking the code with a goal of handling double-buffering and searching for race conditions. I'll submit it as a PR once I have it working better than the current code. It's entirely possible that ISO transfers will work with the block-buffer interface once I make these changes.

Iktek, 2022-08-01:
> The endpoint-ix vs endpoint-ix separation isn't necessarily needed if the application doesn't have a second endpoint with that particular ID (e.g., if the ISO EP is 3OUT, then there couldn't be a 3IN endpoint).

Hi pigrew: Unfortunately the UAC2-Specification (Audio-Streaming) is expecting exactly that. An explicit sync endpoint needs necessarily have the same endpoint-ix as its corresponding stream endpoint. 
I wonder if there are also other classes which expect this to be possible.
So for my opinion the discussed separation is mandatory.
Best Regards Pascal

pigrew, 2022-08-08:
I've started working on rewriting the fsdev driver, to add the layer of indirection between endpoint addresses and hardware endpoints, plus double-buffering... and finding tons of bugs.

Some bugs include:
* Having different EP types with the same address was broken
* Receiving odd numbers of bytes potentially causes memory overruns
* Driver was broken if one attempts to open a control EP with address != 0 (changing to just error out if attempted)
* Add check for CMSIS >=5.1
* Broken with EP >=256 bytes

The changes are about 30% complete... lots are left to do. Hopefully the reworks won't bloat code size and memory usage too much.

My code is currently broken, but here it is if your curious:

https://github.com/pigrew/tinyusb/tree/stm32_doublebuf

Iktek, 2022-08-26:
Hi @pigrew , 
thank you for taking the job pushing this further. 
Is there anything I could do for you? (testing / take over detail implementations)
Or is there anything where you document your "missing bits" so I could do some groundwork?
Regards Pascal

skuep, 2022-10-21:
Yeah, thanks from me as well! I am currently trying to put the UAC2 audio class into a stm32. 
I have built a lot of systems with USB interface and getting CDC class to run with the internal UART was exceptionally easy and I am really impressed by tinyusb.
This evening I tried the UAC2 stuff and I was very sad, when I found out, that it does not work at all.
I haven't done a lot with STM32's USB stuff, but if there is anything I could do to help, let me know.

skuep, 2022-10-23:
Hi guys, I had a look at the work from @Iktek and I could solve the issue with every odd buffer data being 0. The reason for this was as already suspected, that the second buffer in the double-buffer system was never written to. 

With these changes I got the ``audio_text`` example working without buffer dropouts:
https://github.com/skuep/tinyusb/tree/0.14.0-isofix

skuep, 2022-10-23:
Hi PanRe,
The untested portion is from @Iktek. I have not yet concerned myself with it, sorry. I am still working on verifying whether the opposite direction (SPEAKER instead of MICROPHONE) works with my code.

Iktek, 2022-10-24:
Hi guys, I did not continue with my approach as I was waiting for a good state of @pigrew 's work. Because: my first approach still has the problem with different endpoint directions having the same address and this had to be tidied up first. This is mandatory because of UAC2 sync endpoint needing to have the same address as the corresponding data endpoint. Unfortunately I don't have much time to dig in the current changes of @pigrew, so I need to wait for some "stable" state of the code with commented missing bits to continue work here.. or maybe some of you will bring the bits togehter :-).. Regards Pascal

skuep, 2022-10-24:
Hi Iktek,
Are you sure that the endpoint address requirement is mandatory? I have had a quick look in Release 1.0 of the UAC specification (here https://www.usb.org/sites/default/files/audio10.pdf) and on page 61 in the "Standard AS Isochronous Audio Data Endpoint Descriptor" Table, there is at Offset 8 a field for ``bSynchAddress``to define the synchronization endpoint address.
Can't we use that?

Iktek, 2022-10-24:
Hi @skuep, yes you're mixing up UAC and UAC2. 
I also hat tests in implementing UAC class but the UAC-driver on windows does not really allow syncing. So in order to get a valid synced audio you would need to implement UAC2. And UAC2 has removed the bSyncAddress field. From UAC2 on the sync-address needs to sit on the same index as the corresponding data endpoint. So if you don't want to implement an ASRC on your device, you'd either need to have a sync-endpoint or you need to live with frame-drops. (One could also implement an tricky jitterbuffer which adds or removes just one sample) but with a sine wave this would be hearable.
Regards Pascal

skuep, 2022-10-24:
Hm, very dissatisfying, but thanks for the information. I would have gone on to work on using UAC1, since that is actually enough in my case. No way to get asynchronous clocks to work on a Windows machine?
Simon

skuep, 2022-10-24:
Hi again, @Iktek,
Can you give me a link to the @pigrew tinyusb fork? Is he currently working on this feature?
I just had some spare time and I have implemented a rather crude way of handling this issue.
See this commit https://github.com/hathach/tinyusb/commit/af46a53bdfaf925332bbff4446bc2196972b2c69
By default, the tinyusb stack works just as normal, but I added an application callback, that you can use to alter the "mapping" of the endpoint address to the internal endpoint registers.
From my application:

````
/* Endpoints */
#define EPNUM_AUDIO_IN      0x81
#define EPNUM_AUDIO_OUT     0x02
#define EPNUM_AUDIO_FB      0x82
    ....
````

````
// We have ISOCHRONOUS endpoints defined that share the same endpoint number, but have opposing directions.
// However with STM32 hardware, ISOCHRONOUS endpoints use both RX and TX structures of the same endpoint register in hardware
// We circumvent a clash by defining our own custom endpoint map for the tiny usb stack
uint8_t tu_stm32_edpt_number_cb(uint8_t addr)
{
    switch (addr) {
    case 0x00:
    case 0x80:
        /* Endpoint Zero */
        return 0x00;

    case EPNUM_AUDIO_IN:
        return 0x01;

    case EPNUM_AUDIO_OUT:
        return 0x02;

    case EPNUM_AUDIO_FB:
        return 0x03;

    default:
        TU_BREAKPOINT();
        return 0x00;
    }
}

From my limited testing of this functionality, this seems to work quite well, I can just shift around the endpoint registers willy-nilly by changing the return values in the ``tu_stm32_edpt_number_cb``callback. However I have not yet tested the actual speaker feature.

pigrew, 2022-10-24:
> Hi again, @Iktek, Can you give me a link to the @pigrew tinyusb fork? Is he currently working on this feature?

I got sidetracked, and didn't get nearly as far as I had hoped.

The tricky part is handling the allocations of the PMA for double-buffered endpoints (and I found that the existing code had some rather substantial bugs). I was rewriting the PMA allocation code and endpoint housekeeping to handle double-buffering.

The link is https://github.com/pigrew/tinyusb/tree/stm32_doublebuf (and you can see that I've not worked on it in a few months....

pigrew, 2022-10-24:
> @pigrew: Thanks for the heads-up. I will dig through your changes and see what's what :-) What issues with the PMA for double buffered endpoints did you have? Since tinyusb does not seem to implement any double-buffering features (for now?) I basically now just left everything as-is with PMA allocation and just set both (TX and RX) (double-) buffer locations to the same buffer.
> 
> EDIT: Had a look at your code. Yeah, you are totally right with the PMA allocation bug. I am going to steal that if you don't mind :-)

You are correct, no double-buffered in the existing code, but other bugs did exist (I think it was that the TX and RX on a given endpoint were assumed to be of the same type...).

There was a use case where there must be a TX and RX endpoint for a given USB logical endpoint number (required by audio, IIRC? I think it was mentioned above). We need to allocate two double-buffered physical endpoint addresses (one for TX and one for RX), to properly support this, creating an extra layer of indirection.

During my implementation, I kept getting confused about concurrency issues: mostly if we had to set various things as `volatile` since they were used both inside and outside interrupts, and also if I should be using C atomic types. (or maybe the RTOS abstraction layer).

skuep, 2022-10-25:
Yeah, this is exactly the problem we are running into here. I am currently still using the method of custom mapping the endpoint address to the hardware endpoint using a callback function. However I have made some progress in my fork:
- I added your fix regarding the buffer size during allocation to be aligned to the allowable hardware sizes
- I added what you call "indirection" into the code where it was necessary. I re-used the already present ``xfer_ctl_t``structure

https://github.com/hathach/tinyusb/compare/master...skuep:0.14.0-sk

So now we could theoretically go ahead and implement a ``dcd_edpt_alloc`` function that is to be called instead of ``tu_edpt_number`` in ``dcd_edpt_open`` for finding the next free endpoint to use. For now I will work on different parts of my project, since I am currently satisfied using the custom mapping using the callback function (as explained above).

skuep, 2022-11-12:
Hi guys, so I got UAC2 speaker and microphone working (thanks to @Iktek for the descriptor example above). See my tinyusb branch here:
https://github.com/hathach/tinyusb/compare/master...skuep:0.14.0-sk

Here is the device I have implemented. Note that currently, although feedback generation is working, it is lacking the long-term buffer drift prevention Feature required by the USB spec. So when playing an audio file for half an hour, buffer creeps dangerously low.
https://github.com/skuep/AIOC

HiFiPhile, 2022-11-12:
>  it is lacking the long-term buffer drift prevention Feature required by the USB spec (it is not implemented in tinyusb). 

I've implemented a feedback method by keeping fifo at a stable level. https://github.com/hathach/tinyusb/pull/1463

Maybe it can help you.

skuep, 2022-11-12:
Cool thanks! I will have a look at it.

I have been reading and thinking about how to do this and (as other people do) I think that a feedback purely based on FIFO count is not according to specs because you basically implement a P controller on your device (i.e. you use some kind of feedback to keep the FIFO count constant). 
However, only the host is supposed to have a controller, the device only states the number of samples played. The spec then goes on and says that over the long term, the buffer fill level may drift and only in case this is detected, are you allowed to modify the feedback value.

So I am not sure if the dual controller scenario can result in instability issues in case the parameters are not chosen wisely. However that is just my understanding at the moment. If any of you are the wiser, please let me know

HiFiPhile, 2022-11-12:
> So I am not sure if the dual controller scenario can result in instability issues in case the parameters are not chosen wisely.

In theory it's possible, so I've limited feedback min/max to a small interval.

I did all tests on HS, on which fifo level is easier to be regulated than FS, result is satisfying amount 44.1KHz to 384kHz.

> However, only the host is supposed to have a controller, the device only states the number of samples played.

IMHO is the UAC specification who should be blamed. What it want to achieve is a regulated loop, but in real it's not a closed-loop system as the feedback is only an estimation.

skuep, 2022-11-12:
> IMHO is the UAC specification who should be blamed. What it want to achieve is a regulated loop, but in real it's not a closed-loop system as the feedback is only an estimation.

I agree. The UAC spec is not really the USB consortium finest work if you ask me. Why didn't they just specify that the device regularly send a monotonically increasing counter (integral) of all samples played up to now (and let it wrap around at 2**32-1). There would be no issue whatsoever with buffer level drift, because the host could directly control the amount of samples send using the amount of samples played. 🤔
Instead they settled in sending the derivative of this counter resulting in integration inaccuracies that then need to be "hacked" in... Anyway

skuep, 2022-12-10:
Jep, It is working so far with a STM32F302 which is, as far as I can tell, largely compatible with the STM32F103, therefore it should work.
However it needed a lot of tweaking of the STM32 driver in TinyUSB and also still has some crude solutions, like the static mapping of endpoint addresses to hardware endpoints using a callback function. Thus I use my own TinyUSB fork here: https://github.com/skuep/tinyusb/tree/0.14.0-sk

While the project I have linked above is not a USB audio speaker sample code, it is in fact basically only a soundcard with 1 mono microphone channel and 1 mono speaker channel.

vorosj, 2022-12-10:
Thank you very much for the fast reply.

vorosj, 2022-12-10:
I did a quick test: downloaded your AIOC and TinyUSB code, and compiled it in STM32CubeIDE. Success, no errors.
Unfortunately I have no STM32F302, so I blindly uploaded the firmware to a STM32F303CCT6. 
It enumerated as mono speaker and mic, plus serial port. Wow, this uC-s has to be very similar... :-)
No sound output on PA4, the pin is floating - probably the chip is different here. Also checked the Led outputs, 36MHz signal on one of them.
Next I try to port the code to the 303, figure out what is different.
Thank you very much for the great work.

vorosj, 2022-12-11:
_Btw, did you connect the AIOC to a Windows PC?_
Sure, it enumerated as mono mic, speaker and serial port. When I plug in the usb, win11 selects it as audio output, and silence. If I select the audio output to the normal and back to the uC, it's always working. However no sound on PA4, probably there is a difference in the DAC peripheral. The absolute bare metal DAC init is a bit steep for me :-), I usually use CubeMX for the job.
I touched the pins to check if it's working as mic, no response. It was just a short unreliable trial, so not sure if there is difference too, or just missed the input pin.

In order to port the code to the 303, I first started DAC with DMA in a new empty project. CubeMX generated me a faulty code. It initiated the DMA and the DAC in the wrong order, set a register in the DAC before it has clock. Takes me several hours to figure out, but now works. However, at first look no need for DMA in your project, so dont't know why I bother with it...

I don't yet learnt the code in details, not sure how the in and out audio stream keep sync. I am not familiar with the usb standards, not even know if the device itself has to sync to the stream, or send feedback to the PC and the PC side is responsible for the sync.

My project is amplitude modulated signal for an old tube radio. :-) Not too much broadcast are still working in the 0.5...1.5MHz band, and I like old radios. USB sound stream -> AM signal transforms the old radio to netradio, of course without any mod on the radio side.

The job is harder then I estimated, because the controllers I have either not working with usb audio class (CubeMX code is broken, TinyUSB audio class missing for fsdev), or no DAC in the uC.  USB is complicated. I don't even know what fsdev and dcw2 are. Why - for example - the STM32F103 has fsdev and not working, but the very similar GD32VF103 has dcw2 and working? 

My first succesful effort was an F411, it works with the CubeMX generated code, and with TinyUSB as well. But no DA. I tried PWM, it's working well, but filtering out the 48kHz sampling freq requires steep filter, and I like to keep it as  simple as possible. Of course it's for fun, challenge accepted, I want to solve it without steep active analog filter in the audio part.  :-)

PanRe, 2022-12-11:
Up to my knowlegde, dcw2 is intended to fuse all the device drivers of the individuel MCUs together. This, however, is work in progress and unfortunately bugs fixed in dwc2 may not be fixed in the individual drivers and vice versa. I think the docs should be improved to inform everybody of its purpose...

The audio class works fine so far, most bugs reported regarding audio concern the ISO EP support in the device drivers. The same holds true for fsdev.

I am happy there is progress for the 103 and 303 devices! It would be cool if the driver would become mature enough for a PR! :)

Iktek, 2022-12-11:
> In order to port the code to the 303, I first started DAC with DMA in a new empty project. CubeMX generated me a faulty code. It initiated the DMA and the DAC in the wrong order, set a register in [...]

Just FYI: I also use CubeMX and had this issue. I posted it to the forum and it should be fixed in the current CubeMX releases. Nevertheless if you're using an "old" ioc file you should manually edit the order of the Init functions with is coded in a human readable form.

> Up to my knowlegde, dcw2 is intended to fuse all the device drivers of the individuel MCUs together. This, however, is work in progress and unfortunately bugs fixed in dwc2 may not be fixed in the [...]

IMHO it's more a point of which IP-Core is used in the Product. So a few products of ST use the "fsdev" core of ST. Other Products use the DWC2 core which is a bought in core of DesignWare. 
So as long as u use a chip which has the "fsdev" core and it's register set implemented, the driver should work. 
For STM32L4xx devices you need to use dwc2 driver.

Regards Pascal

skuep, 2022-12-11:
> In order to port the code to the 303, I first started DAC with DMA in a new empty project. CubeMX generated me a faulty code. It initiated the DMA and the DAC in the wrong order, set a register in [...]

I would just load the STM32CubeIDE project fromy AIOC repository. Then go into project settings and change the MCU to the STM32F303 instead of 302. Hit compile again and try the code. It is well possible that this might already be enough to get it working. There is no need for DMA in my project. There would be no real benefit here. Note that it might still be possible that there is a problem with Windows compatibility, that I have not yet found.

> The job is harder then I estimated, because the controllers I have either not working with usb audio class (CubeMX code is broken, TinyUSB audio class missing for fsdev), or no DAC in the uC. USB is complicated.

I hear you. I thought my project would be a fun little project.. since I found out that there is basically no existing way to make a USB class device. Since then most of the work was getting my head around the USB audio stack and implementation specifics.

> I am happy there is progress for the 103 and 303 devices! It would be cool if the driver would become mature enough for a PR! :)

At least SOME progress 😃. Some stuff is still not quite up to standard I would say, like the manual endpoint mapping. Also some things I have modified and changed and these changes need to be tested against other tinyusb classes. You know where to find my branch if you want to have a go at testing or developing.

vorosj, 2022-12-11:
_I would just load the STM32CubeIDE project from my AIOC repository. Then go into project settings and change the MCU to the STM32F303 instead of 302. Hit compile again and try the code._ 

It compiled without errors for STM32F303CC.
Enumeration on win11: mono speaker, mic (TinyUSB Device), serial port
**The serial port is working.**(tested using putty terminal and scope)

It's possible to select it from win11 as sound otput, volume control works (except no sound).
No sound from the DAC tested by scope, stable low state on PA4.
The mic is not working (just touched the pin in mic test mode).
Led pins: 
B9 high, 
B8 1msec pulses, 50% or 95% high pulse stream, not sure why one or the other, doesn't change if I select/deselect the board as sound output

Considering it enumerates and the serial port is working, the main USB functions are working. Next I debug it, check if the buffer filled, the DAC register loaded with data etc.

Thank you very much for help.

I am HAM, HA6TI, and often use FT8 mode on my Xiegu G90. I have a xgcomm USB-digimode-3 interface, it's basically an usb soundcard, no serial port, so no built-in CAT control. It has a high bit adc/dac, but it's probably not important, as FT8 is not sensitive. I am sure your AIOC could do the job, plus CAT. :-) The xgcomm is isolated, it's important because the possible strong EM radiation.

vorosj, 2022-12-11:
Forgot to mention, in device manager/sound, video etc
device configured (usbaudio2.inf)
device started (usbaudio2)

vorosj, 2022-12-12:
Debugging captured my mistake immediately... :-)
I started the program in debug mode, then plugged in the USB cable. It enumerated. When I touched the volume control on the PC, debugging stopped, TU_Assert Isochronous not supperted yet.
It means I copied the official Tinyusb into your AIOC project instead of your mod. (I downloaded it in .zip format, this way without the linked TinyUSB).
After correcting my mistake it's working. :-)
Worth to mention, that in debug mode it still halts on an Assert, but without the Jlink adapter it works.

Thank you for the great work and help.
This way my project is almost ready, just have to start a free timer for the carrier signal, a simple hw AM modulator, RF filter, and ready.

skuep2, 2022-12-12:
> It compiled without errors for STM32F303CC. Enumeration on win11: mono speaker, mic (TinyUSB Device), serial port **The serial port is working.**(tested using putty terminal and scope)
Nice! you can also theoretically wire those pins in loopback configuration to test it.

> I am HAM, HA6TI, and often use FT8 mode on my Xiegu G90. I have a xgcomm USB-digimode-3 interface, it's basically an usb soundcard, no serial port, so no built-in CAT control. It has a high bit [...]

Very cool. Theoretically you could also use the serial interface for CAT right? Right now the AIOC is supposed to be for HTs only and the serial is for programming it. Yeah, the AIOC does not have isolation, because I would think it is not required for an HT. Might be different for a bigger rig.

> I started the program in debug mode, then plugged in the USB cable. It enumerated. When I touched the volume control on the PC, debugging stopped, TU_Assert Isochronous not supperted yet.
> It means I copied the official Tinyusb into your AIOC project instead of your mod. (I downloaded it in .zip format, this way without the linked TinyUSB).
> I cloned it using git, it's working. :-)

Nice! I think the best way is to clone the AIOC project and then run a ``git submodule update --init --recursive`` and it will pull the correct TinyUSB fork automatically.

> Thank you for the great work and help.
> This way my project is almost ready, just have to start a free timer for the carrier signal, a simple hw AM modulator, RF filter, and ready.

No worries, I am glad it is working well. Is it still throwing the assertion? I remember enabling the ISOCHRONOUS stuff only for a few hand-picked STM32 types here: https://github.com/skuep/tinyusb/blob/0.14.0-sk/src/portable/st/stm32_fsdev/dcd_stm32_fsdev_pvt_st.h. Look for ``#define ISOCHRONOUS_DOUBLEBUFFER``. However as far as I can tell, the STM32F303CC is included.

skuep, 2022-12-12:
> It compiled without errors for STM32F303CC. Enumeration on win11: mono speaker, mic (TinyUSB Device), serial port **The serial port is working.**(tested using putty terminal and scope)
Nice! you can also theoretically wire those pins in loopback configuration to test it.

> I am HAM, HA6TI, and often use FT8 mode on my Xiegu G90. I have a xgcomm USB-digimode-3 interface, it's basically an usb soundcard, no serial port, so no built-in CAT control. It has a high bit [...]

Very cool. Theoretically you could also use the serial interface for CAT right? Right now the AIOC is supposed to be for HTs only and the serial is for programming it. Yeah, the AIOC does not have isolation, because I would think it is not required for an HT. Might be different for a bigger rig. 

**Out of interest**: Is there any standardized or at least widely-used connector? It might be worthwhile to fork the k1-aioc PCB (has K1 connector) and modify the solder pads to solder on a DIN connector (or whatever is used). The firmware would be the same :-)

> I started the program in debug mode, then plugged in the USB cable. It enumerated. When I touched the volume control on the PC, debugging stopped, TU_Assert Isochronous not supperted yet.
> It means I [...]

Nice! I think the best way is to clone the AIOC project and then run a ``git submodule update --init --recursive`` and it will pull the correct TinyUSB fork automatically.

> Thank you for the great work and help.
> This way my project is almost ready, just have to start a free timer for the carrier signal, a simple hw AM modulator, RF filter, and ready.

No worries, I am glad it is working well. Is it still throwing the assertion? I remember enabling the ISOCHRONOUS stuff only for a few hand-picked STM32 types here: https://github.com/skuep/tinyusb/blob/0.14.0-sk/src/portable/st/stm32_fsdev/dcd_stm32_fsdev_pvt_st.h. Look for ``#define ISOCHRONOUS_DOUBLEBUFFER``. However as far as I can tell, the STM32F303CC is included.

vorosj, 2022-12-12:
_Is it still throwing the assertion?_

No, it!s working in debug mode with connected jlink too.

vorosj, 2022-12-12:
_Out of interest: Is there any standardized or at least widely-used connector?_

AFAIK no. My rig has a 3.5 stereo Jack for serial communication (CAT and fw update), and a 8 pin mini Din for audio IO (for fans of contact failure).

HiFiPhile, 2022-12-28:
@skuep 
Hi, I made some commits based on your fork, which includes:
- Rebased on master
- Minor fixes
- Reworked FIFO based transfer
- Added an endpoint allocator

I've tested with my CDC & HID composite device (mine FIFO enabled CDC class fork), could you test with your UAC enabled project ?
Especially if endpoint alloc/free goes well, you can also modify UAC class to enable FIFO based transfer.

https://github.com/HiFiPhile/tinyusb/tree/0.14.0-sk

skuep, 2022-12-28:
Cool! I will have a look tomorrow 😊

[3 more replies]

closed 2023-03-02 by hathach