hathach/tinyusb #1250: Raspberry Pi Pico Host with Hub Fails to get first bytes HID keyboard descriptor

rppicomidi, 2021-12-18:
### Operating System

Others

### Board

raspberry pi pico (rp2040)

### Firmware

examples/host/cdc_msc_hid commit cadfcd153e8fce804f38f0e19997c13bf6931a49

### What happened ?

The hub appears to recognize the keyboard is attached, does a hub port clear feature reset change, and then fails to complete a transfer to get 8 bytes of the Dev. Descriptor. Unplug of keyboard is not recognized after that. If I unplug the hub, the code recognizes that and enumerates the hub again on plug in. 
@hathach is this a known issue?

### How to reproduce ?

1. Modify the family.cmake file for the rp2040 to set TINYUSB_DEBUG_LEVEL to 3
2. Build cdd msc hid example within Visual Studio Code, debug level 3
3. Use a USB OTG adapter microB to A adapter to give the pico board a host connector
4. Connect a picoprobe to the pico to power the pico, then plug the picoprobe to the debug host PC to power the picoprobe and target pico board.
5. Use the debugger of Visual Studio Code with a picoprobe to run the code
6. Code starts normally. Plug a pluggable brand USB 2.0 powered 4-port hub to the pico board via the OTG adapter.
7. Observe the hub appears to enumerate normally
8. Plug in an Asus KKU-0833 keyboard to a hub port
9. Observe the enumeration process starts, but stall on getting the first 8-bytes of the device descriptor.

### Debug Log

TinyUSB Host CDC MSC HID Example
USBH init
CDC init
MSC init
HID init
HUB init
hcd_init 0
USBH DEVICE ATTACH
hcd_port_connect_status
Open EP0 with Size = 8 (addr = 0)
hcd_edpt_open dev_addr 0, ep_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Get 8 byte of Device Descriptor
Control Setup (addr = 0): 80 06 00 01 00 00 08 00 
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Transfer complete
Sent setup packet
on EP 00 with 0 bytes
hcd_edpt_xfer dev_addr 0, ep_addr 0x80, len 8
hw_endpoint_init dev 0 ep 0 in xfer 0
dev 0 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
  Prepare BufCtrl: [0] = 0x29704  [1] = 0x0000
Buffer complete
buf_status 0x00000001
Single Buffered: ep_ctrl = A0000180
  Sync BufCtrl: [0] = 0x57352  [1] = 0x0000
Completed transfer of 8 bytes on ep 0 in
Transfer complete
on EP 80 with 8 bytes
Control data (addr = 0):
  0000:  12 01 00 02 09 00 00 40                          |.......@|
hcd_edpt_xfer dev_addr 0, ep_addr 0x0, len 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
[... 447 lines cut ...]
on EP 00 with 0 bytes
hcd_edpt_xfer dev_addr 5, ep_addr 0x80, len 0
hw_endpoint_init dev 5 ep 0 in xfer 0
dev 5 ep 0 in setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
  Prepare BufCtrl: [0] = 0x29696  [1] = 0x0000
Buffer complete
buf_status 0x00000001
Single Buffered: ep_ctrl = A0000180
  Sync BufCtrl: [0] = 0x57344  [1] = 0x0000
  Short packet on buffer 0 with 0 bytes
Completed transfer of 0 bytes on ep 0 in
Transfer complete
on EP 80 with 0 bytes

Open EP0 with Size = 8 (addr = 0)
hcd_edpt_open dev_addr 0, ep_addr 0
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180
Get 8 byte of Device Descriptor
Control Setup (addr = 0): 80 06 00 01 00 00 08 00 
hw_endpoint_init dev 0 ep 0 out xfer 0
dev 0 ep 0 out setup buffer @ 0x50100180
endpoint control (0x50100100) <- 0xa0000180

hathach, 2021-12-20:
> is this a known issue?
yeah, since I never tried to get hub work with rp2040, there is no wonder if it doesn't work. But hub calss works with other mcu, it is just a matter of porting. Thank you for the issue, though could you put the log into attached txt for readibility.

jeroentaverne, 2021-12-20:
For more logfiles check my issue: https://github.com/hathach/tinyusb/discussions/1248#discussioncomment-1844605

rppicomidi, 2021-12-20:
Sorry. Should have attached the log as a file. I am investigating the issue
[pico-hub-kb.log](https://github.com/hathach/tinyusb/files/7746802/pico-hub-kb.log)
.

hathach, 2021-12-20:
> Sorry. Should have attached the log as a file. I am investigating the issue [pico-hub-kb.log](https://github.com/hathach/tinyusb/files/7746802/pico-hub-kb.log) .

no problems at all, I just update the 1st post to make it easier to read. Thanks for looking into the issue, I will try to test/troubleshoot this out later on as well.

rppicomidi, 2021-12-20:
Here is a log file for attaching a Dell USB mouse. The same hub enumeration log as previous issue, so only shows after mouse attachment detected. The enumeration gets further along, but the pico receives stall status when it attempts to set the mouse device address
[pico-hub-mouse.log](https://github.com/hathach/tinyusb/files/7746973/pico-hub-mouse.log)
.

hathach, 2022-03-15:
should be fixed by #1397, there is lots of host improvement recently, there is probably minor issues, but I got my hub + keyboard + mouse running fine with stock example on pico. Feel free to re-open if it is still an issue

closed 2022-03-15 by hathach

jeroentaverne, 2022-03-21:
Thanks for this fix.
Without USB HUB the keyboard LEDs will turn on using:
static uint8_t value = 7;
tuh_hid_set_report(keyboard_dev_addr, keyboard_instance, 0, HID_REPORT_TYPE_OUTPUT, (void*)&value, 1);
With USB hub the LEDs will not turn on with the code above. What can I do to control the LEDs through a USB hub?

hathach, 2022-03-29:
@jeroentaverne that is a different issue, please open an new issue with detailed information to reproduce, also include your hub info, and keyboard info with its configuration and report descriptor (when plugged in PC).