hathach/tinyusb #1275: HID Physical Descriptors

nonchip, 2022-01-12:
**Is your feature request related to a problem? Please describe.**

There's a `HID_DESC_TYPE_PHYSICAL` but `hidd_control_xfer_cb` ignores that case instead of calling a `tud_hid_descriptor_physical_cb`.

**Describe the solution you'd like**

For physical descriptors to be supported.

hathach, 2022-01-17:
yeah, indeed it is lacking the physical descriptor support. It seems to be an straight forward modification. Since you are working on it, do you have time to make an PR for this ? I could help if you share your example that has the physical descriptor. It probably need to include the physical descriptor in the interface as well.

nonchip, 2022-01-20:
the issue there would be that i don't actually have any working example yet, because i'm currently still stuck struggling getting HID to work. having issues with the standard contradicting itself yet again, and the old classic of applications not giving a damn about the standard to begin with... the funniest bit is that a piece of code copied essentially verbatim from mbed-os which definitely used to work 100% correctly with that application suddenly seems to be not even seen as a HID device by it after i port it over...

sooo yeah, tl;dr: doubt i can provide much indepth help or a working example there, was just noticing the lack of functionality which might be helpful when i get the remaining stuff to work :P

hathach, 2022-01-22:
Physical descriptor is rarely used I haven't seen one therefore it isn't supported just yet. If there is no practical usage, there is no need to write code for features that isn't used.

hathach, 2022-03-10:
closed due to lack of activities, feel free to submit an PR when it is ready.

closed 2022-03-10 by hathach