hathach/tinyusb #1295: Multi MIDI Device with RP2040

DatanoiseTV, 2022-01-21:
### Operating System

Linux

### Board

Raspberry Pico

### Firmware

examples/device/midi_test with modifications

### What happened ?

When trying to add multiple MIDI ports, the device will enumerate, but aconnect won't show MIDI ports anymore.

### How to reproduce ?

1. Clone the midi-test example
2. Change #define CFG_TUD_MIDI 4 in tusb_config.h
3. In usb_descriptors.c add..
```
To enum:
  ITF_NUM_MIDI_0 = 0,
  ITF_NUM_MIDI_STREAMING_0,
  ITF_NUM_MIDI_1,
  ITF_NUM_MIDI_STREAMING_1,
  ITF_NUM_MIDI_2,
  ITF_NUM_MIDI_STREAMING_2,
  ITF_NUM_MIDI_3,
  ITF_NUM_MIDI_STREAMING_3,

#define CONFIG_TOTAL_LEN (TUD_CONFIG_DESC_LEN + (TUD_MIDI_DESC_LEN * CFG_TUD_MIDI))

#define USBD_MIDI_0_EP_OUT 0x01
#define USBD_MIDI_1_EP_OUT 0x03
#define USBD_MIDI_2_EP_OUT 0x05
#define USBD_MIDI_3_EP_OUT 0x07

#define USBD_MIDI_0_EP_IN 0x81
#define USBD_MIDI_1_EP_IN 0x83
#define USBD_MIDI_2_EP_IN 0x85
#define USBD_MIDI_3_EP_IN 0x87

uint8_t const desc_fs_configuration[] =
    {
        // Config number, interface count, string index, total length, attribute, power in mA
        TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),

        // Interface number, string index, EP Out & EP In address, EP size
        TUD_MIDI_DESCRIPTOR(ITF_NUM_MIDI_0, 0, USBD_MIDI_0_EP_OUT, USBD_MIDI_0_EP_IN, 64),
        TUD_MIDI_DESCRIPTOR(ITF_NUM_MIDI_1, 1, USBD_MIDI_1_EP_OUT, USBD_MIDI_1_EP_IN, 64),
        TUD_MIDI_DESCRIPTOR(ITF_NUM_MIDI_2, 2, USBD_MIDI_2_EP_OUT, USBD_MIDI_2_EP_IN, 64),
        TUD_MIDI_DESCRIPTOR(ITF_NUM_MIDI_3, 3, USBD_MIDI_3_EP_OUT, USBD_MIDI_3_EP_IN, 64),

    };
```

hathach, 2022-02-01:
1. I have never tried multiple MIDI, however string index for interface cannot be 1,2,3 since those are already used for manufacturer, product string. If you dont have any use for those, try to change it to all 0
2. Please provide your example code which is compilable for reproducing
3. Also provide usb log with `LOG=2` and/or 3

DatanoiseTV, 2022-02-02:
> 1. I have never tried multiple MIDI, however string index for interface cannot be 1,2,3 since those are already used for manufacturer, product string. If you dont have any use for those, try to [...]

I have tried with string index 0.

Here are the files attached:
https://gist.github.com/DatanoiseTV/2499fee93d23234048e8d7b6cea88a38

rppicomidi, 2022-02-21:
@DatanoiseTV I do not think the issue is with TinyUSB. I modified the example descriptors.c using the following code snip. Now amidi -l shows all 4 virtual cables. Please note the change to the CONFIG_TOTAL_LEN macro. I hope this helps.
```
$ amidi -l
Dir Device    Name
IO  hw:1,0,0  TinyUSB Device MIDI 1
IO  hw:1,0,1  TinyUSB Device MIDI 2
IO  hw:1,0,2  TinyUSB Device MIDI 3
IO  hw:1,0,3  TinyUSB Device MIDI 4
```
Code Snip:
```
#define CONFIG_TOTAL_LEN(num_cables_)  (TUD_CONFIG_DESC_LEN + (TUD_MIDI_DESC_HEAD_LEN + TUD_MIDI_DESC_JACK_LEN * num_cables_ + TUD_MIDI_DESC_EP_LEN(num_cables_) * 2))

#if CFG_TUSB_MCU == OPT_MCU_LPC175X_6X || CFG_TUSB_MCU == OPT_MCU_LPC177X_8X || CFG_TUSB_MCU == OPT_MCU_LPC40XX
  // LPC 17xx and 40xx endpoint type (bulk/interrupt/iso) are fixed by its number
  // 0 control, 1 In, 2 Bulk, 3 Iso, 4 In etc ...
  #define EPNUM_MIDI   0x02
#else
  #define EPNUM_MIDI   0x01
#endif

#define MIDI_NUM_CABLES 4
uint8_t const desc_fs_configuration[] =
{
  // Config number, interface count, string index, total length, attribute, power in mA
  TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN(MIDI_NUM_CABLES), 0x00, 100),

  // Interface number, string index, EP Out & EP In address, EP size
  //TUD_MIDI_DESCRIPTOR(ITF_NUM_MIDI, 0, EPNUM_MIDI, 0x80 | EPNUM_MIDI, 64)
  TUD_MIDI_DESC_HEAD(ITF_NUM_MIDI, 0, MIDI_NUM_CABLES),
  TUD_MIDI_DESC_JACK(1),
#if MIDI_NUM_CABLES>1
  TUD_MIDI_DESC_JACK(2),
#if MIDI_NUM_CABLES>2
  TUD_MIDI_DESC_JACK(3),
#if MIDI_NUM_CABLES>3
  TUD_MIDI_DESC_JACK(3),
#endif
#endif
#endif
  TUD_MIDI_DESC_EP(EPNUM_MIDI, 64, MIDI_NUM_CABLES),
  TUD_MIDI_JACKID_IN_EMB(1),
#if MIDI_NUM_CABLES>1
  TUD_MIDI_JACKID_IN_EMB(2),
#if MIDI_NUM_CABLES>2
  TUD_MIDI_JACKID_IN_EMB(3),
#if MIDI_NUM_CABLES>3
  TUD_MIDI_JACKID_IN_EMB(4),
#endif
#endif
#endif
  TUD_MIDI_DESC_EP(0x80 | EPNUM_MIDI, 64, MIDI_NUM_CABLES),
  TUD_MIDI_JACKID_OUT_EMB(1),
#if MIDI_NUM_CABLES>1
  TUD_MIDI_JACKID_OUT_EMB(2),
#if MIDI_NUM_CABLES>2
  TUD_MIDI_JACKID_OUT_EMB(3),
#if MIDI_NUM_CABLES>3
  TUD_MIDI_JACKID_OUT_EMB(4),
#endif
#endif
#endif
};
```

AndrewCapon, 2022-11-11:
Thanks for that @rppicomidi 

There is a typo though which causes issues on a mac:
```
#if MIDI_NUM_CABLES>1
  TUD_MIDI_DESC_JACK(2),
#if MIDI_NUM_CABLES>2
  TUD_MIDI_DESC_JACK(3),
#if MIDI_NUM_CABLES>3
  TUD_MIDI_DESC_JACK(3),
```

should be

```
#if MIDI_NUM_CABLES>1
  TUD_MIDI_DESC_JACK(2),
#if MIDI_NUM_CABLES>2
  TUD_MIDI_DESC_JACK(3),
#if MIDI_NUM_CABLES>3
  TUD_MIDI_DESC_JACK(4),
```

rppicomidi, 2022-11-11:
Oops! Thanks @AndrewCapon. Sorry @DatanoiseTV if my typo messed you up.

rppicomidi, 2023-05-08:
I have created an example USB MIDI device with 2 MIDI In and 6 MIDI Out [here](https://github.com/rppicomidi/midi-multistream2usbdev). You may find the files `midi_device_multistream.h` helpful for building a USB descriptor with an
arbitrary number of virtual cables. It uses the Boost Preprocessor library to help.

hathach, 2023-05-09:
thank you @rppicomidi for the update, and the example, I think this could be closed now.

closed 2023-05-09 by hathach