hathach/tinyusb #1297: Mention cmake in getting started documentation

sharpie7, 2022-01-22:
According to [the getting started documentation](https://github.com/hathach/tinyusb/blob/master/docs/reference/getting_started.rst) the examples can be built using make, but some (all?) of them require cmake to be run first. I am not sure which config options should be supplied to make and which to cmake.

It would be nice if this page was updated to reflect the recommended way to use cmake to build the examples.

hathach, 2022-01-24:
cmake is required since the sdk of mcues such as rp2040 and esp32s2 uses CMake. Likewise some specific mcus such as sony spresense, dialog requires additional binary. This repo "assumes" you are already following those MCU setup guide from their vendor.

sharpie7, 2022-01-24:
Thanks. I have created a pull-request based on your comment. Feel free to close this issue in any case.

Thanks for the hard work on the project. It has amazing coverage.

closed 2022-03-08 by hathach