hathach/tinyusb #1324: Restriction on interface number/endpoint number

Huckies, 2022-02-14:
### Operating System

MacOS

### Board

ESP32-S3 Devkitm

### Firmware

Based on example/hid_composite_freertos and modified to have more interfaces.
Just some modifies to `usb_descriptors.c` and 

```
#define CFG_TUD_HID 3
#define CFG_TUD_HID_EP_BUFSIZE 64
```

```
uint8_t const desc_hid_report1[] =
    {
        TUD_HID_REPORT_DESC_KEYBOARD(HID_REPORT_ID(REPORT_ID_KEYBOARD))};
uint8_t const desc_hid_report2[] =
    {
        TUD_HID_REPORT_DESC_MOUSE(HID_REPORT_ID(REPORT_ID_MOUSE))};
uint8_t const desc_hid_report3[] =
    {
        TUD_HID_REPORT_DESC_CONSUMER(HID_REPORT_ID(REPORT_ID_CONSUMER_CONTROL)),
        TUD_HID_REPORT_DESC_GAMEPAD(HID_REPORT_ID(REPORT_ID_GAMEPAD))};

// Invoked when received GET HID REPORT DESCRIPTOR
// Application return pointer to descriptor
// Descriptor contents must exist long enough for transfer to complete
uint8_t const *tud_hid_descriptor_report_cb(uint8_t instance)
{
  printf("hid des called\r\n");
  if (instance == 0)
    return desc_hid_report1;
  else if (instance == 1)
    return desc_hid_report2;
  else if (instance == 2)
    return desc_hid_report3;
  return NULL;
}

//--------------------------------------------------------------------+
// Configuration Descriptor
//--------------------------------------------------------------------+

enum
{
  ITF_NUM_HID1,
  ITF_NUM_HID2,
  ITF_NUM_HID3,
  ITF_NUM_TOTAL
};

#define CONFIG_TOTAL_LEN (TUD_CONFIG_DESC_LEN + TUD_HID_INOUT_DESC_LEN + TUD_HID_INOUT_DESC_LEN + TUD_HID_INOUT_DESC_LEN)

// the first 8 stands for IN endpoint and 0 stands for OUT endpoint
// and the last number must not be repeated
#define EPNUM_HID1_1 0x81
#define EPNUM_HID1_2 0x02
#define EPNUM_HID2_1 0x83
#define EPNUM_HID2_2 0x04
#define EPNUM_HID3_1 0x85
#define EPNUM_HID3_2 0x06

uint8_t const desc_configuration[] =
    {
        // Config number, interface count, string index, total length, attribute, power in mA
        TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 4, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 500),

        // Interface number, string index, protocol, report descriptor len, EP In address, size & polling interval
        TUD_HID_INOUT_DESCRIPTOR(ITF_NUM_HID1, 5, HID_ITF_PROTOCOL_KEYBOARD, sizeof(desc_hid_report1), EPNUM_HID1_2, EPNUM_HID1_1, CFG_TUD_HID_EP_BUFSIZE, 1),
        TUD_HID_INOUT_DESCRIPTOR(ITF_NUM_HID2, 6, HID_ITF_PROTOCOL_NONE, sizeof(desc_hid_report2), EPNUM_HID2_2, EPNUM_HID2_1, CFG_TUD_HID_EP_BUFSIZE, 1),
        TUD_HID_INOUT_DESCRIPTOR(ITF_NUM_HID3, 7, HID_ITF_PROTOCOL_NONE, sizeof(desc_hid_report3), EPNUM_HID3_2, EPNUM_HID3_1, CFG_TUD_HID_EP_BUFSIZE, 1)};

// Invoked when received GET CONFIGURATION DESCRIPTOR
// Application return pointer to descriptor
// Descriptor contents must exist long enough for transfer to complete
uint8_t const *tud_descriptor_configuration_cb(uint8_t index)
{
  return desc_configuration;
}
```

### What happened ?

Device failed to get enumerated with the code above while giving up the the third OUT endpoint fix the problem. Like this:
```
uint8_t const desc_hid_report1[] =
    {
        TUD_HID_REPORT_DESC_KEYBOARD(HID_REPORT_ID(REPORT_ID_KEYBOARD))};
uint8_t const desc_hid_report2[] =
    {
        TUD_HID_REPORT_DESC_MOUSE(HID_REPORT_ID(REPORT_ID_MOUSE))};
uint8_t const desc_hid_report3[] =
    {
        TUD_HID_REPORT_DESC_CONSUMER(HID_REPORT_ID(REPORT_ID_CONSUMER_CONTROL)),
        TUD_HID_REPORT_DESC_GAMEPAD(HID_REPORT_ID(REPORT_ID_GAMEPAD))};

// Invoked when received GET HID REPORT DESCRIPTOR
// Application return pointer to descriptor
// Descriptor contents must exist long enough for transfer to complete
uint8_t const *tud_hid_descriptor_report_cb(uint8_t instance)
{
  if (instance == 0)
    return desc_hid_report1;
  else if (instance == 1)
    return desc_hid_report2;
  else if (instance == 2)
    return desc_hid_report3;
  return NULL;
}

//--------------------------------------------------------------------+
// Configuration Descriptor
//--------------------------------------------------------------------+

enum
{
  ITF_NUM_HID1,
  ITF_NUM_HID2,
  ITF_NUM_HID3,
  ITF_NUM_TOTAL
};

#define CONFIG_TOTAL_LEN (TUD_CONFIG_DESC_LEN + TUD_HID_INOUT_DESC_LEN + TUD_HID_INOUT_DESC_LEN + TUD_HID_DESC_LEN)

// the first 8 stands for IN endpoint and 0 stands for OUT endpoint
// and the last number must not be repeated
#define EPNUM_HID1_1 0x81
#define EPNUM_HID1_2 0x02
#define EPNUM_HID2_1 0x83
#define EPNUM_HID2_2 0x04
#define EPNUM_HID3_1 0x85

uint8_t const desc_configuration[] =
    {
        // Config number, interface count, string index, total length, attribute, power in mA
        TUD_CONFIG_DESCRIPTOR(1, ITF_NUM_TOTAL, 4, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 500),

        // Interface number, string index, protocol, report descriptor len, EP In address, size & polling interval
        TUD_HID_INOUT_DESCRIPTOR(ITF_NUM_HID1, 5, HID_ITF_PROTOCOL_KEYBOARD, sizeof(desc_hid_report1), EPNUM_HID1_2, EPNUM_HID1_1, CFG_TUD_HID_EP_BUFSIZE, 1),
        TUD_HID_INOUT_DESCRIPTOR(ITF_NUM_HID2, 6, HID_ITF_PROTOCOL_MOUSE, sizeof(desc_hid_report2), EPNUM_HID2_2, EPNUM_HID2_1, CFG_TUD_HID_EP_BUFSIZE, 1),
        TUD_HID_DESCRIPTOR(ITF_NUM_HID3, 7, HID_ITF_PROTOCOL_NONE, sizeof(desc_hid_report3), EPNUM_HID3_1, CFG_TUD_HID_EP_BUFSIZE, 5)};

// Invoked when received GET CONFIGURATION DESCRIPTOR
// Application return pointer to descriptor
// Descriptor contents must exist long enough for transfer to complete
uint8_t const *tud_descriptor_configuration_cb(uint8_t index)
{
  (void)index; // for multiple configurations
  return desc_configuration;
}
```

### How to reproduce ?

By modifying the `usb_descriptors.c` and `tusb_config.h` of example/hid_composite_freertos, the problem repeats every single time.

Huckies, 2022-02-15:
Further tests show that `EPNUM_HID` cannot exceed 0x86 or `tud_hid_descriptor_report_cb` will never be called. Is it a esp32s3 specific issue or the tusb component problem?

Huckies, 2022-02-15:
Further tests show that `EPNUM_HID` cannot exceed 0x85 or `tud_hid_descriptor_report_cb` will never be called.
Even a minor change of `hid_composite_freertos/src/usb_descriptors.c line.106` from `#define EPNUM_HID   0x81` to `#define EPNUM_HID   0x86` can crash the host enumeration.

 Is it a esp32s3 specific issue or the tusb component problem?

hathach, 2022-02-15:
> Alright I saw the restriction imposed on espressif chips described in `dcd_attr.h`. But still wonder why to limit it to six endpoints.

it is hardware limit on S2/S3

closed 2022-02-15 by hathach

Huckies, 2022-02-15:
> > Alright I saw the restriction imposed on espressif chips described in `dcd_attr.h`. But still wonder why to limit it to six endpoints.
> 
> it is hardware limit on S2/S3

But I do managed to get the esp32s3 show 6 endpoints(excluding endpoint 0)in USB Device Viewer
[image]

dhalbert, 2022-02-15:
The S2 and the S3 datasheets describe the endpoints as follows:

> Device Mode Features
> • Endpoint number 0 always present (bi-directional, consisting of EP0 IN and EP0 OUT)
> • Six additional endpoints (endpoint numbers 1 to 6), configurable as IN or OUT
> • Maximum of five IN endpoints concurrently active at any time (including EP0 IN)
> • All OUT endpoints share a single RX FIFO
> • Each IN endpoint has a dedicated TX FIFO

me-no-dev, 2022-02-15:
To clarify a bit what @hathach and @dhalbert said: Yes, you do have a total of 7 endpoints available, and they will show in the device viewer, but only 5 of them will have FIFOs to service the requests. The other 2 will not be able to communicate with the host.