hathach/tinyusb #1325: STM32 MSC host - HCD port

xxx-88, 2022-02-14:
**Is your feature request related to a problem? Please describe.**
There is supported MSC host classes for many MCUs but I couldn't find any host part for STM32 ARMs.
Is there any possibility of creating HCD drivers for STM32 for USB host applications?

**Describe the solution you'd like**
I need the HCD port implementation for STM32F7 (STM32F756) device to use as MSC host application. I've already tried my best to implement that port, but I've ended with infinite waiting loop for control block receive function. There is (probably) bad endpoint management in my port because I don't receiving any data from control block endpoint I've requested (trasfer looks OK, but there is no data). Same problem is represented in other code parts too. In some previous library operations, for example reading config descriptor - I'm successfully receiving correct "9 bytes of Configuration Descriptor" in "enum_get_device_desc_complete" function, But in next reception of EP0 I'm receiving zero bytes, and repeat.
My code is quite a big mess, so I'll try to tidy it up a little and I'll post it there when it will be ready to be readed. I hope that I'll find "source of this problem" when I'll be tidying that code, but. 

Anyway, is there any plans of that implementation, so I can use that instead of my messy code please?

Ryzee119, 2022-02-20:
I have made a STM32F7 HCD driver before using HAL_HCD though. Might be a starting point, you still to setup the USB gpio, clocks and the otg_irq etc.

https://gist.github.com/Ryzee119/5a1b74e98449fdf5ed24b3f361ff4e2c

xxx-88, 2022-02-20:
Thanks a lot for your input. I've looked it up but it's hard to understand it for me, because there is no comments and it looks (style) like my code - no comments but with attempt of something like self-descriptioning code by usefull naming patterns. 
Anyway, I've created CubeMX project with HAL layer for my F756 MCU and I've implemented TinyUSB and your HCD driver there. After succesfull port with no error build I've tried to run it, but unfortuanelly, it didn't work well. 
After attaching my flash drive, I see that it connects up, set some EP transfers, get correct ID etc etc, but it ends with assert failed.
There is some problem with SCSI tests, where it falls on one of two places depending on type of flash drive. First place is ASSERT at line 320 of MSC_HOST file (msch_xfer_cb CMD didn't bring back any data -> xfered_bytes = 0). 
Second place is line 421 -> function msch_set_config, with unknown reason for me, it seems like descriptor isn't fitting correctly?
Did you used your code with MSC, or just with other classes ? it seems like there is missing something needed for bulk transfers of MSC, but at this time, I don't see what it is. I'll try to compare your steps in code with mine to see what is differently handled. I didn't expect that I'll get too far in that, because it will be difficult (I'm not using HALs so it will be completely different). I'll try to get some systematic scheme from that. I believed it will be no too difficult task, because it's just low level layer and all hard work is already done in TinyUSB, but that was probably misstake.
Althrough thanks a lot for your input, I'll try to make my own asap and I'll write up there what was probmenatic in my case.

Ryzee119, 2022-02-20:
Yea admittedly, I hadnt tested bulk transfers so that may be an issue :)

> like my code - no comments but with attempt of something like self-descriptioning code by usefull naming patterns. 😁

xxx-88, 2022-03-14:
I've get myself a bit closer to function part of the code. I've continued in my own hcd driver for that STM and now it's cappable of getting full identification of the flash drive, its size and that's all for now. I'm bit confused, because all transfer seems correct but in attempt of reading from flash drive I'm receiving just missmash from the bus. There isnt any real data transfer nor control packets etc after identification.
I'm also done some refactorization of first 1/3 of the code so it is now partialy readable, I'll continue with next two parts and then I'll post it there. There is just one big problem, I'm currently in big time squeeze and my time for this part of the project is probably counted in negative numbers, so please be patient with that.

friesendrywall, 2022-07-28:
I spent some time with this port trying to understand the reason for the NAKS.  Is it possible that the issue could be in the MSC layer?  Here is a capture from a failure, I can't really understand the NAK.  Also included a good capture from elsewhere.
[usb-captures.zip](https://github.com/hathach/tinyusb/files/9213477/usb-captures.zip)
.

mcuenthusiast, 2023-07-27:
@xxx-88 is it possible to share the code for STM32 host MSC ? I would like to take a reference from that.

@Ryzee119 I used your code as well but I am stopping at the same place as shown by you in logs above. did you update the code after that and got that working?

xxx-88, 2023-07-27:
I'll look what I still have, this project was done with another usbdrv because there wasnt much time alocation for usb development. I!ll get to you as soon as I'll find it.

mcuenthusiast, 2023-07-31:
Hi @xxx-88 ,

did you find anything related to above query?

Thanks

xxx-88, 2023-07-31:
Not yet, I was home on my weekend, I'll be in office today so I'll look at that

xxx-88, 2023-08-01:
So, I've searched through my project folder and also checked my backup of this project, but I couldn't find any symbol related to tinyusb. I also looked through the old commits of that project, but I only found that tinyusb was added as a submodule. However, my porting part for the MSC Host was not there. I believe it was probably in some branch which got deleted after the successful implementation of another library.

mcuenthusiast, 2023-08-01:
Okay No, issues thanks for the checking.