hathach/tinyusb #1334: Adding XInput library

Huckies, 2022-02-20:
While the existing examples include several DirectInput ones, the emulation of recent xbox controllers is missing. 
Xbox controllers are very common today, many games don't accept the input from a DirectInput device, so I think it'll be useful.
Here is a repo that may help: https://github.com/FeralAI/MPG

Ryzee119, 2022-02-20:
I assume non standard class drivers are a non-goal of this project.
I have a xinput  USB host driver for TinyUSB, https://github.com/Ryzee119/tusb_xinput, but 'emulating them' as a usb device shouldnt be hard if someone hasnt already done this.

Ive done Original Xbox controllers before, however those arent 'X-input'.

Huckies, 2022-02-21:
@Ryzee119 Thank you for replying! I build my own driver refer to the repo I mentioned and finally get a device called 'Xbox 360 Controller for Windows' is that the goal? Because my first aim was to get a 'Xbox One Controller', am I getting a wrong descriptor or something else?

hathach, 2022-03-06:
Thank for your issue, to be honest, I am not familiar with xinput and xbox controller ( I dont have one to test it out), however, we can thing easier for user with predefined hid descriptor and structure report. I could made an example for ps4 controller since it is what I have. Hopefully that would pave the way for adding similar pr for xbox one. Will post ps4 here whenever it is available

Huckies, 2022-03-06:
Thank you for that:) I've done my implementation of an Xbox 360 controller since cracking an Xbox One controller might not be legal, although in a rough way. I could post my code later if you are interested, it'll be great to have it polished and be a part of the tinyusb stack.