hathach/tinyusb #1339: EHCI Interrupt Handler Clears Interrupt Status Register Incorrectly

yoyopod, 2022-02-22:
### Operating System

Others

### Board

Custom i.MX1064RT, running FreeRTOS and custom application

### Firmware

src/portable/ehci/ehci.c

hcd_int_handler() clears the interrupt status register in a way that can result in lost status and system hang.

I looked through the repo, and can see the error in commits going back at least nine years, through present day.

### What happened ?

hcd_int_handler() clears the interrupt status register using an |=, rather than a simple =.  The status register is already read two lines earlier, and the value read is then masked by the interrupt-enable settings to produce the pending status bits that are also enabled.  It is that pending status value that should be written back to the status register to clear the conditions that the handler is about to service.  Instead, by using |=, you read the status AGAIN, possibly introducing NEW status bits.  Any such status bits will end up being bit-wise ORed into the correct mask value, and then written back to the status register immediately, resulting in their being cleared unconditionally and blindly, whether they're enabled or not.  If they are enabled, they will never be seen by the handler.

### How to reproduce ?

I discovered this while running my target in host mode, reading an S-Record file from an MSC device and writing the content to Flash.  This was in a test loop.  After a variable number of iterations, the system would lock up.  I looked at a LOT of code and data before I found that a completion interrupt was being lost due to the status-clearing technique.

The window is so small, you wouldn't expect this to be a frequent error, yet I was seeing it quite readily.  During my testing, I inadvertently enlarged the window by placing some debug data-grabs between the two existing reads of the status register, and the lock-ups became more frequent.

hathach, 2022-02-23:
thank you for filing the issue and detail explanation. Since you could already reproduce the issue, would you mind making an PR for the changes? That's way we know for sure the issue is fixed and tested on your hardware. I could make the PR, but that would take time since I needs to find a way to reproduce the issue first.

yoyopod, 2022-02-23:
Will do.  It is a simple change, but I won't be able to get to it until tomorrow, CST.

yoyopod, 2022-02-24:
Okay, I didn't get to this today, but I did finish a major task today, so I will block out some time tomorrow for this (different tomorrow than the one yesterday, I know; call it second-tomorrow).

hathach, 2022-02-25:
take your time, not a rush at all

closed 2022-02-26 by hathach