hathach/tinyusb #1355: MCUboot support

beriberikix, 2022-02-26:
[MCUboot](https://www.mcuboot.com/) is a very capable bootloader that is being adopted across the industry and has many advanced features, but a special emphasis on security. While some of the ports have USB DFU support (I think just Zephyr atm,) there isn't a complete USB solution in the bootloader.

What inspired this idea was me looking into UF2 support for Zephyr, but being able to use the secure boot features of MCUboot (see mcu-tools/mcuboot#1309.) I'd imagine it could look similar to [TinyUF2](https://github.com/adafruit/tinyuf2). Note that this is different and probably unrelated to hathach/tinyusb#458 since an MCUboot integration would apply across (currently) 7 OSes.

Thoughts on the idea? Suggestions on an approach?

Thanks!

hathach, 2022-02-26:
> Thoughts on the idea? Suggestions on an approach?

this isn't a feature of a usb stack, this issue should be opened on the mcuboot repo for their maintainer to add the support tinyusb.

closed 2022-02-26 by hathach

beriberikix, 2022-02-26:
Makes sense! I opened mcu-tools/mcuboot#1309 moments ago.

Would it be more appropriate to convert this to a Discussion instead?

hathach, 2022-02-26:
I don't see any points of `adding bootloader support to a usb stack` feature unless the mcuboot has some specific needs for the stack. Top of my head it would be a special minimal for size optimization, you can open an ticket later with more specific requirement from the mcuboot team.