hathach/tinyusb #1364: How to control the Caps/Scroll/Num Lock LEDs in host mode?

SukkoPera, 2022-03-02:
First of all, thanks a lot for this project!

I would like the [cdc_msc_hid example](https://github.com/hathach/tinyusb/tree/master/examples/host/cdc_msc_hid) to be extended in order control the keyboard Caps/Scroll/Num Lock leds when a keyboard is connected in host mode.

Or if you can provide some quick code, I might integrate it into the example and provide a PR.

Thanks in any case :).

hathach, 2022-03-08:
this is not a feature, for question please use discussion

closed 2022-03-08 by hathach

SukkoPera, 2022-03-09:
Sorry, didn't know about the Discussion section.

Anyway I found what I was looking for [here](https://github.com/hathach/tinyusb/discussions/1191), I'm leaving this for reference.