hathach/tinyusb #1382: Missing iso packets in audio_test

kaspernyhus, 2022-03-07:
### Operating System

Linux

### Board

esp32s3

### Firmware

https://github.com/kaspernyhus/tinyusb/tree/audio_test_freertos/examples/device/audio_test_freertos

### What happened ?

I made a freertos version of the audio_test example.
Audio samples are coming through, but every other usb packet is missing data (see wireshark screenshots)
48kHz/16bit, 1 channel.

### How to reproduce ?

clone my fork
go to tinyusb/examples/device/audio_test_freertos
make BOARD=espressif_s3_devkitc all
flash

### Debug Log

I (24) boot: ESP-IDF v5.0-dev-1867-g5faf116d26-dirty 2nd stage bootloader
I (25) boot: compile time 21:20:35
I (25) boot: chip revision: 0
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 2MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08d28h ( 36136) map
I (100) esp_image: segment 1: paddr=00018d50 vaddr=3fc922d0 size=02814h ( 10260) load
I (105) esp_image: segment 2: paddr=0001b56c vaddr=40374000 size=04aach ( 19116) load
I (115) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1ed68h (126312) map
I (142) esp_image: segment 4: paddr=0003ed90 vaddr=40378aac size=09818h ( 38936) load
I (151) esp_image: segment 5: paddr=000485b0 vaddr=50000000 size=00010h (    16) load
I (157) boot: Loaded app from partition at offset 0x10000
I (157) boot: Disabling RNG early entropy source...
I (171) cpu_start: Pro cpu up.
I (171) cpu_start: Starting app cpu, entry point is 0x40375254
I (0) cpu_start: App cpu up.
I (185) cpu_start: Pro cpu start user code
I (185) cpu_start: cpu freq: 160000000 Hz
I (185) cpu_start: Application information:
I (188) cpu_start: Project name:     audio_test_freertos
I (194) cpu_start: App version:      0.13.0-66-g6248f648
I (200) cpu_start: Compile time:     Mar  7 2022 21:34:57
I (206) cpu_start: ELF file SHA256:  ca7e3fa86253da6a...
ELF file not found. You need to build & flash the project before running 'monitor', and the binary on the device must match the one in the build directory exactly. 
I (212) cpu_start: ESP-IDF:          v5.0-dev-1867-g5faf116d26-dirty
I (219) heap_init: Initializing. RAM available for dynamic allocation:
I (227) heap_init: At 3FC96C88 len 00049378 (292 KiB): D/IRAM
I (233) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (240) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (246) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (253) spi_flash: detected chip: gd
I (256) spi_flash: flash io: dio
W (260) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (274) sleep: Configure to isolate all GPIO pins in sleep state
I (280) sleep: Enable automatic switching of GPIO sleep configuration
I (288) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

### Screenshots

[image]
[image]
[image]

PanRe, 2022-03-08:
Hello,
sorry for the possibly stupid question but is there any error in the functionality of this example? Anyway, for starting, stopping, etc.  a few command messages from EP0 have to be transmitted. Have you found those in your wireshark? I saw only EP 0x81 (which is supposed to be so for data) in your pictures. Right now, i can not see any error!

PanRe, 2022-03-09:
Ah now i am on track, thanks for the clarification. Indeed that seems weired, there should be a packet scheduled every ms. What puzzels me right now is that the saw-tooth-test signal looks without gaps. The `audio_test` example sends increasing values from 0 - 2^16-1. So in the first packet, there sould be numbers ranging from 0 - 47, in the second from 48 - 95, etc. If every second packet is missing, there sould be jumps of 48 in that signal. There "might" be jumps in your plot but i am not sure. Could you verify that? You may use the python scipts as well.
A second possibility is, that the saw-tooth signal needs double the time to reach the maximum of 2^16-1. If everything is ok,  the saw-tooth should have a length of (2^16 / 48) / 1000 ~= 1.37 seconds. In your plot, that seems to be the case (2.7 + 1.37 = 4.07). Hence either there are jumps or wireshark is weired. ;) Let me know what you find!

kaspernyhus, 2022-03-09:
Here I have decoded three concecutive packets containing data 2ms appart:

```
raw data packet 1 [f0fff1fff2fff3fff4fff5fff6fff7fff8fff9fffafffbfffcfffdfffeffffff00000100020003000400050006000700080009000a000b000c000d000e000f0010001100120013001400150016001700180019001a001b001c001d001e001f00]
raw data packet 2 [20002100220023002400250026002700280029002a002b002c002d002e002f0030003100320033003400350036003700380039003a003b003c003d003e003f0040004100420043004400450046004700480049004a004b004c004d004e004f00]
raw data packet 3 [50005100520053005400550056005700580059005a005b005c005d005e005f0060006100620063006400650066006700680069006a006b006c006d006e006f0070007100720073007400750076007700780079007a007b007c007d007e007f00]
```

```
timestamp: 15.361629, [65520, 65521, 65522, 65523, 65524, 65525, 65526, 65527, 65528, 65529, 65530, 65531, 65532, 65533, 65534, 65535, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31]
timestamp: 15.363626, [32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79]
timestamp: 15.365626, [80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127]
```
so there are no gaps!

And the timing also seems correct: 1.365 seconds for one period

[image]

So the host somehow accepts a this halving of the bandwidth?

kaspernyhus, 2022-03-09:
```
MicNode
Manufacturer: PaniRCorp
Serial Number: 123456
Speed: 12Mb/s (full)
Bus:   3
Address:   6
USB Version:  2.00
Device Class: ef(misc )
Device Subclass: 02
Device Protocol: 01
Maximum Default Endpoint Size: 64
Number of Configurations: 1
Vendor Id: cafe
Product Id: 4010
Revision Number:  1.00

Config Number: 1
	Number of Interfaces: 2
	Attributes: 80
	MaxPower Needed: 100mA

	Interface Number: 0
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 01
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 1
		Name: snd-usb-audio
		Alternate Number: 0
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 0

	Interface Number: 1
		Name: snd-usb-audio
		Alternate Number: 1
		Class: 01(audio) 
		Sub Class: 02
		Protocol: 20
		Number of Endpoints: 1

			Endpoint Address: 81
			Direction: in
			Attribute: 5
			Type: Isoc
			Max Packet Size: 96
			Interval: 1ms
```

kaspernyhus, 2022-03-09:
btw I also sniffed the usb traffic using an ellisys USB Explorer with same results as in wireshark :)

PanRe, 2022-03-09:
Interesting. However, we may not trust the time scale of audicity (maybe, i don't know audicity this well). It could be, that audicity just uses the information, that 48 kHz is the sample frequency. Can you test this another way? Can you schedule a recording lasting exactly 2 * 1.365 seconds? Not the corresponding amount of bytes but really in time. If the received signal has two saw-tooths also the timing is ok. If the bandwidth was halfed, only one saw-tooth should be received (half the bytes).

Your decoded packets are 48 values each... every 2ms suggests the halfing of the bandwidth. I guess you will see only one saw-tooth. If you can confirm this, then we have an issue on the MCU (might be too slow in loading in data).

kaspernyhus, 2022-03-10:
I'm not sure how I would do that.
However I think you are correct in not trusting audacity!
I tried to do a recording on my macbook using reaper and the results are very different indeed! Are we seeing the consequences of the missing packets here maybe?

To have a reference I made a recording of the original audio_test example on an rp2040.

[image]
[image]

Also I noticed audacity running more "smoothly" with the rp2040 plugged in, although there the waveforms look identical - 1.365 seconds

kaspernyhus, 2022-03-10:
I had to reread your post earlier. I can confirm that the actual recording time is affected!
I tried to time what should be 4 * saw-tooth ≈ 5.5seconds:
the esp32s3 has 2
the pico has 4
!!

PanRe, 2022-03-10:
You may conduct this test by use of the [python script](https://github.com/hathach/tinyusb/blob/master/examples/device/audio_test/src/plot_audio_samples.py) in the audio test example. Set the duration to roughly 2.7 seconds.

PanRe, 2022-03-10:
Ok, then the MCU is too slow in loading the 48 samples or 96 bytes into the fifo and from the fifo into the usb buffer. Can you increase the clock frequency to speed the MCU up? Another possibility would be to reduce the workload in the example... but this would be a bit of work. One idea would be to trick the fifo into thinking something is contained in the fifo (by use of `tu_fifo_advance_read_pointer()`). Another one would be, to use a DMA to load data into the fifo.

PanRe, 2022-03-10:
Or the rtos intervenes somehow and keeps the MCU too busy.

kaspernyhus, 2022-03-10:
[image]
recording time ≈5.5 in both cases

kaspernyhus, 2022-03-10:
The mcu is running at full speed 240MHz

kaspernyhus, 2022-03-10:
[image]
Not sure this is of much value, but it shows the tud_task (second from the bottom) is scheduled to run with 2ms interval triggered by an interrupt (labeled "ETH_MAC" here for some reason)
I also notice the tud_task being interrupted by the same interrupt before finishing.

PanRe, 2022-03-10:
Ok that seems to be some configuration error regarding rtos. Performance is more than enough available ^^ and the USB protocol polls every ms but only every second triggers an interrupt. Unfortunately, i am not that experienced with freeRTOS. @hathach do you have an idea on this?

kaspernyhus, 2022-03-11:
for the record I'm finding the same exact issues on the esp32s2 as the esp32s3

LiTongXue98, 2022-03-12:
[image]
I found the same problem when I used the IO flip as a marker in the _dcd_int_handler and found that the RTOS generated only one interrupt request for every two DATA_IN requests on the physical line

hathach, 2022-03-12:
sorry guys, I am currently busy with other works and couldn't conduct any test and/or look into this. Though keep posting what you find, I will join in if I could spot anything suspicious

kaspernyhus, 2022-03-14:
If it is indeed an rtos error, I'm not sure where to start looking. If we don't get the interrupt because the system is blocking, how can we test for that? Any ideas?

PanRe, 2022-03-14:
Die you try the example without rtos?

kaspernyhus, 2022-03-15:
I'm not sure that I can use the esp32 without freertos? I think the second cpu can be configured to run without freertos, but I wouldn't know how to configure that using the tinyusb examples..

kaspernyhus, 2022-03-15:
I have been digging around in the dcd_esp32sx.c file trying to identify possible bottlenecks.
In the interrupt handler `_dcd_int_handler` I taken some timestamps to see if we got stuck somewhere - one at the top and one at the end:
__________________ name ______timestamp (us) ___ deltaT
```
I (8154) : 706: _dcd top          935207        1
I (8164) : 707: _dcd end          935213        6
I (8174) : 708: _dcd top          935222        9
I (8184) : 709: _dcd end          935226        4
I (8194) : 710: _dcd top          935237        11
I (8204) : 711: _dcd end          935244        7
I (8214) : 712: _dcd top          935272        28
I (8224) : 713: _dcd end          935274        2
I (8234) : 714: _dcd top          935276        2
I (8244) : 715: _dcd end          935278        2
I (8254) : 716: _dcd top          935279        1
I (8264) : 717: _dcd end          935285        6
I (8274) : 718: _dcd top          935621        336
I (8284) : 719: _dcd end          935623        2
I (8294) : 720: _dcd top          935627        4
I (8304) : 721: _dcd end          935629        2
I (8314) : 722: _dcd top          935631        2
I (8324) : 723: _dcd end          935637        6
I (8334) : 724: _dcd top          935682        45
I (8344) : 725: _dcd end          935689        7
I (8354) : 726: _dcd top          939841        4152
I (8364) : 727: _dcd end          939847        6
I (8374) : 728: _dcd top          939872        25
I (8384) : 729: _dcd end          939880        8
I (8394) : 730: _dcd top          941969        2089
I (8404) : 731: _dcd end          941975        6
I (8414) : 732: _dcd top          941992        17
I (8424) : 733: _dcd end          942000        8
I (8434) : 734: _dcd top          943969        1969
I (8444) : 735: _dcd end          943975        6
I (8454) : 736: _dcd top          943992        17
I (8464) : 737: _dcd end          944000        8
I (8474) : 738: _dcd top          945969        1969
I (8484) : 739: _dcd end          945975        6
I (8494) : 740: _dcd top          945992        17
I (8504) : 741: _dcd end          946000        8
I (8514) : 742: _dcd top          947969        1969
I (8524) : 743: _dcd end          947975        6
I (8534) : 744: _dcd top          947992        17
I (8544) : 745: _dcd end          948000        8
I (8554) : 746: _dcd top          949969        1969
I (8564) : 747: _dcd end          949975        6
I (8574) : 748: _dcd top          949992        17
I (8584) : 749: _dcd end          950000        8
I (8594) : 750: _dcd top          951969        1969
I (8604) : 751: _dcd end          951975        6
I (8614) : 752: _dcd top          951992        17
I (8624) : 753: _dcd end          952000        8
I (8634) : 754: _dcd top          953968        1968
I (8644) : 755: _dcd end          953975        7
I (8654) : 756: _dcd top          953992        17
I (8664) : 757: _dcd end          954000        8
I (8674) : 758: _dcd top          955968        1968
I (8684) : 759: _dcd end          955975        7
I (8694) : 760: _dcd top          955992        17
I (8704) : 761: _dcd end          956000        8
I (8714) : 762: _dcd top          957968        1968
I (8724) : 763: _dcd end          957975        7
I (8734) : 764: _dcd top          957991        16
I (8744) : 765: _dcd end          958000        9
I (8754) : 766: _dcd top          959968        1968
I (8764) : 767: _dcd end          959975        7
I (8774) : 768: _dcd top          959992        17
I (8784) : 769: _dcd end          960000        8
I (8794) : 770: _dcd top          961968        1968
I (8804) : 771: _dcd end          961975        7
I (8814) : 772: _dcd top          961991        16
I (8824) : 773: _dcd end          962000        9
I (8834) : 774: _dcd top          963968        1968
I (8844) : 775: _dcd end          963975        7
I (8854) : 776: _dcd top          963991        16
I (8864) : 777: _dcd end          964000        9
```
This is started right before the enumeration process ends and as soon as the iso IN data transfers start, we see the pattern with a ~2ms delay. The delay is happening **outside** of the `_dcd_int_hnadler` !
This might be showing that we lose an interrupt? And might suggest that it is not an rtos problem?

Could it be a missed clearing of an interrupt flag or something like that? In `handle_epin_ints` ?

/K

PanRe, 2022-03-15:
Can you somehow measure the CPU load? If it is very high the mcu might Hang in some buggy loop, if the load is low there might be a missing interrupt flag... The error might be in the dcd ISO related transfer functions...

PanRe, 2022-03-15:
Can you profile how long the write functions for loading the data into the buffers last?

HiFiPhile, 2022-03-16:
> Mh i see... @HiFiPhile have you any experience with rtos and uac?

I didn't use RTOS on SAME70 & LPC4322. I doubt there is some issue in esp-idf since personally I'm not impressed by it's robustness. (5-6 issues submitted in 2 weeks and more minor glitches)

kaspernyhus, 2022-03-16:
> Can you somehow measure the CPU load? If it is very high the mcu might Hang in some buggy loop, if the load is low there might be a missing interrupt flag... The error might be in the dcd ISO related transfer functions...>
My tracing shows that the cpu running the tud_task is mostly idle - the other cpu has only systicks. So cpu load is assumed to be quite low, right?

kaspernyhus, 2022-03-16:
> Can you profile how long the write functions for loading the data into the buffers last?>

`tud_audio_write` ?

kaspernyhus, 2022-03-16:
> > Mh i see... @HiFiPhile have you any experience with rtos and uac?
> 
> I didn't use RTOS on SAME70 & LPC4322. I doubt there is some issue in esp-idf since personally I'm not impressed by it's robustness. (5-6 issues submitted in 2 weeks and more minor glitches)

I have contacted Espressif as well, and they just replyed they will provide me with a test example later this week !

LiTongXue98, 2022-03-16:
> > Can you profile how long the write functions for loading the data into the buffers last?>
> 
> `tud_audio_write` ?

I have measured the time between `handle_epin_ints()` and data filling in the buffer（`dcd_edpt_xfer`），It's much less than 1ms

PanRe, 2022-03-18:
Could you monitor the corresponding interrupt enable registers sucht that you can say if it gets disabled at some point? Maybe by placing a breakpoint which hits when the register flag gets changed?

LiTongXue98, 2022-03-21:
I'm sorry, I didn't find at [https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf](url) relevant introduction of the usb register, but I did an experiment, in the hid device 1ms to send out a data, is can produce the corresponding interrupt signal ,I wonder if it has something to do with endpoint initialization or configuration, but I don't know how to verify this

PanRe, 2022-03-21:
Look info the device controler driver file tinyusb/src/portable/espressif/esp32sx/dcd_esp32sx.c. The registers are set within dcd_edpt_open().Check if the registers are correctly set for the ISO endpoint!

PanRe, 2022-03-23:
Well, according to the [manual ](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) on page 602, the interrupt for the IN EP is always set (bit 18). Unfortunately, the manual does not tell the meaning of the other bits. Can you also check for the masking register `USB_GINTSTS_REG` and `SB_GLBLINTRMSK`. In this registers you can check if the interrupt was masked out. Furthermore, please check `USB_DAINT_REG` to check the number of the EP interrupt and finally the corresponding `USB_DIEPINTn_REG` for the interrupt source. These should indicate the endpoint 0x81 i guess.

kaspernyhus, 2022-03-24:
[image]
Yes, this part of the documentation leaves a bit to be desired.

Here is another run:
```
I (5281) : -------------------------------------------------
I (5281) : USB0.gintsts:         0x00040020      32
I (5291) : USB0.gintmsk:         0x208C3C16      1
I (5291) : int_status  :         0x00040000      1
I (5301) : daint       :         0x00000001      2
I (5301) : daintmsk    :         0x00010003      1
I (5311) : dsts        :         0x00425D06      2
I (5311) : -------------------------------------------------
I (5321) : USB0.gintsts:         0x00000030      30
I (5321) : USB0.gintmsk:         0x208C3C16      1
I (5331) : int_status  :         0x00000010      2
I (5331) : daint       :         0x00000000      1
I (5341) : daintmsk    :         0x00010003      2
I (5341) : dsts        :         0x00425D06      1
I (5351) : -------------------------------------------------
I (5351) : USB0.gintsts:         0x00000030      4
I (5361) : USB0.gintmsk:         0x208C3C16      2
I (5361) : int_status  :         0x00000010      1
I (5371) : daint       :         0x00000000      1
I (5371) : daintmsk    :         0x00010003      2
I (5381) : dsts        :         0x00425D06      1
I (5381) : -------------------------------------------------
I (5391) : USB0.gintsts:         0x00080020      4
I (5391) : USB0.gintmsk:         0x208C3C16      2
I (5401) : int_status  :         0x00080000      1
I (5401) : daint       :         0x00010000      2
I (5411) : daintmsk    :         0x00010003      1
I (5411) : dsts        :         0x00425D06      1
I (5421) : -------------------------------------------------
I (5421) : USB0.gintsts:         0x00000038      424
I (5431) : USB0.gintmsk:         0x208C3C16      2
I (5431) : int_status  :         0x00000010      1
I (5441) : daint       :         0x00000000      2
I (5441) : daintmsk    :         0x00010003      1
I (5451) : dsts        :         0x00425E06      1
I (5451) : -------------------------------------------------
I (5461) : USB0.gintsts:         0x00000030      5
I (5461) : USB0.gintmsk:         0x208C3C16      1
I (5471) : int_status  :         0x00000010      2
I (5471) : daint       :         0x00000000      1
I (5481) : daintmsk    :         0x00010003      1
I (5481) : dsts        :         0x00425E06      2
I (5491) : -------------------------------------------------
I (5491) : USB0.gintsts:         0x00080020      4
I (5501) : USB0.gintmsk:         0x208C3C16      1
I (5501) : int_status  :         0x00080000      2
I (5511) : daint       :         0x00010000      1
I (5511) : daintmsk    :         0x00010003      1
I (5521) : dsts        :         0x00825E06      2
I (5521) : -------------------------------------------------
I (5531) : USB0.gintsts:         0x00040020      58
I (5531) : USB0.gintmsk:         0x208C3C16      2
I (5541) : int_status  :         0x00040000      1
I (5541) : daint       :         0x00000001      1
I (5551) : daintmsk    :         0x00010003      2
I (5551) : dsts        :         0x00425E06      1
I (5551) : -------------------------------------------------
I (5561) : USB0.gintsts:         0x00148028      5552
I (5571) : USB0.gintmsk:         0x208C3C16      1
I (5571) : int_status  :         0x00040000      2
I (5581) : daint       :         0x00000002      1
I (5581) : daintmsk    :         0x00010003      1
I (5581) : dsts        :         0x00426406      2
I (5591) : -------------------------------------------------
I (5601) : USB0.gintsts:         0x00040020      41
[... 106 lines cut ...]
    volatile uint32_t dvbusdis;             // 0x0828 Device VBUS discharge Register
    volatile uint32_t dvbuspulse;           // 0x082c Device VBUS Pulse Register
    volatile uint32_t dthrctl;              // 0x0830 Device Thresholding control register (Read/Write)
    volatile uint32_t dtknqr4_fifoemptymsk; // 0x0834 Device IN Endpoint FIFO Empty Interrupt Mask register
    uint32_t reserved_0x0838_0x0900[50];    // 0x0838 to 0x0900
    // Input Endpoints
    usb_in_endpoint_t in_ep_reg[USB_IN_EP_NUM];     // 0x0900 to 0x09e0 IN EP registers
    uint32_t reserved_0x09e0_0x0b00[72];    // 0x09e0 to 0x0b00
    // Output Endpoints
    usb_out_endpoint_t out_ep_reg[USB_OUT_EP_NUM];  // 0x0b00 to 0x0be0 OUT EP registers
    uint32_t reserved_0x0be0_0x0d00[72];    // 0x0be0 to 0x0d00
    uint32_t reserved_0x0d00_0x0e00[64];    // 0x0d00 to 0x0e00
    /**
     * Power Control and direct FIFO access
     */
    uint32_t pcgctrl;                       // 0x0e00 Power and Clock Gating Control Register
    uint32_t reserved_0x0e04;               // 0x0e04
    uint8_t reserved8[0x1000 - 0xe08];      // 0x0d00 to 0x1000
    uint32_t fifo[16][0x400];               // 0x1000 to 0x2000 Device EP i/Host Channel i FIFO
    uint8_t reserved0x11000[0x20000 - 0x11000];
    uint32_t dbg_fifo[0x20000];             // 0x2000 to 0x22000 Direct Access to Data FIFO RAM for Debugging
} usb_dev_t;

extern usb_dev_t USB0;
```

kaspernyhus, 2022-03-30:
Is this essentially a job for Espressif to fix, since it seems to be quite deep down in the handling of interrupts?

PanRe, 2022-03-30:
> when the 2ms delay starts USB0.in_ep_reg[1].diepint is oscillating between 0x80/0x81 - what does that mean??
I guess the interrupt flag is not cleared. The error may be in the function `handle_epin_ints()` as there these flags are cleared. Or this function does not get called every second time?!

> Is this essentially a job for Espressif to fix, since it seems to be quite deep down in the handling of interrupts?
I guess yes, the device controler `dcd_esp32sx.c` was also written by Espressif Systems.

rheckhau, 2022-09-01:
Is there any update on this?

kaspernyhus, 2022-09-01:
> Is there any update on this?

unfortunately there is not. However I'm in contact with Espressif to try to solve this. I will keep you posted

tore-espressif, 2022-09-26:
@kaspernyhus That's really an interesting problem. Could you please test your setup with Windows host? And with ESP32-S2, if you have one? (just to rule out any multi-CPU issues).

tore-espressif, 2022-10-17:
@kaspernyhus @PanRe I'm sorry this has sat around for an extraordinary long time without further feedback.

The current DCD implementation in `dcd_esp32sx.c` does not fully support isochronous transfers. Could you please try applying the attached patch, and let me know whether it works for you?

```
diff --git a/src/portable/espressif/esp32sx/dcd_esp32sx.c b/src/portable/espressif/esp32sx/dcd_esp32sx.c
index a9c31132..dee2c855 100644
--- a/src/portable/espressif/esp32sx/dcd_esp32sx.c
+++ b/src/portable/espressif/esp32sx/dcd_esp32sx.c
@@ -60,6 +60,7 @@ typedef struct {
     uint16_t queued_len;
     uint16_t max_size;
     bool short_packet;
+    uint8_t interval;
 } xfer_ctl_t;
 
 static const char *TAG = "TUSB:DCD";
@@ -260,6 +261,7 @@ bool dcd_edpt_open(uint8_t rhport, tusb_desc_endpoint_t const *desc_edpt)
 
   xfer_ctl_t *xfer = XFER_CTL_BASE(epnum, dir);
   xfer->max_size = tu_edpt_packet_size(desc_edpt);
+  xfer->interval = desc_edpt->bInterval;
 
   if (dir == TUSB_DIR_OUT) {
     out_ep[epnum].doepctl |= USB_USBACTEP1_M |
@@ -372,6 +374,14 @@ bool dcd_edpt_xfer(uint8_t rhport, uint8_t ep_addr, uint8_t *buffer, uint16_t to
     USB0.in_ep_reg[epnum].dieptsiz = (num_packets << USB_D_PKTCNT0_S) | total_bytes;
     USB0.in_ep_reg[epnum].diepctl |= USB_D_EPENA1_M | USB_D_CNAK1_M; // Enable | CNAK
 
+    // For ISO endpoint with interval=1 set correct DATA0/DATA1 bit for next frame
+    if ((USB0.in_ep_reg[epnum].diepctl & USB_D_EPTYPE0_M) == (1 << USB_D_EPTYPE1_S) && xfer->interval == 1)
+    {
+      // Take odd/even bit from frame counter.
+      uint32_t const odd_frame_now = (USB0.dsts & (1u << USB_SOFFN_S));
+      USB0.in_ep_reg[epnum].diepctl |= (odd_frame_now ? USB_DI_SETD0PID1 : USB_DI_SETD1PID1);
+    }
+
     // Enable fifo empty interrupt only if there are something to put in the fifo.
     if(total_bytes != 0) {
       USB0.dtknqr4_fifoemptymsk |= (1 << epnum);
@@ -380,6 +390,14 @@ bool dcd_edpt_xfer(uint8_t rhport, uint8_t ep_addr, uint8_t *buffer, uint16_t to
     // Each complete packet for OUT xfers triggers XFRC.
     USB0.out_ep_reg[epnum].doeptsiz |= USB_PKTCNT0_M | ((xfer->max_size & USB_XFERSIZE0_V) << USB_XFERSIZE0_S);
     USB0.out_ep_reg[epnum].doepctl  |= USB_EPENA0_M | USB_CNAK0_M;
+
+    // For ISO endpoint with interval=1 set correct DATA0/DATA1 bit for next frame
+    if ((USB0.out_ep_reg[epnum].doepctl & USB_D_EPTYPE0_M) == (1 << USB_D_EPTYPE1_S) && xfer->interval == 1)
+    {
+      // Take odd/even bit from frame counter.
+      uint32_t const odd_frame_now = (USB0.dsts & (1u << USB_SOFFN_S));
+      USB0.out_ep_reg[epnum].doepctl |= (odd_frame_now ? USB_DO_SETD0PID1 : USB_DO_SETD1PID1);
+    }
   }
   return true;
 }
```
(GH won't allow me to attach .patch file)
I'll submit a PR then, thank you!

kaspernyhus, 2022-10-18:
[image]
I'm seeing payload data in every packet now!!
Fantastic!

tore-espressif, 2022-10-18:
> I'm seeing payload data in every packet now!!
Fantastic!

Good to hear that! Thank you for your patience

PanRe, 2022-10-18:
Very nice! Thanks a lot!!

closed 2022-10-21 by hathach