hathach/tinyusb #1411: Docs build failling

perigoso, 2022-03-30:
Something happened and the docs builds are now failling.

 I don't think it's related to anything on our side, probably just an upstream update that broke something, let's wait to see if it fixes itself.

```
/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/bin/python -m sphinx -T -E -b html -d _build/doctrees -D language=en . _build/html
Traceback (most recent call last):
  File "/home/docs/.pyenv/versions/3.8.6/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/docs/.pyenv/versions/3.8.6/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/lib/python3.8/site-packages/sphinx/__main__.py", line 13, in <module>
    from sphinx.cmd.build import main
  File "/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/lib/python3.8/site-packages/sphinx/cmd/build.py", line 25, in <module>
    from sphinx.application import Sphinx
  File "/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/lib/python3.8/site-packages/sphinx/application.py", line 43, in <module>
    from sphinx.registry import SphinxComponentRegistry
  File "/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/lib/python3.8/site-packages/sphinx/registry.py", line 24, in <module>
    from sphinx.builders import Builder
  File "/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/lib/python3.8/site-packages/sphinx/builders/__init__.py", line 26, in <module>
    from sphinx.util import import_object, logging, progress_message, rst, status_iterator
  File "/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/lib/python3.8/site-packages/sphinx/util/rst.py", line 21, in <module>
    from jinja2 import Environment, environmentfilter
ImportError: cannot import name 'environmentfilter' from 'jinja2' (/home/docs/checkouts/readthedocs.org/user_builds/tinyusb/envs/1381/lib/python3.8/site-packages/jinja2/__init__.py)
```

hathach, 2022-03-30:
I will trigger rebuild to see if it would fix.

perigoso, 2022-03-30:
`jinja2` updated to 3.1.0 / 3.1.1 a couple days ago, which explains the breakage, I think we can force a version, I will try it out.

https://pypi.org/project/Jinja2/#history

PR #1413 (2022-03-30): Fix docs
`jinja2` updated to `3.1.0` / `3.1.1` which broke the docs, fix it to a known working version

Fixes #1411 

~~Testing, do not merge.~~