hathach/tinyusb #1414: dcd_pic32mz.c: line 565

zj2004, 2022-03-31:
### Operating System

Windows 10

### Board

new MCU port from pic32mz

### Firmware

cdc_msc demo

### What happened ?

the second CBW , bulk out EP got 62 bytes

### How to reproduce ?

dcd_pic32mz.c: line 565
after xfer_complete(xfer, XFER_RESULT_SUCCESS, true);
add xfer->transferred = 0;

hathach, 2022-04-01:
please be specific on the MCU, what is your exact MCU and board. Also provide your LOG=2 as attached txt as well.

kasjer, 2022-05-23:
@zj2004 could you please check if #1451 fixes problem you described?

zj2004, 2022-08-29:
> @zj2004 could you please check if #1451 fixes problem you described?

yes ，it fixes. 
the problem appear at LOG=2
as you say “For PIC32 OUT packets that were sent from host and not handled fast enough could lead to memory corruption.”

closed 2022-08-29 by zj2004