hathach/tinyusb #1417: Control CTS and DSR lines

reiser4, 2022-04-03:
**Is your feature request related to a problem? Please describe.**
Hi, I'm trying to use an rp2040 to create an hardware project which uses CTS and DSR lines.
I see the current CDC code doesn't support them.

**Describe the solution you'd like**
I'd like to know how to implement CTS and DSR control in tinyusb.
I'm also able to read DTR, but RTS is not update. Host is Windows 11.
Thanks

hathach, 2022-04-04:
to speed up the support, do you know
1. on PC host, which API is used to get cts and dsr, preferably with python script 
2. And which CDC-ACM request that host issue when (1) is called to get CTS and DSR from device

reiser4, 2022-04-04:
Hi, I don't have those information. I'm using comtestserial to interact with the CDC port. I had success moving DSR with this function:
```

void set_dsr(uint8_t itf, bool value) {

    cdcd_interface_t* p_cdc = &_cdcd_itf[itf];

    uint8_t packet[10];

    packet[0] = 0xA1;                        //   bmRequestType
    packet[1] = CDC_NOTIF_SERIAL_STATE;                //   bNotification
    packet[2] = 0x00;                        //   wValue
    packet[3] = 0x00;
    packet[4] = 0x00;                        //   wIndex
    packet[5] = 0x00;
    packet[6] = 0x02;                        //   wLength
    packet[7] = 0x00;
    packet[8] = value ?  0x02 : 0x00; 
    packet[9] = 0x00;

    usbd_edpt_xfer(TUD_OPT_RHPORT, p_cdc->ep_notif, packet, 10);

}
```

with this code, CTS is always High. Before calling this function, I see it as low.
Thanks

hathach, 2022-04-04:
thanks for info, though it doesn't answer the 1st question, which script/API you run on windows to get CTS DSR state

reiser4, 2022-04-04:
On Windows 10/11 I open this software to read the serial port state:
https://www.microridge.com/comtestserial.htm

hathach, 2022-04-04:
Thanks, I don't use windows, but I could give it a try on my VM when I have the time to work on this later on.

reiser4, 2022-04-04:
Thanks, I can try something on Linux and OSX to verify my code, which platform are you working on?

hathach, 2022-04-04:
I am on Linux

just-jason, 2022-08-26:
The pin state can be checked with `pySerial`

Something along these lines

```
import serial

tty = serial.Serial(port = '/dev/AMA0', baudrate = baud, parity = serial.PARITY_NONE, stopbits = serial.STOPBITS_ONE, bytesize = serial.EIGHTBITS, xonxoff = True, timeout = 2.0)

if (tty.getCTS()):
   print("CTS")
```
You can also use [coolterm](https://freeware.the-meiers.org/) under Linux for a checking the state of the cts/dsr pins with a nice GUI.

AKASHWORLD614, 2023-08-05:
Hi I want to implement the same of having CTS flow control with pre existing RTS. Have you got it to work if yes can u share them please.

asesidaa, 2024-05-09:
> Hi, I don't have those information. I'm using comtestserial to interact with the CDC port. I had success moving DSR with this function:
>
> ```
>
> void set_dsr(uint8_t itf, bool value) { [...]

Are you able to figure out how to change CTS to low after calling this

HiFiPhile, 2024-10-30:
You can try https://github.com/hathach/tinyusb/pull/2593

closed 2025-08-07 by hathach