hathach/tinyusb #1427: TinyUSB Hangs on Set Configuration waiting for usbh_lock()

alimansfield2016, 2022-04-14:
### Operating System

Linux

### Board

Raspberry Pi Pico

### Firmware

Custom CDC class extension to support notify (similar to HID class reports) and mount/umount callbacks. The hardware is a Telit LE910C1 LTE module that interfaces via "CDC" interfaces over USB, although they aren't quite CDC, they are 99% (class is vendor and no data interface, just two extra endpoints instead.
[telit_driver.zip](https://github.com/hathach/tinyusb/files/8489957/telit_driver.zip)
)

### What happened ?

During the ENUM_SET_CONFIG phase of process_enumeration(), tuh_configuration_set() hangs while waiting for usbh_lock() mutex. This is running using a single core
. I have already confirmed that the mutex has been freed and not acquired elsewhere.
Having done some digging, it turns out that the _usbh_mutex is being overwritten, currently not sure what by.

### How to reproduce ?

I don't really know what to put in here, upload to pico and plug in device? It only seems to happen with this driver, I've checked the HID (in the same program) and it works fine.

### Debug Log as txt file

[telit_tinyusb_log.txt](https://github.com/hathach/tinyusb/files/8489969/telit_tinyusb_log.txt)

alimansfield2016, 2022-04-14:
Turns out the USB device has more endpoints than ep2drv had reserved and there is no ASSERT() to check this! Which in turn caused an out of bounds write!

hathach, 2022-04-17:
Ah right, 16 is probably a better default value, this is mainly for mcu with lack of memory. I will change that and add assert as well.