hathach/tinyusb #1430: Corrupted data in Interrupt OUT pipe

Huckies, 2022-04-16:
### Operating System

MacOS

### Board

ESP32-S3-DevKitC-1

### Firmware

Custom, with only few changes.

### What happened ?

Every single hid example seems to declare no Interrupt OUT pipe so devices like keyboards retrieve host control data from the control pipe. I made it to have an Interrupt OUT pipe and the host PC do leave the control pipe sending instructions (NumLock, CapsLock etc.)

However when I try to retrieve these using ```tud_hid_set_report_cb()```, the data is not as expected, ```instance``` ```report_id``` ```report_type``` are constant zeros while ```buffer``` works, storing the whole message from host

### How to reproduce ?

Just declare an Interrupt OUT endpoint in descriptor and flash it, this should be a stack issue and is irrelevant to chip.

hathach, 2022-04-17:
Thanks for the report, though the bug should be exactly reproduced, can you 
1. Provide link to modified example that can be compiled to flash to reproduce the issue
2. Which steps that you do on host pc to cause the issue
3. “The data is not as expected” is not analyzable, can you provide the expected vs the actual data (contents of those variables in the callback)
4. Attach debug log as txt file for analysis

Huckies, 2022-04-17:
1. [https://github.com/Huckies/tinyusb-bugreport](url) I do the minimal modification on example ```hid_boot_interface``` to reproduce this. All I did was adding an Interrupt OUT endpoint to the keyboard interface and left everything else 'as is'.
2. Press 'NumLock', 'CapsLock' or 'ScrollLock'
3. I expected to have ```instance=0 report_id=0 report_type=HID_REPORT_TYPE_OUTPUT(aka 2)``` in callback ```void tud_hid_set_report_cb(uint8_t instance, uint8_t report_id, hid_report_type_t report_type, uint8_t const *buffer, uint16_t bufsize)```, but eventually I get ```instance=0 report_id=0 report_type=0```
4. I must be dumb to fail to find the debug log after turning on CFG_TUSB_DEBUG.

Furthermore, if a report id is used let's say keyboard report is in interface 0 with report id 1, after pressing the CapsLock host should transfer ```01 02``` in which ```01``` is the destination report id and ```02```means turning on the CapsLock LED, I expect ```report_id``` to be 0x01 and ```buffer[0]``` presenting 0x02. However, I constantly retrieve a 0 in ```report_id``` and observe ```buffer[0]=0x01``` ```buffer[1]=0x02``` which leaves the raw report totally unparsed.

hathach, 2022-04-18:
If you read the code, the report id as invalid is intended behavior

https://github.com/hathach/tinyusb/blob/9c8c5c1c53214b8954bb28c33a496b5b423d5ecc/src/class/hid/hid_device.c#L408-L412

The reason is report parser is not implemented for hid device and it cannot tell if report id is used and part of the output report or not. And present report as raw as it is. You have to do your own parsing (read report id, and byte++). 

duplicated to #1095, I will add a simple hid parser later on next time I work with hid class. 

PS: for debug log, try `make BOARD=espressif_ LOG=2 all flash` you may need to full clean it first. Regarding

closed 2022-04-18 by hathach

Huckies, 2022-04-18:
I get it.
But a parser is for host, isn't it? The device claims its report using a report descriptor and a parser allows the host to read the report correctly. Since we are implementing a device, we claim our own report descriptor so adding 'report id version' of  ```TUD_HID_DESCRIPTOR``` and ```TUD_HID_INOUT_DESCRIPTOR``` macros should work. After all, you only need to tell the stack if it's using reportids

hathach, 2022-04-19:
Report id is part of report descriptor, we will need a parser to detect whether it is in use or not. I don't quite understand your idea, could you mind making an pr or codes around your suggestion to demonstrate it ?

Huckies, 2022-04-20:
Alright I watched through the code again and realized that the macro is only for generating some hex numbers, so no operation space. Please go ahead with the parser.

hathach, 2022-04-20:
FYI, Those serves as template, and aren't required to use. User can dump a byte array from existing device and feed that byte array to the stack as simulation.