hathach/tinyusb #1448: Please add .DS_store files to .gitignore

rppicomidi, 2022-04-27:
**Is your feature request related to a problem? Please describe.**
I just started using a Mac as a development platform for rp2040. I find .DS_Store files show up as untracked files. It would be too easy to commit them by accident.

**Describe the solution you'd like**
This Apple developer forum [discussion](https://developer.apple.com/forums/thread/30532) suggests to add the following to the .gitignore. I can create a pull request for this if you would like.
```
# macOS-specific files
.DS_Store
.Trashes
Thumbs.db
```

hathach, 2022-04-28:
sure, please make an PR for this, I don't use mac to verify the change.

closed 2022-04-29 by rppicomidi

hathach, 2022-04-30:
No problems at all, thanks for the update