hathach/tinyusb #1457: some examples don't work for imxrt:  cdc_msc_freertos, dynamic_configuration, hid_composite_freertos,  usbtmc

ccrome, 2022-05-04:
### Operating System

Linux

### Board

imxrt 1060, imxrt

### Firmware

examples/device/cdc_msc_freertos
examples/device/dynamic_configuration
examples/device/hid_composite_freertos
examples/device/usbtmc

None of these examples work on the teensy 4.0, using etither tag 0.13.0, or the latest fa895ed3dc46d5aa5e11b19ff07beb6fcfc62655.

**all other device examples work**

### What happened ?

I compile all examples using the tools/build_family.py.  They all compile properly with the latest commit (fa895ed3dc46d5aa5e11b19ff07beb6fcfc62655), but some build errors with 0.13.0.  

In any case, when I attempt to load the binaries into the teensy, the following examples simply stay in the teensy bootloader and never actually work properly:
cdc_msc_freertos, dynamic_configuration, hid_composite_freertos,  usbtmc

I get the same behavior on both the teensy and on the 1060 evk

### How to reproduce ?

run the build_family.py imxrt script, and attemp to load and use the aforementioned binaries.

ccrome, 2022-05-04:
Oh!  Found that on the 1060 EVK, some demos are on USB1 whereas others are on USB2.

So, it appears the 1060 EVK works on all examples, so now just left wondering why it doesn't work on all examples on the tensy 4

| teensy | 1060evk | example |
|--------|---------|---------|
|     OK |  OK USB1  |   audio_4_channel_mic     |
|     OK |  OK USB1  |   audio_test              |
|     OK |  OK       |   board_test              |
|     OK |  OK USB2  |   cdc_dual_ports          |
|     OK |  OK USB2  |   cdc_msc                 |
|     -- |  OK USB2  |   cdc_msc_freertos  device attempts to attach to USB 2, but gets errors on enumeration |
|     OK |  OK USB2  |   dfu                     |
|     OK |  OK USB2  |   dfu_runtime             |
|     -- |  OK USB2  |   dynamic_configuration   |
|     OK |  OK USB2  |   hid_boot_interface      |
|     OK |  OK USB2  |   hid_composite           |
|     -- |  OK USB2  |   hid_composite_freertos  |
|     OK |  OK USB2  |   hid_generic_inout       |
|     OK |  OK USB2  |   hid_multiple_interface  |
|     OK |  OK USB2  |   midi_test               |
|     OK |  OK USB2  |   msc_dual_lun            |
|     OK |  OK USB2  |   net_lwip_webserver      |
|     OK |  OK USB2  |   uac2_headset            |
|     -- |  OK USB2  |   usbtmc                  |
|     OK |  OK USB2  |   video_capture           |
|     OK |  OK USB2  |   webusb_serial           |

ccrome, 2022-05-04:
Update: 
edited `hw/bsp/imxrt/boards/mimxrt1060_evk/board.mk` to change 

```BOARD_DEVICE_RHPORT_NUM = 1```

to 

```BOARD_DEVICE_RHPORT_NUM = 0```

and recompiled.  Now the 1060 works on USB1 instead of USB2 (as expected), and that exact same *1060*.hex file loads into the teensy and runs (since they use the same chip, I was expecting it to).  So... chasing the difference now...

ccrome, 2022-05-04:
PR #1459 sure fixes this problem for me.  I'm still not sure how the pin gets misconfigured in the first place.

hathach, 2022-06-17:
Thanks for filing the issue, I put up a test today. The issue is not caused by mis-configured JTAG but rather the example incorrect bulk size of usb descriptor. For Highspeed the bulk endpoint size must be 512, dyanmic_configuration and usbtmc are hard-coded it into 64 (FS). kernel message does show this issue,

```
[ 4949.159993] usb 3-1: USB disconnect, device number 51
[ 4949.484276] usb 3-1: new high-speed USB device number 52 using xhci_hcd
[ 4949.633227] usb 3-1: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64
[ 4949.633230] usb 3-1: config 1 interface 0 altsetting 0 bulk endpoint 0x81 has invalid maxpacket 64
```   

and if you enable LOG debug of tinyusb it will also throw an assert on the issue as well. For freertos example, the issue is likely due to the irq priority, which you fixed in another PR.

This issue is fixed by

closed 2022-06-17 by hathach

ccrome, 2022-06-17:
> Thanks for filing the issue, I put up a test today. The issue is not caused by mis-configured JTAG but rather the example incorrect bulk size of usb descriptor. For Highspeed the bulk endpoint size [...]

Ah ha!  Thanks for the fix.

PR #1516 (2022-06-17): correct bulk size for highspeed endpoint in dynamic_configuration and usbtmc exmaples
**Describe the PR**
fix #1457, close #1459
bulk endpoint size must be 512 for highspeed and 64 for FS, dynamic_configuration and usbtmc examples hard coded to 64 and cause an assert in the stack

```
[ 4949.159993] usb 3-1: USB disconnect, device number 51
[ 4949.484276] usb 3-1: new high-speed USB device number 52 using xhci_hcd
[ 4949.633227] usb 3-1: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64
[ 4949.633230] usb 3-1: config 1 interface 0 altsetting 0 bulk endpoint 0x81 has  [... cut]