hathach/tinyusb #1464: Maybe remove mcu driver submodules and use some setup script

hathach, 2022-05-16:
**Is your feature request related to a problem? Please describe.**

Due to the wide range support of MCUs and the mcu sdk is often large enough. `git submodule init --recursive` which is convenient for most user project had a hard time with tinyusb. Maybe we should use an setup script to download mcu driver instead of git submodule instead.

perigoso, 2022-05-19:
we could make use of the build system for this, meson supports something like this directly for example #1423, since its only really used for the examples it makes sense to have it a bit less tightly coupled

hathach, 2022-05-21:
yeah, a build management (meson) can be useful way to manage multiple repos. Though I still want to have an simple python script with list of repos and commit hash since not everyone want to install an build management.

perigoso, 2022-05-21:
Well the only requirement for meson would be python, so if you're going with a python script it's not much different, but I hear you, that is also a good option

cr1901, 2022-05-27:
"Automatically retrieving deps for the examples" seems to already work via the `DEPS_SUBMODULES` `make` var. So removing submodules in place of a script that retrieves dependencies (_possibly with a manual override_?) probably won't be _that_ invasive of a change for building examples.

hathach, 2022-06-02:
@cr1901 right, i am planning to have an simple python script that take argument from `DEPS_SUBMODULES` and clone it accordingly.

PR #1947 (2023-03-11): Remove submodules and use python script to manage repo dependencies
**Describe the PR**
fix #1464 remove all submodules in hw/mcu, these repo is now managed by an python script at `tools/get_deps.py` which take 'submodule' path as inputs. The existing way to get dependency with make target `make BOARD={} get-deps` still works by invoking get_deps.py script.

This is due to the wide range support of MCUs and the mcu sdk is often large enough. `git submodule init --recursive` which is convenient for most user project had a hard time with tinyusb.
closed 2023-03-11 by hathach