hathach/tinyusb #1488: RP2040 host msc example non-functional (with fatfs)

pm-mic, 2022-06-02:
### Operating System

Windows 10

### Board

raspberry pi pico

### Firmware

examples/host/cdc_msc_hid

### What happened ?

When trying the msc host example with the fatfs implementation based on the example code, one first runs into the following code at line 97 and 107:
`if ( TUSB_ERROR_NONE != tuh_msc_read10(usb_addr, 0, buff, sector, count) )		return RES_ERROR;`
Which needs to be correct to: 
`if ( tuh_msc_read10(usb_addr, 0, buff, sector, count, NULL) )		return RES_ERROR;`
in order to make it compile. 

However after this: 
- disk_initialize succeeds
- disk_is_ready succeeds
- f_mount succeeds
- f_chdrive succeeds
- f_chdir fails with an error 1

The fix proposed here: https://github.com/hathach/tinyusb/pull/1434 changed the error into an error 13: no filesystem. Although it's clear there's a FAT32 with 512 block size on the USB stick. 

When checking the buffer returned from "disk_read" in the function check_fs() (which is called from f_chdir), there's supposed to be a record signature at offset 510, however this buffer is completely empty. It seems like tuh_msc_read10() did not succeed in filling the buffer with an actual value. 

### How to reproduce ?

1. Run the examples/host/cdc_msc_hid with only the msc example engaged, 
2. comment out the tuh_msc_inquiry() call (this will cause a race condition with the other calls)
3. uncomment the file system code below this call
4. fix the calls on line 97 and 107 in diskio.c
5. see an error 1

When doing those steps with the aforementioned fix from here: https://github.com/hathach/tinyusb/pull/1434
this will result in an code 13 error.

pm-mic, 2022-06-02:
I also tried this example with this fix: https://github.com/hathach/tinyusb/pull/1434 but this didn't change the result.

hathach, 2022-06-02:
Host bulk for rp2040 isn’t supported yet, I will take a look at this (and other related issues/prs) when having time.

zjwhitehead, 2022-09-18:
Hey @hathach any progress on this?
I know you're busy so willing to help sponsor this feature if needed. 
Let me know how I can help you!

rppicomidi, 2022-11-05:
@zjwhitehead Bulk support was added less than one day ago. You might find [this project](https://github.com/rppicomidi/pico-usb-host-msc-demo) helpful. it demonstrates RP2040 MSC Host with FatFs. It uses a terminal CLI for common file manipulation commands.

MasterQ32, 2022-11-07:
thank you so much @rppicomidi, i'm finally unblocked on my pico project and can now continue hacking on this. will keep track of this to finally update to a sane upstream version when this issue is resolved.

pete-pjb, 2023-01-11:
Hi @hathach , @MasterQ32 , @pm-mic @rppicomidi @zjwhitehead 

I am working on a host port for the Xilinx Zynq7000 SOC and I came across [this page](https://www.downtowndougbrown.com/2018/12/usb-mass-storage-with-embedded-devices-tips-and-quirks/) which contains some very useful information about USB and Mass Storage, if you haven't already seen, it I would suggest taking a look. 

 It's helped me a lot with my issues so far...

I hope it is helpful.

Kind Regards,

Pete

hathach, 2023-01-11:
this should be resolved in current master. Try out this example to see if that works for you https://github.com/hathach/tinyusb/tree/master/examples/host/msc_file_explorer

pete-pjb, 2023-01-11:
Hi @hathach ,

Thank you for your reply, I thought the link in my previous post might help everyone who is working on various MSC projects on this thread. I wasn't actually asking for help, but this is where I'm at with my port so far...

I am using FreeRTOS+FAT rather than the elm-Chan FAT so I have written my own drivers for +FAT. Currently I have managed to mount a filesystem and I can read files.  I am currently able to write files but I am having issues with data corruption, these are likely(as always😊) to be cache related as I am at least getting files of the correct size.  I also have HID - mouse, keyboard and an Illitek touch screen working with tinyUSB and I have connected Mouse/Touch Screen and MSC stick simultaneously via a USB2.0 four port hub.  Mostly things are working well but I do have some occasional instabilities, but I am sure I will get to the bottom of those when my USB protocol analyser arrives.

I have based my MSC operations on the `msc_file_explorer` example as you mentioned, the main issue I had was I needed to modify it so read and writes wait for their call-backs before continuing, I can share my code if anyone wants to see it at any point. Just ask 🙂 

When I get everything working with both the MSC and EHCI caching implementation, I will submit a PR and if the code is useful to the wider community that's all good. 🙂 I would also be happy to contribute the Zynq7000 port drivers if I can find a good way to integrate them with your BSP structure. 

Thanks again for your hard work bringing this library to the community. 👍 😃  

Kind Regards,

Pete

hathach, 2023-01-12:
@pete-pjb thank you for your good will, though each port is different. rp2040 host does not support Bulk endpoints previously, which is the main reason for this issue. 

Anyway, I think this issue is resolved and can now be closed. OP feel free to comment/re-open this if [host/msc_file_explorer](https://github.com/hathach/tinyusb/tree/master/examples/host/msc_file_explorer) does not work for you.

closed 2023-01-12 by hathach